MichaIng/DietPi #3422: Add Ombi to software

TristBella, 2020-03-16:
### Creating a software request

#### Formal software information
- Software title | OMBI
- Short description | Want a Movie or TV Show on Plex or Emby? Use Ombi!
- Official URL (if available) | https://ombi.io/
- GitHub URL (if available) | https://github.com/tidusjar/Ombi
- (Official) install documentation | https://github.com/tidusjar/Ombi/wiki/Installation

#### Are there similar/alternative software titles available with DietPi-Software?
Not that I see.
- ...

#### What makes your requested software better than the above solutions, if available?

- ...

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?
Normal apt works, but may be able to optimize. Not that familiar with the workings of DietPi.
- ...

#### Can you provide the installation steps that you would suggest DietPi-Software to do?
Not outside normal installation.

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.
I have limited knowledge, but would be willing to test and code where I can.
- ...

#### Vote for this software on FeatHub: https://feathub.com/MichaIng/DietPi/+147

MichaIng, 2020-03-16:
@TristBella 
Many thanks for your request.

Sounds like a reasonable extension to Plex and Emby and probably it will support further media players in the future.

elgringo81, 2021-05-29:
Im supprised that Ombi is not included. This service gives us a nice UI to request new shows and movies from Radarr, Sonarr and more. Its a must have.

kd9352, 2022-11-24:
Hello!
Maybe wrong place to write,
Someone using ombi on dietpi with nginx wan to share nginx config file?
I have nextcloud and jellyfin using nginx i am accessing them with my domain name. I couldnt fix ombi :/

Joulinar, 2022-11-24:
Ombi has on own documentation for this. https://docs.ombi.app/info/reverse-proxy/