MichaIng/DietPi #3425: Docker not Installing

tenryou-17, 2020-03-17:
#### Details:
- Date           | Tue 17 Mar 22:25:36 GMT 2020
- Bug report     | edecd16c-5053-416f-9934-caa1a1ea2953
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux DietPi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:
1. Install docker via dietpi-software

#### Expected behaviour:

-  Docker should install

#### Actual behaviour:

- Error being thrown when trying to install docker

#### Extra details:

- See bug report ref# edecd16c-5053-416f-9934-caa1a1ea2953

#### Additional logs:
```
Setting up docker-ce (5:19.03.8~3-0~raspbian-buster) ...
Job for docker.service failed because the control process exited with error code.
See "systemctl status docker.service" and "journalctl -xe" for details.
invoke-rc.d: initscript docker, action "start" failed.
● docker.service - Docker Application Container Engine
   Loaded: loaded (/lib/systemd/system/docker.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2020-03-17 22:25:21 GMT; 28ms ago
     Docs: https://docs.docker.com
  Process: 4475 ExecStart=/usr/bin/dockerd -H fd:// --containerd=/run/containerd/containerd.sock (code=exited, status=1/FAILURE)
 Main PID: 4475 (code=exited, status=1/FAILURE)
dpkg: error processing package docker-ce (--configure):
 installed docker-ce package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 docker-ce
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

I saw that there was a docker service that didn't start. So I tried uninstalling it. Here are the results:
```
root@DietPi:~# apt-get purge docker
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package 'docker' is not installed, so not removed
The following packages were automatically installed and are no longer required:
  libbsd0 libx11-6 libx11-data libxau6 libxcb1 libxdmcp6 wmdocker
Use 'apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up docker-ce (5:19.03.8~3-0~raspbian-buster) ...
Job for docker.service failed because the control process exited with error code.
See "systemctl status docker.service" and "journalctl -xe" for details.
invoke-rc.d: initscript docker, action "start" failed.
● docker.service - Docker Application Container Engine
   Loaded: loaded (/lib/systemd/system/docker.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2020-03-17 22:36:23 GMT; 27ms ago
     Docs: https://docs.docker.com
  Process: 4988 ExecStart=/usr/bin/dockerd -H fd:// --containerd=/run/containerd/containerd.sock (code=exited, status=1/FAILURE)
 Main PID: 4988 (code=exited, status=1/FAILURE)
dpkg: error processing package docker-ce (--configure):
 installed docker-ce package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 docker-ce
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

dweee, 2020-03-17:
I had this issue and I think it because I have a non-standard IP range on my network. Do `journalctl -u docker` and see if it mentions anything to do with allocating IP addresses or "no available network" and something to do with bridges. If so, try disabling any VPN clients on the system or disconnecting NordVPN and `apt purge docker-ce` then try reinstalling manually or through dietpi-software. 

Good luck figuring it out, also if you're thinking of using Nord on some of the dockers, try using a solution like [this](https://github.com/bubuntux/nordvpn) as it can help if you have any programs you don't want running through the VPN.

tenryou-17, 2020-03-18:
Thanks for the response @dweee. `journalctl -u docker` doesn't reveal any network related issues. I don't have any VPNs running on the Pi, so that isn't an issue.

I did try `apt remove --purge docker` and  `apt remove --purge docker-ce`. That removed any trace of the installation. I tried again using Dietpi-Software with no luck. However, when I install Docker the conventional way using `apt-get install docker`, it works like a charm. 

I'm not sure what the `./DockerInstall.sh` script is doing when going the Dietpi-Software route. I will have to investigate further. 

Also tried uninstalling and installing on my VM and it seems to work.

MichaIng, 2020-03-18:
@tenryou-17 
Many thanks for your report. Seems Docker vs RPi issues are a never-ending story... 😢.

We use the official Docker installer from https://get.docker.com. The script is hosted on GitHub: https://github.com/docker/docker-install/blob/master/install.sh
- It installs the official APT repo
- It installs a specific docker-ce-cli and docker-ce version from this repo

And btw. the [docker](https://packages.debian.org/buster/docker) package that you installed, has nothing to do with "Docker", it is a totally different app that only shares the same name, which is the reason why the actual "Docker" package is called `docker-ce` 😉.

So basically the script does everything correct, but the Docker service fails, please check:
```
journalctl -u docker
journalctl -u containerd
```

tenryou-17, 2020-03-18:
> And btw. the docker package that you installed, has nothing to do with "Docker", it is a totally different app that only shares the same name, which is the reason why the actual "Docker" package is called docker-ce 😉.

Well I'll be...the more you now. I guess the lesson here is 'never assume' :P

So after trying again through `Dietpi-Software` here is the output from `journalctl -u docker`

```
-- Logs begin at Tue 2020-03-17 22:14:42 GMT, end at Wed 2020-03-18 15:17:28 GMT. --
Mar 17 22:19:24 DietPi systemd[1]: Starting Docker Application Container Engine...
Mar 17 22:19:24 DietPi dockerd[2050]: time="2020-03-17T22:19:24.714132116Z" level=info msg="Starting up"
Mar 17 22:19:24 DietPi dockerd[2050]: failed to load listeners: no sockets found via socket activation: make sure the service was started by systemd
Mar 17 22:19:24 DietPi systemd[1]: docker.service: Main process exited, code=exited, status=1/FAILURE
Mar 17 22:19:24 DietPi systemd[1]: docker.service: Failed with result 'exit-code'.
Mar 17 22:19:24 DietPi systemd[1]: Failed to start Docker Application Container Engine.
Mar 17 22:19:26 DietPi systemd[1]: docker.service: Service RestartSec=2s expired, scheduling restart.
Mar 17 22:19:26 DietPi systemd[1]: docker.service: Scheduled restart job, restart counter is at 1.
Mar 17 22:19:26 DietPi systemd[1]: Stopped Docker Application Container Engine.
Mar 17 22:19:26 DietPi systemd[1]: Starting Docker Application Container Engine...
Mar 17 22:19:27 DietPi dockerd[2088]: time="2020-03-17T22:19:27.010992479Z" level=info msg="Starting up"
Mar 17 22:19:27 DietPi dockerd[2088]: time="2020-03-17T22:19:27.016254823Z" level=info msg="parsed scheme: \"unix\"" module=grpc
Mar 17 22:19:27 DietPi dockerd[2088]: time="2020-03-17T22:19:27.016891125Z" level=info msg="scheme \"unix\" not registered, fallback to default scheme" module=grpc
Mar 17 22:19:27 DietPi dockerd[2088]: time="2020-03-17T22:19:27.017315709Z" level=info msg="ccResolverWrapper: sending update to cc: {[{unix:///run/containerd/container
d.sock 0  <nil>}] <nil>}" module=grpc
Mar 17 22:19:27 DietPi dockerd[2088]: time="2020-03-17T22:19:27.017689979Z" level=info msg="ClientConn switching balancer to \"pick_first\"" module=grpc
Mar 17 22:19:27 DietPi dockerd[2088]: time="2020-03-17T22:19:27.023709250Z" level=info msg="parsed scheme: \"unix\"" module=grpc
Mar 17 22:19:27 DietPi dockerd[2088]: time="2020-03-17T22:19:27.024338365Z" level=info msg="scheme \"unix\" not registered, fallback to default scheme" module=grpc
Mar 17 22:19:27 DietPi dockerd[2088]: time="2020-03-17T22:19:27.024757479Z" level=info msg="ccResolverWrapper: sending update to cc: {[{unix:///run/containerd/container
d.sock 0  <nil>}] <nil>}" module=grpc
Mar 17 22:19:27 DietPi dockerd[2088]: time="2020-03-17T22:19:27.025148886Z" level=info msg="ClientConn switching balancer to \"pick_first\"" module=grpc
Mar 17 22:19:28 DietPi dockerd[2088]: time="2020-03-17T22:19:28.117496594Z" level=warning msg="Your kernel does not support swap memory limit"
Mar 17 22:19:28 DietPi dockerd[2088]: time="2020-03-17T22:19:28.119103364Z" level=warning msg="Your kernel does not support cgroup cfs period"
Mar 17 22:19:28 DietPi dockerd[2088]: time="2020-03-17T22:19:28.119540239Z" level=warning msg="Your kernel does not support cgroup cfs quotas"
Mar 17 22:19:28 DietPi dockerd[2088]: time="2020-03-17T22:19:28.119900864Z" level=warning msg="Your kernel does not support cgroup rt period"
Mar 17 22:19:28 DietPi dockerd[2088]: time="2020-03-17T22:19:28.120230500Z" level=warning msg="Your kernel does not support cgroup rt runtime"
Mar 17 22:19:28 DietPi dockerd[2088]: time="2020-03-17T22:19:28.121619771Z" level=info msg="Loading containers: start."
Mar 17 22:19:30 DietPi dockerd[2088]: time="2020-03-17T22:19:30.627551072Z" level=info msg="stopping event stream following graceful shutdown" error="<nil>" module=libc
ontainerd namespace=moby
Mar 17 22:19:30 DietPi dockerd[2088]: failed to start daemon: Error initializing network controller: list bridge addresses failed: PredefinedLocalScopeDefaultNetworks L
ist: [172.17.0.0/16 172.18.0.0/16 172.19.0.0/16 172.20.0.0/16 172.21.0.0/16 172.22.0.0/16 172.23.0.0/16 172.24.0.0/16 172.25.0.0/16 172.26.0.0/16 172.27.0.0/16 172.28.0
.0/16 172.29.0.0/16 172.30.0.0/16 172.31.0.0/16 192.168.0.0/20 192.168.16.0/20 192.168.32.0/20 192.168.48.0/20 192.168.64.0/20 192.168.80.0/20 192.168.96.0/20 192.168.1
12.0/20 192.168.128.0/20 192.168.144.0/20 192.168.160.0/20 192.168.176.0/20 192.168.192.0/20 192.168.208.0/20 192.168.224.0/20 192.168.240.0/20]: no available network
Mar 17 22:19:30 DietPi systemd[1]: docker.service: Main process exited, code=exited, status=1/FAILURE
Mar 17 22:19:30 DietPi systemd[1]: docker.service: Failed with result 'exit-code'.
Mar 17 22:19:30 DietPi systemd[1]: Failed to start Docker Application Container Engine.
Mar 17 22:19:32 DietPi systemd[1]: docker.service: Service RestartSec=2s expired, scheduling restart.
Mar 17 22:19:32 DietPi systemd[1]: docker.service: Scheduled restart job, restart counter is at 2.
Mar 17 22:19:32 DietPi systemd[1]: Stopped Docker Application Container Engine.
Mar 17 22:19:32 DietPi systemd[1]: Starting Docker Application Container Engine...
Mar 17 22:19:33 DietPi dockerd[2158]: time="2020-03-17T22:19:33.030145654Z" level=info msg="Starting up"
Mar 17 22:19:33 DietPi dockerd[2158]: time="2020-03-17T22:19:33.035205446Z" level=info msg="parsed scheme: \"unix\"" module=grpc
```

[... cut ...]
`Mar 17 22:19:30 DietPi dockerd[2088]: failed to start daemon: Error initializing network controller: list bridge addresses failed` I googled this issue and came up with [#18283](https://github.com/moby/moby/issues/18283) and [#17083](https://github.com/moby/moby/issues/17083). 

What I tried: after I `apt autoremove --purge docker docker-ce` I stopped the OpenVPN service and I did `ip link delete tun0` to delete the VPN bridge. Per the feedback from the two issues I linked before I deleted `/var/lib/docker` and tried reinstalling via Dietpi-Software.

Still no luck 😞

dweee, 2020-03-18:
I had the same issue, If I recall correctly, I used a solution from [here](https://github.com/docker/for-linux/issues/123#issuecomment-346546953) to fix the issue. I should be resetting my DietPi soon anyways as I've figured out a better setup for it. You could try resetting and installing docker first? If it still fails, try purging it, doing the solution I referenced earlier and then attempt installing it again. If I run into this issue again, I will let you know if I fixed it differently.

Hope you fix it soon, king.

tenryou-17, 2020-03-18:
> I had the same issue, If I recall correctly, I used a solution from [here](https://github.com/docker/for-linux/issues/123#issuecomment-346546953) to fix the issue. I should be resetting my DietPi [...]

Well this is indeed odd. 

I purged docker from my system, deleted the `/var/lib/docker` folder, tried the recommended fix from [here](https://github.com/docker/for-linux/issues/123#issuecomment-346546953), and installed docker via Dietpi-Software. Installation threw an error again. I then threw my hands up in defeat and gave up. I rebooted the pi and checked on the status of my services. Everything, including docker, was active and running. 

To double check I ran `docker run hello-world`. I got the "Hello from Docker!" message indicating everything was working correctly. 

I guess it's solved by accident? Sage old advice: did you restart your computer? 😆

MichaIng, 2020-03-18:
@tenryou-17 
> Docker does not appear on the list of software to uninstall

Yes if install failed, it is not saved as installed, hence cannot be uninstalled. Not idea, would be good to store it somehow as "install started", so uninstall can be done to clean everything.

> I guess it's solved by accident?

Probably your kernel modules were updated? I found that earlier, would make sense to block install if active kernel does not match /lib/modules/ entry, which usually indicates a kernel upgrade. In such case a reboot is required to install things like Docker and WireGuard.

> "modprobe: FATAL: Module aufs not found in directory /lib/modules/4.19.97-v7+\n": exit status 1"

That actually indicates the thing.

@dweee 
Thanks for sharing your solution. While in some cases it indeed might be some network-related issue, also in the thread you linked the missing reboot after kernel upgrade appeared to be the culprit 😉: https://github.com/docker/for-linux/issues/123#issuecomment-491102033

tenryou-17, 2020-03-18:
More and more I am continually impressed by this community. Thank you both for your help! And I'm glad my 'accident' got added as a milestone to achieve 😝

MichaIng, 2020-06-24:
#### Done
[image]

closed 2020-06-30 by MichaIng