MichaIng/DietPi #3427: redis-server error interferes with resetting of Nextcloud

edo0, 2020-03-21:
#### Details:
- Date           | Sat 21 Mar 15:59:18 CET 2020
- Bug report     | 4c3ad55a-bcb4-4389-85b5-86813af122f9
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | Linux TheBox 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
- Distro         | stretch (ID=4)
- Command        | systemctl restart redis-server
- Exit code      | 1
- Software title | DietPi-Software

#### Steps to reproduce:

1. Had an issue while updating Nextcloud, and my instance got broken
2. I tried restoring a Dietpi backup I had made, but afterwards realized it was too old and that lead to a series of issues (among which, expired certificate and more)
3. I decided to go for dietpi-software reinstall 114 to reinstall nextcloud. That, and subsequent attemps to remore and install again Nextcloud, always give me the error I am reporting 

#### Expected behaviour:

- Reinstallation of Nextcloud should happen smoothly

#### Actual behaviour:

- The installation hangs with the reported error

#### Additional logs:
```
Job for redis-server.service failed because the control process exited with error code.
See "systemctl status redis-server.service" and "journalctl -xe" for details.
```

I thank wholeheartedly in advance whoever will be able to give me a hand troubleshoot this!!
 

---------------------------------------------------------------------

MichaIng, 2020-03-21:
@edo0 
Many thanks for your report. Please paste the output of:
```
journalctl -u redis-server
```

edo0, 2020-03-21:
Thanks to you Micha! Here is the report
``` root@TheBox:~# journalctl -u redis-server
-- Logs begin at Thu 2016-11-03 18:16:42 CET, end at Sat 2020-03-21 17:06:54 CET
. --
Mar 21 15:45:22 TheBox systemd[1]: Starting Advanced key-value store...
Mar 21 15:45:22 TheBox run-parts[780]: run-parts: executing /etc/redis/redis-ser
ver.pre-up.d/00_example
Mar 21 15:45:22 TheBox systemd[1]: redis-server.service: Control process
 exited, code=exited status=1
Mar 21 15:45:22 TheBox systemd[1]: Failed to start Advanced key-value st
ore.
Mar 21 15:45:22 TheBox systemd[1]: redis-server.service: Unit entered fa
iled state.
Mar 21 15:45:22 TheBox systemd[1]: redis-server.service: Failed with res
ult 'exit-code'.
Mar 21 15:45:23 TheBox systemd[1]: redis-server.service: Service hold-off time o
ver, scheduling restart.
Mar 21 15:45:23 TheBox systemd[1]: Stopped Advanced key-value store.
Mar 21 15:45:23 TheBox systemd[1]: Starting Advanced key-value store...
Mar 21 15:45:23 TheBox run-parts[821]: run-parts: executing /etc/redis/redis-ser
ver.pre-up.d/00_example
Mar 21 15:45:23 TheBox systemd[1]: redis-server.service: Control process
 exited, code=exited status=1
Mar 21 15:45:23 TheBox systemd[1]: Failed to start Advanced key-value st
ore.
Mar 21 15:45:23 TheBox systemd[1]: redis-server.service: Unit entered failed state.
Mar 21 15:45:23 TheBox systemd[1]: redis-server.service: Failed with result 'exit-code'.
Mar 21 15:45:23 TheBox systemd[1]: redis-server.service: Service hold-off time over, scheduling restart.
Mar 21 15:45:23 TheBox systemd[1]: Stopped Advanced key-value store.
Mar 21 15:45:23 TheBox systemd[1]: Starting Advanced key-value store...
Mar 21 15:45:24 TheBox redis-server[844]: *** FATAL CONFIG FILE ERROR ***
Mar 21 15:45:24 TheBox redis-server[844]: Reading the configuration file, at line 1053
Mar 21 15:45:24 TheBox redis-server[844]: >>> 'always-show-logo no'
Mar 21 15:45:24 TheBox redis-server[844]: Bad directive or wrong number of arguments
Mar 21 15:45:24 TheBox systemd[1]: redis-server.service: Control process exited, code=exited status=1
Mar 21 15:45:24 TheBox systemd[1]: Failed to start Advanced key-value store.
Mar 21 15:45:24 TheBox systemd[1]: redis-server.service: Unit entered failed state.
Mar 21 15:45:24 TheBox systemd[1]: redis-server.service: Failed with result 'exit-code'.
Mar 21 15:45:24 TheBox systemd[1]: redis-server.service: Service hold-off time over, scheduling restart.
Mar 21 15:45:24 TheBox systemd[1]: Stopped Advanced key-value store.
Mar 21 15:45:24 TheBox systemd[1]: Starting Advanced key-value store...
Mar 21 15:45:24 TheBox run-parts[938]: run-parts: executing /etc/redis/redis-server.pre-up.d/00_example
Mar 21 15:45:24 TheBox systemd[1]: redis-server.service: Control process exited, code=exited status=1
Mar 21 15:45:24 TheBox systemd[1]: Failed to start Advanced key-value store.
Mar 21 15:45:24 TheBox systemd[1]: redis-server.service: Unit entered failed state.
Mar 21 15:45:24 TheBox systemd[1]: redis-server.service: Failed with result 'exit-code'.
Mar 21 15:45:25 TheBox systemd[1]: redis-server.service: Service hold-off time over, scheduling restart.
Mar 21 15:45:25 TheBox systemd[1]: Stopped Advanced key-value store.
Mar 21 15:45:25 TheBox systemd[1]: Starting Advanced key-value store...
Mar 21 15:45:25 TheBox run-parts[954]: run-parts: executing /etc/redis/redis-server.pre-up.d/00_example
Mar 21 15:45:25 TheBox systemd[1]: redis-server.service: Control process exited, code=exited status=1
Mar 21 15:45:25 TheBox systemd[1]: Failed to start Advanced key-value store.
Mar 21 15:45:25 TheBox systemd[1]: redis-server.service: Unit entered failed state.
Mar 21 15:45:25 TheBox systemd[1]: redis-server.service: Failed with result 'exit-code'.
Mar 21 15:45:25 TheBox systemd[1]: redis-server.service: Service hold-off time over, scheduling restart.
Mar 21 15:45:25 TheBox systemd[1]: Stopped Advanced key-value store.
Mar 21 15:45:25 TheBox systemd[1]: redis-server.service: Start request repeated too quickly.
Mar 21 15:45:25 TheBox systemd[1]: Failed to start Advanced key-value store.
Mar 21 15:45:25 TheBox systemd[1]: redis-server.service: Unit entered failed state.
Mar 21 15:45:25 TheBox systemd[1]: redis-server.service: Failed with result 'exit-code'.
Mar 21 15:45:25 TheBox systemd[1]: redis-server.service: Service hold-off time over, scheduling restart.
Mar 21 15:45:25 TheBox systemd[1]: Stopped Advanced key-value store.
Mar 21 15:45:25 TheBox systemd[1]: redis-server.service: Start request repeated too quickly.
Mar 21 15:45:25 TheBox systemd[1]: Failed to start Advanced key-value store.
Mar 21 15:45:25 TheBox systemd[1]: redis-server.service: Unit entered failed state.
Mar 21 15:45:25 TheBox systemd[1]: redis-server.service: Failed with result 'exit-code'.
Mar 21 15:50:49 TheBox systemd[1]: Starting Advanced key-value store...
Mar 21 15:50:49 TheBox run-parts[6139]: run-parts: executing /etc/redis/redis-server.pre-up.d/00_example
Mar 21 15:50:50 TheBox redis-server[6144]: *** FATAL CONFIG FILE ERROR ***
Mar 21 15:50:50 TheBox redis-server[6144]: Reading the configuration file, at line 1053
[... 24 lines cut ...]
Mar 21 15:50:51 TheBox systemd[1]: redis-server.service: Unit entered failed state.
Mar 21 15:50:51 TheBox systemd[1]: redis-server.service: Failed with result 'exit-code'.
Mar 21 15:50:51 TheBox systemd[1]: redis-server.service: Service hold-off time over, scheduling restart.
Mar 21 15:50:51 TheBox systemd[1]: Stopped Advanced key-value store.
[... cut ...]
Mar 21 15:50:53 TheBox systemd[1]: redis-server.service: Service hold-off time over, scheduling restart.
Mar 21 15:50:53 TheBox systemd[1]: Stopped Advanced key-value store.
Mar 21 15:50:53 TheBox systemd[1]: redis-server.service: Start request repeated too quickly.
Mar 21 15:50:53 TheBox systemd[1]: Failed to start Advanced key-value store.
Mar 21 15:50:53 TheBox systemd[1]: redis-server.service: Unit entered failed state.
Mar 21 15:50:53 TheBox systemd[1]: redis-server.service: Failed with result 'exit-code'.
```

MichaIng, 2020-03-21:
@edo0 
Hmm, the logs are not helpful please try to reinstall it:
```
apt purge redis-server
rm -Rf /etc/redis
apt install redis-server
G_CONFIG_INJECT 'loglevel[[:blank:]]' 'loglevel warning' /etc/redis/redis.conf
G_CONFIG_INJECT 'logfile[[:blank:]]' 'logfile ""' /etc/redis/redis.conf
G_CONFIG_INJECT 'syslog-enabled[[:blank:]]' 'syslog-enabled yes' /etc/redis/redis.conf
G_CONFIG_INJECT 'always-show-logo[[:blank:]]' 'always-show-logo no' /etc/redis/redis.conf
G_CONFIG_INJECT 'unixsocket[[:blank:]]' 'unixsocket /run/redis/redis-server.sock' /etc/redis/redis.conf
G_CONFIG_INJECT 'unixsocketperm[[:blank:]]' 'unixsocketperm 770' /etc/redis/redis.conf
usermod -aG redis www-data
```

edo0, 2020-03-21:
The same issue presented itself again ([bug report](https://pastebin.com/jj3Z33Pp), omitted since identical)

During removal, this message appeared, not sure if it can be of any help

```
the following packages will be REMOVED:
  redis-server*
0 upgraded, 0 newly installed, 1 to remove and 1 not upgraded.
After this operation, 877 kB disk space will be freed.
Do you want to continue? [Y/n] 
(Reading database ... 25718 files and directories currently installed.)
Removing redis-server (3:3.2.6-3+deb9u3) ...
(Reading database ... 25709 files and directories currently installed.)
**Purging configuration files for redis-server (3:3.2.6-3+deb9u3) ...
userdel: group redis not removed because it has other members.**
Processing triggers for systemd (232-25+deb9u12) ...
```

edo0, 2020-03-21:
These are the more detailed logs

```
root@TheBox:~# journalctl -u redis-server
-- Logs begin at Thu 2016-11-03 18:16:42 CET, end at Sun 2020-03-22 00:30:55 CET. --
Mar 21 15:45:22 TheBox systemd[1]: Starting Advanced key-value store...
Mar 21 15:45:22 TheBox run-parts[780]: run-parts: executing /etc/redis/redis-server.pre-up.d/00_example
Mar 21 15:45:22 TheBox systemd[1]: redis-server.service: Control process exited, code=exited status=1
Mar 21 15:45:22 TheBox systemd[1]: Failed to start Advanced key-value store.
Mar 21 15:45:22 TheBox systemd[1]: redis-server.service: Unit entered failed state.
Mar 21 15:45:22 TheBox systemd[1]: redis-server.service: Failed with result 'exit-code'.
Mar 21 15:45:23 TheBox systemd[1]: redis-server.service: Service hold-off time over, scheduling restart.
Mar 21 15:45:23 TheBox systemd[1]: Stopped Advanced key-value store.
Mar 21 15:45:23 TheBox systemd[1]: Starting Advanced key-value store...
Mar 21 15:45:23 TheBox run-parts[821]: run-parts: executing /etc/redis/redis-server.pre-up.d/00_example
Mar 21 15:45:23 TheBox systemd[1]: redis-server.service: Control process exited, code=exited status=1
Mar 21 15:45:23 TheBox systemd[1]: Failed to start Advanced key-value store.
Mar 21 15:45:23 TheBox systemd[1]: redis-server.service: Unit entered failed state.
Mar 21 15:45:23 TheBox systemd[1]: redis-server.service: Failed with result 'exit-code'.
Mar 21 15:45:23 TheBox systemd[1]: redis-server.service: Service hold-off time over, scheduling restart.
Mar 21 15:45:23 TheBox systemd[1]: Stopped Advanced key-value store.
Mar 21 15:45:23 TheBox systemd[1]: Starting Advanced key-value store...
Mar 21 15:45:24 TheBox redis-server[844]: *** FATAL CONFIG FILE ERROR ***
Mar 21 15:45:24 TheBox redis-server[844]: Reading the configuration file, at line 1053
Mar 21 15:45:24 TheBox redis-server[844]: >>> 'always-show-logo no'
Mar 21 15:45:24 TheBox redis-server[844]: Bad directive or wrong number of arguments
Mar 21 15:45:24 TheBox systemd[1]: redis-server.service: Control process exited, code=exited status=1
Mar 21 15:45:24 TheBox systemd[1]: Failed to start Advanced key-value store.
Mar 21 15:45:24 TheBox systemd[1]: redis-server.service: Unit entered failed state.
Mar 21 15:45:24 TheBox systemd[1]: redis-server.service: Failed with result 'exit-code'.
Mar 21 15:45:24 TheBox systemd[1]: redis-server.service: Service hold-off time over, scheduling restart.
Mar 21 15:45:24 TheBox systemd[1]: Stopped Advanced key-value store.
Mar 21 15:45:24 TheBox systemd[1]: Starting Advanced key-value store...
Mar 21 15:45:24 TheBox run-parts[938]: run-parts: executing /etc/redis/redis-server.pre-up.d/00_example
Mar 21 15:45:24 TheBox systemd[1]: redis-server.service: Control process exited, code=exited status=1
Mar 21 15:45:24 TheBox systemd[1]: Failed to start Advanced key-value store.
Mar 21 15:45:24 TheBox systemd[1]: redis-server.service: Unit entered failed state.
Mar 21 15:45:24 TheBox systemd[1]: redis-server.service: Failed with result 'exit-code'.
Mar 21 15:45:25 TheBox systemd[1]: redis-server.service: Service hold-off time over, scheduling restart.
Mar 21 15:45:25 TheBox systemd[1]: Stopped Advanced key-value store.
Mar 21 15:45:25 TheBox systemd[1]: Starting Advanced key-value store...
Mar 21 15:45:25 TheBox run-parts[954]: run-parts: executing /etc/redis/redis-server.pre-up.d/00_example
Mar 21 15:45:25 TheBox systemd[1]: redis-server.service: Control process exited, code=exited status=1
Mar 21 15:45:25 TheBox systemd[1]: Failed to start Advanced key-value store.
Mar 21 15:45:25 TheBox systemd[1]: redis-server.service: Unit entered failed state.
Mar 21 15:45:25 TheBox systemd[1]: redis-server.service: Failed with result 'exit-code'.
Mar 21 15:45:25 TheBox systemd[1]: redis-server.service: Service hold-off time over, scheduling restart.
Mar 21 15:45:25 TheBox systemd[1]: Stopped Advanced key-value store.
Mar 21 15:45:25 TheBox systemd[1]: redis-server.service: Start request repeated too quickly.
Mar 21 15:45:25 TheBox systemd[1]: Failed to start Advanced key-value store.
Mar 21 15:45:25 TheBox systemd[1]: redis-server.service: Unit entered failed state.
Mar 21 15:45:25 TheBox systemd[1]: redis-server.service: Failed with result 'exit-code'.
Mar 21 15:50:49 TheBox systemd[1]: Starting Advanced key-value store...
Mar 21 15:50:49 TheBox run-parts[6139]: run-parts: executing /etc/redis/redis-server.pre-up.d/00_example
Mar 21 15:50:50 TheBox redis-server[6144]: *** FATAL CONFIG FILE ERROR ***
Mar 21 15:50:50 TheBox redis-server[6144]: Reading the configuration file, at line 1053
Mar 21 15:50:50 TheBox redis-server[6144]: >>> 'always-show-logo no'
Mar 21 15:50:50 TheBox redis-server[6144]: Bad directive or wrong number of arguments
Mar 21 15:50:50 TheBox systemd[1]: redis-server.service: Control process exited, code=exited status=1
Mar 21 15:50:50 TheBox systemd[1]: Failed to start Advanced key-value store.
Mar 21 15:50:50 TheBox systemd[1]: redis-server.service: Unit entered failed state.
Mar 21 15:50:50 TheBox systemd[1]: redis-server.service: Failed with result 'exit-code'.
Mar 21 15:50:50 TheBox systemd[1]: redis-server.service: Service hold-off time over, scheduling restart.
Mar 21 15:50:50 TheBox systemd[1]: Stopped Advanced key-value store.
Mar 21 15:50:50 TheBox systemd[1]: Starting Advanced key-value store...
Mar 21 15:50:50 TheBox run-parts[6158]: run-parts: executing /etc/redis/redis-server.pre-up.d/00_example
[... cut ...]
Mar 22 00:30:55 TheBox redis-server[12044]: *** FATAL CONFIG FILE ERROR ***
Mar 22 00:30:55 TheBox redis-server[12044]: Reading the configuration file, at line 1053
Mar 22 00:30:55 TheBox redis-server[12044]: >>> 'always-show-logo no'
Mar 22 00:30:55 TheBox redis-server[12044]: Bad directive or wrong number of arguments
Mar 22 00:30:55 TheBox systemd[1]: redis-server.service: Control process exited, code=exited status=1
Mar 22 00:30:55 TheBox systemd[1]: Failed to start Advanced key-value store.
```

MichaIng, 2020-03-21:
@edo0 
Ah, the `always-show-logo` is not yet supported by the old Redis version on Stretch 🤔. Please do:
```
sed -i '/always-show-logo/d' /etc/redis/redis.conf
systemctl restart redis-server
```

edo0, 2020-03-22:
Thank you, that did indeed solve it, and allowed me to finalize the reinstallation.

I was able to get the core of nextcloud up and running (WebDAV, CalDAV, Notes, App synchronization etc), but as of immediately after the installation, I still seem to have some issues with the web interface.

I link my [nextcloud config file](https://pastebin.com/EH8UMYEU)  and attach a couple of screenshots

[image]
[image]

MichaIng, 2020-03-22:
@edo0 
Please try flush the browser cache (often crtl+f5).

edo0, 2020-03-22:
Yes, I thought of that. Unfortunately, even flushing cache, opening in incognito, or on other laptop and  mobile devices (mine and of my significant other), the interface is still broken.

MichaIng, 2020-03-22:
@edo0 
Hmm, the installer failed in the middle before?
Please try to reinstall Nextcloud, after removing the culprit:
```
sed -i '/always-show-logo/d' /DietPi/dietpi/dietpi-software
dietpi-software reinstall 114
```

edo0, 2020-03-22:
@MichaIng 
I have tried reinstalling again a couple of times, but the outcome is always the same. Passed the login, the web interface comes out as broken

MichaIng, 2020-03-22:
@edo0 
Did you enable any themes on Nextcloud or such?

edo0, 2020-03-22:
No, I had actually disabled the provided Theming application

MichaIng, 2020-03-22:
@edo0 
I checked your bug report and there are some more failing/conflicting services and some that I cannot see at all, even that their holding packages are installed 🤔:
```
systemctl disable --now dhcpcd
apt purge dhcpcd5
```
- Not required, failing, since conflicting with ifupdown+dhclient

```
systemctl disable --now wpa_supplicant
```
- The static service is not required, failing, since conflicting with ifupdown per-interface invocation

```
ls -Al /lib/systemd/system/nginx.service
```
- Does this exist? If not:
```
apt install --reinstall nginx-common
systemctl status nginx
```

```
ls -Al /lib/systemd/system/php7.3-fpm.service
```
- Does this exist? If not:
```
apt install --reinstall php7.3-fpm
systemctl status php7.3-fpm
```

edo0, 2020-03-22:
I got rid of the first two services, and confirmed that 

```
● redis-server.service - Advanced key-value store
   Loaded: loaded (/lib/systemd/system/redis-server.service; disabled; vendor pr
eset: enabled)
   Active: active (running) since Sun 2020-03-22 18:30:37 CET; 1h 8mi
```

```
● php7.3-fpm.service - The PHP 7.3 FastCGI Process Manager
   Loaded: loaded (/lib/systemd/system/php7.3-fpm.service; disabled; vendor pres
et: enabled)
   Active: active (running) since Sun 2020-03-22 18:30:42 CET; 1h 8mi
```
and
```
● nginx.service - A high performance web server and a reverse proxy s
erver
   Loaded: loaded (/lib/systemd/system/nginx.service; disabled; vendor preset: e
nabled)
   Active: active (running) since Sun 2020-03-22 18:30:43 CET; 1h 8mi
```

edo0, 2020-03-22:
However, the issue still persists

MichaIng, 2020-03-22:
@edo0 
Can you try to update Nextcloud:
```
sudo -u www-data /var/www/nextcloud/updater/updater.phar
ncc upgrade
```

edo0, 2020-03-22:
Thank you so much @MichaIng .
That solved it for me, and now everything is back up and running!!

MichaIng, 2020-03-22:
@edo0 
Great, probably there was some mismatch with the installed version + database or similar.

edo0, 2020-03-22:
Yes, that seems to have been the case!

closed 2020-05-03 by MichaIng