MichaIng/DietPi #3430: Jackett service not active on boot

Dr0bac, 2020-03-22:
DietPi v6.28.0 - RPi 3 Model B (armv7l)

I noticed for few days that indexers in radarr that come with jackett are not displayed.
Than i noticed in htop that jackett is not present. And with jackett still not present with few system reboots i went to dietpi-service and saw this

[image]

MichaIng, 2020-03-22:
@Dr0bac 
Many thanks for your report.

Please paste the output of:
```
systemctl status jackett
journalctl -u jackett
```

Dr0bac, 2020-03-22:
````
root@DietPi:~# systemctl status jackett
● jackett.service - Jackett (DietPi)
   Loaded: loaded (/etc/systemd/system/jackett.service; disabled; vendor preset: enabled)
   Active: activating (auto-restart) since Sun 2020-03-22 13:06:24 CET; 2s ago
  Process: 8004 ExecStart=/bin/dash -c /opt/jackett/jackett --NoRestart; while pgrep -u jackett JackettUpdater; do sleep 1; done (code=exited, status=0/SUCCESS) Main PID: 8004 (code=exited, status=0/SUCCESS)
root@DietPi:~# journalctl -u jackett
-- Logs begin at Sun 2020-03-22 11:43:59 CET, end at Sun 2020-03-22 13:06:30 CET. --
Mar 22 11:44:32 DietPi systemd[1]: Started Jackett (DietPi).
Mar 22 11:44:32 DietPi jackett[774]: /bin/dash: 1: /opt/jackett/jackett: not found
Mar 22 11:44:32 DietPi systemd[1]: jackett.service: Succeeded.
Mar 22 11:44:37 DietPi systemd[1]: jackett.service: Service RestartSec=5s expired, scheduling restart.
Mar 22 11:44:37 DietPi systemd[1]: jackett.service: Scheduled restart job, restart counter is at 1.
Mar 22 11:44:37 DietPi systemd[1]: Stopped Jackett (DietPi).
Mar 22 11:44:37 DietPi systemd[1]: Started Jackett (DietPi).
Mar 22 11:44:38 DietPi jackett[864]: /bin/dash: 1: /opt/jackett/jackett: not found
Mar 22 11:44:38 DietPi systemd[1]: jackett.service: Succeeded.
Mar 22 11:44:43 DietPi systemd[1]: jackett.service: Service RestartSec=5s expired, scheduling restart.
Mar 22 11:44:43 DietPi systemd[1]: jackett.service: Scheduled restart job, restart counter is at 2.
Mar 22 11:44:43 DietPi systemd[1]: Stopped Jackett (DietPi).
Mar 22 11:44:43 DietPi systemd[1]: Started Jackett (DietPi).
Mar 22 11:44:43 DietPi jackett[873]: /bin/dash: 1: /opt/jackett/jackett: not found
Mar 22 11:44:43 DietPi systemd[1]: jackett.service: Succeeded.
Mar 22 11:44:48 DietPi systemd[1]: jackett.service: Service RestartSec=5s expired, scheduling restart.
Mar 22 11:44:48 DietPi systemd[1]: jackett.service: Scheduled restart job, restart counter is at 3.
Mar 22 11:44:48 DietPi systemd[1]: Stopped Jackett (DietPi).
Mar 22 11:44:48 DietPi systemd[1]: Started Jackett (DietPi).
Mar 22 11:44:49 DietPi jackett[879]: /bin/dash: 1: /opt/jackett/jackett: not found
Mar 22 11:44:49 DietPi systemd[1]: jackett.service: Succeeded.
Mar 22 11:44:54 DietPi systemd[1]: jackett.service: Service RestartSec=5s expired, scheduling restart.
Mar 22 11:44:54 DietPi systemd[1]: jackett.service: Scheduled restart job, restart counter is at 4.
--More--
````

MichaIng, 2020-03-22:
@Dr0bac 
The binary is somehow missing, probably the internal updater failed?
```
systemctl stop jackett
ls -Al /opt/jackett
```

Dr0bac, 2020-03-22:
````
root@DietPi:~# systemctl stop jackett
root@DietPi:~# ls -Al /opt/jackett
total 105536
-rw-r--r-- 1 jackett jackett   829952 Mar 16 23:35 AngleSharp.dll
-rw-r--r-- 1 jackett jackett   248832 Mar 16 23:35 Autofac.dll
-rw-r--r-- 1 jackett jackett    15872 Mar 16 23:35 Autofac.Extensions.DependencyInjection.dll
-rw-r--r-- 1 jackett jackett   296960 Mar 16 23:35 AutoMapper.dll
-rw-r--r-- 1 jackett jackett    74240 Mar 16 23:35 BencodeNET.dll
-rw-r--r-- 1 jackett jackett    77312 Mar 16 23:35 CloudflareSolverRe.dll
-rw-r--r-- 1 jackett jackett   204800 Mar 16 23:35 CommandLine.dll
drwxr-xr-x 7 jackett jackett     4096 Feb 13 07:38 Content
-rw-r--r-- 1 jackett jackett    95284 Mar 16 23:35 createdump
-rw-r--r-- 1 jackett jackett    14336 Mar 16 23:35 DateTimeRoutines.dll
-rw-r--r-- 1 jackett jackett     2620 Mar 16 23:35 DateTimeRoutines.pdb
drwxr-xr-x 2 jackett jackett    20480 Mar 15 03:03 Definitions
-rw-r--r-- 1 jackett jackett   198144 Mar 15 03:02 ICSharpCode.SharpZipLib.dll
-rwxr--r-- 1 jackett jackett     3318 Mar 15 03:02 install_service_systemd.sh
drwxr-xr-x 4 jackett jackett     4096 Mar 11 19:04 Jackett
-rw-r--r-- 1 jackett jackett  1189888 Mar 15 03:02 Jackett.Common.dll
-rw-r--r-- 1 jackett jackett   203571 Mar 15 03:02 jackett.deps.json
-rw-r--r-- 1 jackett jackett   394752 Mar 15 03:02 jackett.dll
-rwxr--r-- 1 jackett jackett      406 Mar 15 03:02 jackett_launcher.sh
-rw-r--r-- 1 jackett jackett      328 Mar 15 03:02 jackett.runtimeconfig.json
-rwxr--r-- 1 jackett jackett    77496 Mar 15 03:02 JackettUpdater
-rw-r--r-- 1 jackett jackett    57103 Mar 15 03:02 JackettUpdater.deps.json
-rw-r--r-- 1 jackett jackett   329216 Mar 15 03:02 JackettUpdater.dll
-rw-r--r-- 1 jackett jackett      240 Mar 15 03:02 JackettUpdater.runtimeconfig.json
-rw-r--r-- 1 jackett jackett  2159004 Mar 15 03:02 libclrjit.so
-rw-r--r-- 1 jackett jackett  7297744 Mar 15 03:02 libcoreclr.so
-rw-r--r-- 1 jackett jackett   604268 Mar 15 03:02 libcoreclrtraceptprovider.so
-rw-r--r-- 1 jackett jackett   729572 Mar 15 03:02 libdbgshim.so
-rw-r--r-- 1 jackett jackett   314296 Mar 15 03:02 libhostfxr.so
-rw-r--r-- 1 jackett jackett   281572 Mar 15 03:02 libhostpolicy.so
-rw-r--r-- 1 jackett jackett   183240 Mar 15 03:02 libMonoPosixHelper.so
-rw-r--r-- 1 jackett jackett  2763952 Mar 15 03:02 libmscordaccore.so
-rw-r--r-- 1 jackett jackett  2191888 Mar 15 03:02 libmscordbi.so
-rw-r--r-- 1 jackett jackett    18047 Mar 15 03:02 LICENSE
-rw-r--r-- 1 jackett jackett   111488 Mar 15 03:02 Microsoft.AspNetCore.Antiforgery.dll
-rw-r--r-- 1 jackett jackett    62336 Mar 15 03:02 Microsoft.AspNetCore.Authentication.Abstractions.dll
-rw-r--r-- 1 jackett jackett   111488 Mar 15 03:02 Microsoft.AspNetCore.Authentication.Cookies.dll
-rw-r--r-- 1 jackett jackett    86904 Mar 15 03:02 Microsoft.AspNetCore.Authentication.Core.dll
-rw-r--r-- 1 jackett jackett   152448 Mar 15 03:02 Microsoft.AspNetCore.Authentication.dll
-rw-r--r-- 1 jackett jackett   107392 Mar 15 03:02 Microsoft.AspNetCore.Authentication.OAuth.dll
-rw-r--r-- 1 jackett jackett    91000 Mar 15 03:02 Microsoft.AspNetCore.Authorization.dll
-rw-r--r-- 1 jackett jackett    62336 Mar 15 03:02 Microsoft.AspNetCore.Authorization.Policy.dll
-rw-r--r-- 1 jackett jackett    54144 Mar 15 03:02 Microsoft.AspNetCore.Components.Authorization.dll
-rw-r--r-- 1 jackett jackett   394104 Mar 15 03:02 Microsoft.AspNetCore.Components.dll
-rw-r--r-- 1 jackett jackett    66432 Mar 15 03:02 Microsoft.AspNetCore.Components.Forms.dll
-rw-r--r-- 1 jackett jackett   787328 Mar 15 03:02 Microsoft.AspNetCore.Components.Server.dll
-rw-r--r-- 1 jackett jackett   119680 Mar 15 03:02 Microsoft.AspNetCore.Components.Web.dll
-rw-r--r-- 1 jackett jackett    70528 Mar 15 03:02 Microsoft.AspNetCore.Connections.Abstractions.dll
-rw-r--r-- 1 jackett jackett    54144 Mar 15 03:02 Microsoft.AspNetCore.CookiePolicy.dll
-rw-r--r-- 1 jackett jackett    91008 Mar 15 03:02 Microsoft.AspNetCore.Cors.dll
-rw-r--r-- 1 jackett jackett    78720 Mar 15 03:02 Microsoft.AspNetCore.Cryptography.Internal.dll
-rw-r--r-- 1 jackett jackett    45952 Mar 15 03:02 Microsoft.AspNetCore.Cryptography.KeyDerivation.dll
-rw-r--r-- 1 jackett jackett    50048 Mar 15 03:02 Microsoft.AspNetCore.DataProtection.Abstractions.dll
-rw-r--r-- 1 jackett jackett   406400 Mar 15 03:02 Microsoft.AspNetCore.DataProtection.dll
-rw-r--r-- 1 jackett jackett    50048 Mar 15 03:02 Microsoft.AspNetCore.DataProtection.Extensions.dll
-rw-r--r-- 1 jackett jackett    33664 Mar 15 03:02 Microsoft.AspNetCore.Diagnostics.Abstractions.dll
-rw-r--r-- 1 jackett jackett   422784 Mar 15 03:02 Microsoft.AspNetCore.Diagnostics.dll
-rw-r--r-- 1 jackett jackett    54144 Mar 15 03:02 Microsoft.AspNetCore.Diagnostics.HealthChecks.dll
-rw-r--r-- 1 jackett jackett    45952 Mar 15 03:02 Microsoft.AspNetCore.dll
-rw-r--r-- 1 jackett jackett    45952 Mar 15 03:02 Microsoft.AspNetCore.HostFiltering.dll
-rw-r--r-- 1 jackett jackett    45952 Mar 15 03:02 Microsoft.AspNetCore.Hosting.Abstractions.dll
-rw-r--r-- 1 jackett jackett   377728 Mar 15 03:02 Microsoft.AspNetCore.Hosting.dll
-rw-r--r-- 1 jackett jackett    29568 Mar 15 03:02 Microsoft.AspNetCore.Hosting.Server.Abstractions.dll
-rw-r--r-- 1 jackett jackett    41856 Mar 15 03:02 Microsoft.AspNetCore.Html.Abstractions.dll
-rw-r--r-- 1 jackett jackett   217984 Mar 15 03:02 Microsoft.AspNetCore.Http.Abstractions.dll
-rw-r--r-- 1 jackett jackett    58240 Mar 15 03:02 Microsoft.AspNetCore.Http.Connections.Common.dll
[... cut ...]
-rw-r--r-- 1 jackett jackett    25472 Mar 15 03:02 System.Xml.XmlDocument.dll
-rw-r--r-- 1 jackett jackett    25464 Mar 15 03:02 System.Xml.XmlSerializer.dll
-rw-r--r-- 1 jackett jackett    25472 Mar 15 03:02 System.Xml.XPath.dll
-rw-r--r-- 1 jackett jackett    29560 Mar 15 03:02 System.Xml.XPath.XDocument.dll
-rw-r--r-- 1 jackett jackett      460 Mar 15 03:02 web.config
-rw-r--r-- 1 jackett jackett    25472 Mar 15 03:02 WindowsBase.dll
-rw-r--r-- 1 jackett jackett   199168 Mar 15 03:02 YamlDotNet.dll
````

MichaIng, 2020-03-22:
@Dr0bac 
Yes the binary is missing, everything else is there. I compared the content to the online archive...
Please try to reinstall Jackett:
```
dietpi-software reinstall 147
```

Dr0bac, 2020-03-22:
after reinstall jacket is running again, thanks.

Dr0bac, 2020-04-20:
I noticed again that jackett service is in activating status.
```
● jackett.service - Jackett (DietPi)
   Loaded: loaded (/etc/systemd/system/jackett.service; disabled; vendor preset: enabled)
   Active: activating (auto-restart) since Mon 2020-04-20 02:29:52 CEST; 4s ago
  Process: 1227 ExecStart=/bin/dash -c /opt/jackett/jackett --NoRestart; while pgrep -u jackett JackettUpdater; do sleep 1; done (code=exited, status=0/SUCCESS)
 Main PID: 1227 (code=exited, status=0/SUCCESS)
```
status after i stop jackett service with `systemctl stop jackett`
```
● jackett.service - Jackett (DietPi)
   Loaded: loaded (/etc/systemd/system/jackett.service; disabled; vendor preset: enabled)
   Active: inactive (dead)

Apr 20 02:34:04 DietPi systemd[1]: Started Jackett (DietPi).
Apr 20 02:34:05 DietPi jackett[1557]: /bin/dash: 1: /opt/jackett/jackett: not found
Apr 20 02:34:05 DietPi systemd[1]: jackett.service: Succeeded.
Apr 20 02:34:10 DietPi systemd[1]: jackett.service: Service RestartSec=5s expired, scheduling restart.
Apr 20 02:34:10 DietPi systemd[1]: jackett.service: Scheduled restart job, restart counter is at 78.
Apr 20 02:34:10 DietPi systemd[1]: Stopped Jackett (DietPi).
Apr 20 02:34:10 DietPi systemd[1]: Started Jackett (DietPi).
Apr 20 02:34:10 DietPi jackett[1562]: /bin/dash: 1: /opt/jackett/jackett: not found
Apr 20 02:34:10 DietPi systemd[1]: jackett.service: Succeeded.
Apr 20 02:34:14 DietPi systemd[1]: Stopped Jackett (DietPi).
```
and here is jackett directory
```
total 105560
-rw-r--r-- 1 jackett jackett   829952 Apr  9 06:35 AngleSharp.dll
-rw-r--r-- 1 jackett jackett   248832 Apr  9 06:36 Autofac.dll
-rw-r--r-- 1 jackett jackett    15872 Apr  9 06:35 Autofac.Extensions.DependencyInjection.dll
-rw-r--r-- 1 jackett jackett   296960 Apr  9 06:36 AutoMapper.dll
-rw-r--r-- 1 jackett jackett    74752 Apr  9 06:35 BencodeNET.dll
-rw-r--r-- 1 jackett jackett    77312 Apr  9 06:35 CloudflareSolverRe.dll
-rw-r--r-- 1 jackett jackett   204800 Apr  9 06:36 CommandLine.dll
drwxr-xr-x 7 jackett jackett     4096 Mar 22 07:55 Content
-rw-r--r-- 1 jackett jackett    95288 Apr  9 06:36 createdump
-rw-r--r-- 1 jackett jackett    14336 Apr  9 06:35 DateTimeRoutines.dll
-rw-r--r-- 1 jackett jackett     2620 Apr  9 06:36 DateTimeRoutines.pdb
drwxr-xr-x 2 jackett jackett    20480 Apr  8 05:35 Definitions
-rw-r--r-- 1 jackett jackett   198144 Apr  8 05:34 ICSharpCode.SharpZipLib.dll
-rwxr--r-- 1 jackett jackett     3318 Apr  8 05:34 install_service_systemd.sh
drwxr-xr-x 4 jackett jackett     4096 Apr  8 05:35 Jackett
-rw-r--r-- 1 jackett jackett  1184256 Apr  8 05:34 Jackett.Common.dll
-rw-r--r-- 1 jackett jackett   211087 Apr  8 05:34 jackett.deps.json
-rw-r--r-- 1 jackett jackett   394752 Apr  8 05:34 jackett.dll
-rwxr--r-- 1 jackett jackett      406 Apr  8 05:34 jackett_launcher.sh
-rw-r--r-- 1 jackett jackett      328 Apr  8 05:34 jackett.runtimeconfig.json
-rwxr--r-- 1 jackett jackett    77496 Apr  8 05:34 JackettUpdater
-rw-r--r-- 1 jackett jackett    61316 Apr  8 05:34 JackettUpdater.deps.json
-rw-r--r-- 1 jackett jackett   329728 Apr  8 05:34 JackettUpdater.dll
-rw-r--r-- 1 jackett jackett      240 Apr  8 05:34 JackettUpdater.runtimeconfig.json
-rw-r--r-- 1 jackett jackett  2159052 Apr  8 05:34 libclrjit.so
-rw-r--r-- 1 jackett jackett  7298036 Apr  8 05:34 libcoreclr.so
-rw-r--r-- 1 jackett jackett   604268 Apr  8 05:34 libcoreclrtraceptprovider.so
-rw-r--r-- 1 jackett jackett   729572 Apr  8 05:34 libdbgshim.so
-rw-r--r-- 1 jackett jackett   314296 Apr  8 05:34 libhostfxr.so
-rw-r--r-- 1 jackett jackett   281572 Apr  8 05:34 libhostpolicy.so
-rw-r--r-- 1 jackett jackett   183240 Apr  8 05:34 libMonoPosixHelper.so
-rw-r--r-- 1 jackett jackett  2763956 Apr  8 05:34 libmscordaccore.so
-rw-r--r-- 1 jackett jackett  2191888 Apr  8 05:35 libmscordbi.so
-rw-r--r-- 1 jackett jackett    18047 Apr  8 05:34 LICENSE
-rw-r--r-- 1 jackett jackett   111696 Apr  8 05:35 Microsoft.AspNetCore.Antiforgery.dll
-rw-r--r-- 1 jackett jackett    62544 Apr  8 05:34 Microsoft.AspNetCore.Authentication.Abstractions.dll
-rw-r--r-- 1 jackett jackett   111488 Apr  8 05:34 Microsoft.AspNetCore.Authentication.Cookies.dll
-rw-r--r-- 1 jackett jackett    87120 Apr  8 05:34 Microsoft.AspNetCore.Authentication.Core.dll
-rw-r--r-- 1 jackett jackett   152680 Apr  8 05:34 Microsoft.AspNetCore.Authentication.dll
-rw-r--r-- 1 jackett jackett   107600 Apr  8 05:34 Microsoft.AspNetCore.Authentication.OAuth.dll
-rw-r--r-- 1 jackett jackett    91216 Apr  8 05:34 Microsoft.AspNetCore.Authorization.dll
-rw-r--r-- 1 jackett jackett    62544 Apr  8 05:34 Microsoft.AspNetCore.Authorization.Policy.dll
[... 291 lines cut ...]
-rw-r--r-- 1 jackett jackett    25472 Apr  8 05:35 System.Threading.Tasks.Extensions.dll
-rw-r--r-- 1 jackett jackett   131968 Apr  8 05:34 System.Threading.Tasks.Parallel.dll
-rw-r--r-- 1 jackett jackett    29560 Apr  8 05:34 System.Threading.Thread.dll
-rw-r--r-- 1 jackett jackett    25472 Apr  8 05:34 System.Threading.ThreadPool.dll
-rw-r--r-- 1 jackett jackett    25472 Apr  8 05:35 System.Threading.Timer.dll
-rw-r--r-- 1 jackett jackett    25472 Apr  8 05:34 System.Transactions.dll
-rw-r--r-- 1 jackett jackett   394104 Apr  8 05:34 System.Transactions.Local.dll
-rw-r--r-- 1 jackett jackett    25472 Apr  8 05:34 System.ValueTuple.dll
-rw-r--r-- 1 jackett jackett    25472 Apr  8 05:34 System.Web.dll
-rw-r--r-- 1 jackett jackett    58240 Apr  8 05:34 System.Web.HttpUtility.dll
[... cut ...]
-rw-r--r-- 1 jackett jackett    25472 Apr  8 05:34 System.Xml.XmlDocument.dll
-rw-r--r-- 1 jackett jackett    25680 Apr  8 05:34 System.Xml.XmlSerializer.dll
-rw-r--r-- 1 jackett jackett    25472 Apr  8 05:34 System.Xml.XPath.dll
-rw-r--r-- 1 jackett jackett    29568 Apr  8 05:34 System.Xml.XPath.XDocument.dll
-rw-r--r-- 1 jackett jackett      460 Apr  8 05:34 web.config
-rw-r--r-- 1 jackett jackett    25472 Apr  8 05:34 WindowsBase.dll
-rw-r--r-- 1 jackett jackett   199168 Apr  8 05:34 YamlDotNet.dll
```

Joulinar, 2020-04-20:
Looks like you have same error message again 

`/opt/jackett/jackett: not found`

Probably you would need to perform re-installation again 

`dietpi-software reinstall 147`

MichaIng, 2020-04-21:
Strange, probably the internal updater fails somehow. Although files are from April 8, hence no recent update has been installed. I'll spin up a Jackett instance for testing when I find time. Also we should check if the devs found a real solution for systemd unit + internal updater that does not require this pgrep + sleep loop.

Dr0bac, 2020-04-21:
I reinstalled jackett a month ago when you told me, and didn't use it, maybe once or twice, maybe it was broken for some time, i didn't notice it till now..

MichaIng, 2020-04-21:
@Dr0bac 
Possible. I checked recent upstream changes and the increased the stop timeout: https://github.com/Jackett/Jackett/pull/7614/files
I cannot find any explaination for this, but it could be at least related to the updater. I'll do the same in our systemd unit, you could change it as well and see if the issue stays resolved or still reappears.

closed 2020-07-09 by MichaIng