MichaIng/DietPi #3431: Rpi Cam interface not working, possibly permission error

arkhub, 2020-03-22:
### Creating a bug report/issue

#### Required Information
- DietPi version 
> G_DIETPI_VERSION_CORE=6
> G_DIETPI_VERSION_SUB=28
> G_DIETPI_VERSION_RC=0
> G_GITBRANCH='master'
> G_GITOWNER='MichaIng'
- Distro version
> 10.3
- Kernel version
> Linux DietPi 4.19.97+ #1294 Thu Jan 30 13:10:54 GMT 2020 armv6l GNU/Linux
- SBC device
> RPi B (armv6l) (Raspberry Pi 1 Model B)
- Power supply used
> 5V 2.5A
- SDcard used 
> Generic 8Gb microsd card with adapter

#### Additional Information (if applicable)
- Software title
> Rpi Cam Control
- Was the software title installed freshly or updated/migrated?
> Fresh Install
- Can this issue be replicated on a fresh installation of DietPi?
> Yes, I've installed it again fresh

#### Steps to reproduce
1. Fresh install dietpi using etcher
2. Finish setup and reboot
3. Install optimized software -> Rpi Cam
4. Go to web interface -> http://<ip address/rpicam
5. Page loaded, pictures shown, but cannot change / edit any option, the interface just doesn't work

#### Expected behaviour

- rpi cam control interface should be working (change setting, etc)

#### Actual behaviour

- rpi cam control interface not working, even stopping the camera button doesn't work

#### Extra details

After reviewing error.log in lighttpd, here is the first error encountered
(mod_fastcgi.c.421) FastCGI-stderr: PHP message: PHP Warning:  fopen(/var/www/rpicam/scheduleLog.txt): failed to open stream: Permission denied in /var/www/rpicam/config.php on line 66PHP message: PHP Warning:  fwrite() expects parameter 1 to be resource, bool given in /var/www/rpicam/config.php on line 68PHP message: PHP Warning:  fclose() expects parameter 1 to be resource, bool given in /var/www/rpicam/config.php on line 69PHP message: PHP Warning:  fopen(/var/www/rpicam/scheduleLog.txt): failed to open stream: Permission denied in /var/www/rpicam/config.php on line 66PHP message: PHP Warning:  fwrite() expects parameter 1 to be resource, bool given in /var/www/rpicam/config.php on line 68PHP message: PHP Warning:  fclose() expects parameter 1 to be resource, bool given in /var/www/rpicam/config.php on line 69

arkhub, 2020-03-22:
Update : changing the permission of rpicam folder into www-data seems to fix the issue (chown -hR www-data:www-data), although I don't know if this the recommended way.

MichaIng, 2020-03-22:
@arkhub 
Permissions should be granted more targeted:
```
chown -R www-data:www-data /var/www/rpicam
```

MichaIng, 2020-03-22:
Fixed: https://github.com/MichaIng/DietPi/commit/503549e4854eaa37653fc77a65cf4efd1c06d8f0
Changelog: https://github.com/MichaIng/DietPi/commit/2e41b6f543cf28ca855b47c0102d90f408834af0

dsalgueiro, 2020-03-28:
As-is installation with ProFTP and RPI Cam Web Interface: Cannot view RPI Cam Web Interface page.

`With chown -R www-data:www-data /var/www/rpicam`

RPI Cam Web Interface now shows up and works

STILL NOT WORKING: 404 error when trying to download any recorded videos or Gzipped videos.  Note that preview images do appear.  I think this is another related permissions issue.

MichaIng, 2020-03-28:
@dsalgueiro 
Could you please paste:
```
ls -al /mnt/dietpi_userdata/rpicam
```
Omit any actual file names in case for your privacy, I am only interested in the user/modes of the contained files. The videos should be stored there and need to be readable by www-data user.

dsalgueiro, 2020-03-28:
Here's what I get with that:

```
root@DietPi:~# ls -al /mnt/dietpi_userdata/rpicam
total 62916
drwxrwxr-x 2 www-data www-data     4096 Mar 28 15:49 .
drwxrwxr-x 7 dietpi   dietpi       4096 Mar 27 12:57 ..
-rw-r--r-- 1 www-data www-data 14876171 Mar 28 15:49 vi_0000[rest of file name]

All of the other file names have the same owner (www-data) and permissions.
```

MichaIng, 2020-03-28:
@dsalgueiro 
Hmm the permissions look fine as `www-data` is owner of all files and has write access as such. Also strange that it shows one video only, as all present ones have exactly the same permissions. But the shown one is one of those listed in the directory, right?

dsalgueiro, 2020-03-28:
> @dsalgueiro
> Hmm the permissions look fine as `www-data` is owner of all files and has write access as such. Also strange that it shows one video only, as all present ones have exactly the same permissions. But the shown one is one of those listed in the directory, right?

Correct, the only video shown is in the top 2 files, first one is the video, second one is a thumbnail.

closed 2020-05-03 by MichaIng