MichaIng/DietPi #3433: hi,

drieksje, 2020-03-22:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

MichaIng, 2020-03-22:
?

drieksje, 2020-03-22:
sorry,
i have an issue after installing  java for chronium on my 3.16.31 dietpi,(home again) so everything is running verrywell but when i did want to sync. chromium on lxde they did ask to update java for security reasons, so i did but it was not working!!!!, so i did install firfox er , firefox worked fine but now i can't install no dietpi-software annymore, so i think the ubuntu ppa's are the problem!!(unregistered ppa's, i did want to uninstall them but it didden work .

#### Details:
- Date           | Sun 22 Mar 17:12:03 CET 2020
- Bug report     | N/A
- DietPi version | v6.28.0 (Fourdee/master)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- SBC device     | Odroid C2 (aarch64) (ID=12)
- Kernel version | Linux DietPi 3.16.81+ #1 SMP PREEMPT Mon Jan 20 13:45:12 CET 2020 aarch64 GNU/Linux
- Distro         | stretch (ID=4)
- Command        | G_AGUP
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Ign:11 http://ftp.debian.org/debian/stretch main/contrib all Packages
Ign:12 http://ftp.debian.org/debian/stretch main/contrib arm64 Packages
Ign:14 http://ftp.debian.org/debian/stretch main/non-free arm64 Packages
Ign:10 http://ftp.debian.org/debian/stretch main/non-free all Packages
Ign:15 https://deb.debian.org/debian stretch InRelease
Ign:11 http://ftp.debian.org/debian/stretch main/contrib all Packages
Get:16 https://deb.debian.org/debian stretch-updates InRelease [91.0 kB]
Get:17 https://packages.sury.org/php stretch InRelease [6,760 B]
Ign:12 http://ftp.debian.org/debian/stretch main/contrib arm64 Packages
Ign:14 http://ftp.debian.org/debian/stretch main/non-free arm64 Packages
Ign:10 http://ftp.debian.org/debian/stretch main/non-free all Packages
Get:18 https://deb.debian.org/debian-security stretch/updates InRelease [94.3 kB]
Ign:11 http://ftp.debian.org/debian/stretch main/contrib all Packages
Get:19 https://deb.debian.org/debian stretch-backports InRelease [91.8 kB]
Ign:12 http://ftp.debian.org/debian/stretch main/contrib arm64 Packages
Ign:14 http://ftp.debian.org/debian/stretch main/non-free arm64 Packages
Get:20 https://deb.debian.org/debian bullseye InRelease [116 kB]
Ign:10 http://ftp.debian.org/debian/stretch main/non-free all Packages
Ign:11 http://ftp.debian.org/debian/stretch main/contrib all Packages
Hit:21 https://deb.debian.org/debian stretch Release
Ign:12 http://ftp.debian.org/debian/stretch main/contrib arm64 Packages
Ign:14 http://ftp.debian.org/debian/stretch main/non-free arm64 Packages
Ign:10 http://ftp.debian.org/debian/stretch main/non-free all Packages
Ign:11 http://ftp.debian.org/debian/stretch main/contrib all Packages
Err:12 http://ftp.debian.org/debian/stretch main/contrib arm64 Packages
  404  Not Found
Ign:14 http://ftp.debian.org/debian/stretch main/non-free arm64 Packages
Get:22 http://deb.debian.org/debian unstable/main arm64 Packages.diff/Index [27.9 kB]
Get:23 http://deb.debian.org/debian unstable/main arm64 Packages 2020-03-22-0810.58.pdiff [12.7 kB]
Get:24 http://deb.debian.org/debian unstable/main arm64 Packages 2020-03-22-1415.01.pdiff [24.2 kB]
Get:24 http://deb.debian.org/debian unstable/main arm64 Packages 2020-03-22-1415.01.pdiff [24.2 kB]
Ign:5 http://ppa.launchpad.net/mozillateam/firefox-next/ubuntu focal InRelease
Ign:9 http://ppa.launchpad.net/webupd8team/java/ubuntu precise InRelease
Ign:13 http://ppa.launchpad.net/webupd8team/java/ubuntu trusty InRelease
Get:25 https://packages.sury.org/php stretch/main arm64 Packages [133 kB]
Get:26 https://deb.debian.org/debian bullseye/main arm64 Packages.diff/Index [27.9 kB]
Get:27 https://deb.debian.org/debian bullseye/main arm64 Packages 2020-03-22-0810.58.pdiff [4,661 B]
Get:28 https://deb.debian.org/debian bullseye/main arm64 Packages 2020-03-22-1415.01.pdiff [60 B]
Get:28 https://deb.debian.org/debian bullseye/main arm64 Packages 2020-03-22-1415.01.pdiff [60 B]
Fetched 918 kB in 7s (128 kB/s)
Reading package lists...
W: The repository 'http://ftp.debian.org/debian/stretch main Release' does not have a Release file.
W: GPG error: http://ppa.launchpad.net/mozillateam/firefox-next/ubuntu focal InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 9BDB3D89CE49EC21
W: The repository 'http://ppa.launchpad.net/mozillateam/firefox-next/ubuntu focal InRelease' is not signed.
W: GPG error: http://ppa.launchpad.net/webupd8team/java/ubuntu precise InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY C2518248EEA14886
W: The repository 'http://ppa.launchpad.net/webupd8team/java/ubuntu precise InRelease' is not signed.
W: GPG error: http://ppa.launchpad.net/webupd8team/java/ubuntu trusty InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY C2518248EEA14886
W: The repository 'http://ppa.launchpad.net/webupd8team/java/ubuntu trusty InRelease' is not signed.
E: Failed to fetch http://ftp.debian.org/debian/stretch/dists/main/contrib/binary-arm64/Packages  404  Not Found
E: Some index files failed to download. They have been ignored, or old ones used instead.

MichaIng, 2020-03-22:
@drieksje 
Generally, be sure that this PPA is compatible with Debian. We used it in the past but removed it because it caused issues. However, you didn't add its key, but they are not the issue. You should be able to add the key like this:
```
apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 7B2C3B0889BF5709A105D03AC2518248EEA14886
```

Did you add the Debian repo another time? There are wrong URLs:
```
http://ftp.debian.org/debian/stretch main/contrib
```
"main/contrib" does not exist, there is either "main" only or "contrib" only: http://ftp.debian.org/debian/dists/stretch/
In case reset the main sources.list:
```
/DietPi/dietpi/func/dietpi-set_software apt-mirror default
```

drieksje, 2020-03-22:
Hi,
After reboot evertring is working fijne, had been reading 10 things you do
when you install Debian 9, but 10 things you don't have to do would have
been a better choice 😊😊😊, thé one thing that keeps boddering me is that
Synaptic isen't working in dietpi, so thanks for all your good advice
👍👍👍👍👍👍

MichaIng, 2020-03-22:
@drieksje 
Great.

> thé one thing that keeps boddering me is that
Synaptic isen't working in dietpi, so thanks for all your good advice

I'll try it by times.

drieksje, 2020-03-22:
You may close the issue. I do like the Linux, Debian distro's it keeps me
bussy in these homestaying days, still have a lot to learn🤔🤔🤔🤔🤔🤔🤔
Kind regards: Dirk

closed 2020-03-22 by MichaIng