MichaIng/DietPi #3436: Nextcloud: ncc doesn't work anymore (error message: This version of Nextcloud is not compatible with > PHP 7.3.)

Krawei, 2020-03-24:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`:
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=28
G_DIETPI_VERSION_RC=0
G_GITBRANCH='beta'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`:
9.11
- Kernel version | `uname -a`:
 Linux Himbeere 4.19.108-v7+ #1298 SMP Fri Mar 6 18:04:35 GMT 2020 armv7l GNU/Linux
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3):
 RPi 3 Model B (armv7l)
- Power supply used | (EG: 5V 1A RAVpower): 
5V 3A
- SDcard used | (EG: SanDisk ultra):
SanDisk ultra

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud):
Nextcloud
- Was the software title installed freshly or updated/migrated?:
migrated
- Can this issue be replicated on a fresh installation of DietPi?:
yes

#### Steps to reproduce
ncc help

#### Expected behaviour
ncc showing its possible commands

#### Actual behaviour
root@Himbeere:~# ncc help
This version of Nextcloud is not compatible with > PHP 7.3.<br/>You are currently running 7.4.4.root@Himbeere:~# 
- ...

#### Extra details
I've migrated from the Debian 8 to the debian 9 Image some months ago. it seems that there are both PHP7.3 and PHP7.4 installed on my system.

I've read that Nextcloud 18 works with PHP7.4 (see: https://github.com/nextcloud/server/issues/18146). So my question is if the installed nextcloud version is outdated? If yes, please upgrade it to the new major version. If this is not possible please explain how to update nextclpud myself? I didn't found it via apt so it is kinda flatpack/snap-software package, am I right? how I update this?

I hope you can shed some light in the dark :).

KInd Regards,

Krawei.

Joulinar, 2020-03-24:
Hi,

many thanks for your report. Well you can check inside NextCloud application if there are updates available and perform the update from there.

as well you can have a look to the NextCloud Doc on how to upgrade

https://docs.nextcloud.com/server/latest/admin_manual/maintenance/upgrade.html

MichaIng, 2020-03-24:
@Krawei 
Basically it is a known issue. I guess you have phpMyAdmin installed?

Krawei, 2020-03-24:
Hello @Joulinar,

unfortunately  I forgot thre login credentials for the webinterface and wanted to add a new admin user via ncc.

So what about the nextcloud software package? Is it a flatpack/snapd-one? Are there other methods to update it? Or will you update the dietpi-nextcloud-package to v18?

Krawei, 2020-03-24:
> @Krawei
> Basically it is a known issue. I guess you have phpMyAdmin installed?

Well, I don't think so atm....I'll check that.

MichaIng, 2020-03-24:
@Krawei 
Nextcloud is up-to-date, it's upstream version does not yet support PHP7.4, neither is PHP7.4 intended to be installed on DietPi since there are a lot of other web applications which do not yet support it.

Usually PHP7.4 is pulled on DietPi Stretch systems since Ondrejs PHP repo has it made the new default PHP version some moths ago. So if any APT package pulls the PHP meta packages (instead of an explicit PHP version, or the [PHP API](https://packages.debian.org/bullseye/phpapi-20180731) which every package should do!), then PHP7.4 is installed.

phpMyAdmin from Debian repo as an example does this. Since it is anyway implemented instructively and suboptiomal in many other ways, on DietPi we pull the upstream version and implement it according to the official install instructions now. And I see that we actively migrate this on update to v6.27, hence this cannot be the reason anymore.

Okay lets see what would be removed when actively purging the PHP7.4 base package. Please do not blindly confirm when being asked. Other packages (aside of PHP7.4 modules) might depend on it, hence this should be checked carefully first. Instead call the following command, hit `n` in case and paste the output here:
```
apt purge php7.3-common
```

Joulinar, 2020-03-24:
> Or will you update the dietpi-nextcloud-package to v18?

Well there is no DietPi own Nextcloud package. DietPi will pull latest Nectcloud version directly from Nextcloud during installation.

MichaIng, 2020-03-24:
Actually it should have been switched already for MOTD hack 😉.

Ah lol I've overseen that that NC18 already added PHP7.4 support. However I would still not recommend to use it as long as it has not gained wider support. So I would go with the above to purge PHP7.4, if there is not any strong dependency on it.

Joulinar, 2020-03-24:
yeah, I just checked the `dietpi-software` code. So did not take MOTD into account 😉

Krawei, 2020-03-24:
@MichaIng : Well it seems that there's no other software related to php7.4-common on my system, see:

- # apt purge php7.4-common
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.        
Statusinformationen werden eingelesen.... Fertig
Die folgenden Pakete wurden automatisch installiert und werden nicht mehr benötigt:
  libicu64 libonig4
Verwenden Sie »apt autoremove«, um sie zu entfernen.
Die folgenden Pakete werden ENTFERNT:
  php-curl* php-fpm* php-gd* php-mbstring* php-sqlite3* php-xml* php-zip* php7.4-cli* php7.4-common* php7.4-curl* php7.4-fpm* php7.4-gd* php7.4-json* php7.4-mbstring*
  php7.4-opcache* php7.4-readline* php7.4-sqlite3* php7.4-xml* php7.4-zip*
0 aktualisiert, 0 neu installiert, 19 zu entfernen und 0 nicht aktualisiert.
Nach dieser Operation werden 14,3 MB Plattenplatz freigegeben.
Möchten Sie fortfahren? [J/n] n
Abbruch.

I've installed dietpi and nextcloud, lightty atm on my system and baikal in the past. Maybe php7.4 has been installed with the "upgrade"/"switch" to the dietpi Buster Beta-Branch some months ago (is there a way to switch back to the stable branch without reinstalling btw?).

So it should be save to remove it, I guess.

But last but not least: There are plenty of performance gains in PHP7.4, aren't they? So @Joulinar , if I understand you right, dietpi will pull the latest NC-version when I do a re-install, so it should be NC18, am I correct?
If not, will you switch to NC18 soon? There are some nice features being added in that version :).

But I think I will purge PHP7.4 once :).

Joulinar, 2020-03-24:
Well I'm not sure if it's a good idea to simply install NC18 over NC17. This could lead to some damage on your NC installation. I would recommend to follow official instruction of Nextcloud to move from 1 major release to another.

MichaIng, 2020-03-24:
@Krawei 
Yes please always use the internal updater to update Nextcloud, since there are usually required (database) migration steps included.

I would purge PHP7.4 first, upgrade Nextcloud to v18, then you can think of upgrading PHP again. But take care that the `php-apcu` + `php-redis` modules supports PHP7.4, otherwise you will have a large performance decrease (+need to manually edit NC config.txt). The following should provide this info:
```
apt show php-apcu
apt show php-redis
```

And you need to alter `/etc/lighttpd/conf-available/fastcgi-php.conf` (AFAIK, or similar) to use `php7.4-fpm.sock` instead of `php7.4-fpm.sock` AND you need to transfer + enable `/etc/php/7.3/mods-available/dietpi.conf`+`dietpi-nextcloud.conf` over to `/etc/php/7.4/` but that should be it then.

Krawei, 2020-03-24:
Pheeww, thanks for your great support and step-by-step help. :D

MichaIng, 2020-03-24:
@Krawei 
One more step, to enable the two "modules" (more DietPi+Nextcloud specific configs), after PHP7.4 has been installed, the two files copied over and PHP7.3 purged:
```
phpenmod dietpi
phpenmod dietpi-nextcloud
```
I'll also add general support for PHP7.4 to DietPi-Software, in a way that if PHP7.3 is not installed but PHP7.4 is installed, it will install required modules and configure all web applications against PHP7.4.

Krawei, 2020-03-24:
> @Krawei
[...]
> And you need to alter `/etc/lighttpd/conf-available/fastcgi-php.conf` (AFAIK, or similar) to use `php7.4-fpm.sock` instead of `php7.4-fpm.sock` AND you need to transfer + enable `/etc/php/7.3/mods-available/dietpi.conf`+`dietpi-nextcloud.conf` over to `/etc/php/7.4/` but that should be it then.

Do you mean 
And you need to alter `/etc/lighttpd/conf-available/fastcgi-php.conf` (AFAIK, or similar) to use `php7.4-fpm.sock` instead of `php7.3-fpm.sock` AND you need to transfer + enable `/etc/php/7.3/mods-available/dietpi.conf`+`dietpi-nextcloud.conf` over to `/etc/php/7.4/` but that should be it then. 
? ;)

Krawei, 2020-03-24:
@MichaIng well I think I need to do a break....
your suggested packages
php-apcu
php-redis
doesn't seems to support PHP74, see:

- # apt show php-apcu
Package: php-apcu
Version: 5.1.18+4.0.11-1+0~20191219.13+debian9~1.gbpa99079
Priority: optional
Section: php
Maintainer: Debian PHP PECL Maintainers <team+php-pecl@tracker.debian.org>
Installed-Size: 641 kB
Provides: php-user-cache, php5.6-apcu, php7.0-apcu, php7.1-apcu, php7.2-apcu, php7.3-apcu
Pre-Depends: php-common (>= 2:69~)
Depends: phpapi-20180731 | phpapi-20170718 | phpapi-20160303 | phpapi-20151012 | phpapi-20131226, libc6 (>= 2.4)
Recommends: php-apcu-bc
Suggests: php-gd
Conflicts: php-user-cache, php-xcache, php-yac
Replaces: php-user-cache
Homepage: https://pecl.php.net/package/APCu
Download-Size: 83,9 kB
APT-Manual-Installed: yes
APT-Sources: https://packages.sury.org/php stretch/main armhf Packages
Description: APC User Cache for PHP
 The APCu is userland caching: APC (Alternative PHP Cache) stripped of
 opcode caching after the deployment of Zend OpCache in PHP 5.5 as the
 primary solution to opcode caching in future versions of PHP.
 .
 The APCu is a fast solution for userland caching (and dumping) of PHP
 variables locally, it is not distributed like MemcacheD, but they can
 be used together for optimal caching.

# apt show php-redis
Package: php-redis
Version: 5.2.1+4.3.0-1+0~20200321.20+debian9~1.gbp9fe0ce
Priority: optional
Section: php
Maintainer: Debian PHP PECL Maintainers <team+php-pecl@tracker.debian.org>
Installed-Size: 1.808 kB
Provides: php5.6-redis, php7.0-redis, php7.1-redis, php7.2-redis, php7.3-redis
Pre-Depends: php-common (>= 2:69~)
Depends: php-igbinary, phpapi-20180731 | phpapi-20170718 | phpapi-20160303 | phpapi-20151012 | phpapi-20131226, libc6 (>= 2.4)
Suggests: redis-server
Homepage: https://pecl.php.net/package/redis
Download-Size: 328 kB
APT-Manual-Installed: yes
APT-Sources: https://packages.sury.org/php stretch/main armhf Packages
Description: PHP extension for interfacing with Redis
 This extension allows php applications to communicate with the Redis
 persistent key-value store.  The php-redis module provides an easy
 object oriented interface.

Well, at least: Thanks for your help and hints. Seems I need to lean back and wait for the updated apcu and redis-packages ;).

MichaIng, 2020-03-25:
@Krawei 
I just checked the PHP repo Stretch suite and indeed those do not support PHP7.4 so far. In this case I would stay with PHP7.3 as caching and Redis-based file locking are a huge performance bonus which PHP7.4-only can never outweigh.

MichaIng, 2020-10-18:
I mark this as closed. Feel free to reopen if required.

closed 2020-10-18 by MichaIng

gstrauss, 2020-10-25:
With PHP 7.4, you should not need to bake the version in to the path to PHP-FPM.

Proposed patch to Debian lighttpd config to add "lighty-enable-mod fastcgi-php-fpm option":
https://salsa.debian.org/debian/lighttpd/-/merge_requests/19

MichaIng, 2020-10-25:
Jep, however PHP7.4 is not used in any stable Debian release and using an explicit version assures that the exact intended PHP version is used, e.g. if multiple PHP versions are installed. Good to know about the purposed PHP-FPM module for Lighttpd 😃, at least for Debian Bullseye we could be able to skip our `fastcgi-php` override. Also good to see that the purposed module matches our override 100%, so we obviously do well.