MichaIng/DietPi #3439: Cannot use dietpi-software or apt-get

teamallnighter, 2020-03-25:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. ...executing dietpi-software
2. ...executing apt-get upgrade

#### Expected behaviour

- ...I should be able to download and install or uninstall software

#### Actual behaviour

- ...Failed network connection test

#### Extra details

- ...This happens even though I am connect, can access the internet and access the pi from outside the network

Joulinar, 2020-03-25:
Hi,

Pls can you provide the exact error message you are facing.

teamallnighter, 2020-03-25:
yes of course. When i go to "dietpi-software i get 

"cking network connectivity

Command: ping -c 1 -W 5 1.1.1.1
Exit code: 1
DietPi version: v6.28.0 (MichaIng/master) | HW_MODEL:4 | HW_ARCH:2 | DISTRO:5
Image creator: DietPi Core Team
Pre-image: Raspbian Lite

 file content:
G 1.1.1.1 (1.1.1.1) 56(84) bytes of data.

 1.1.1.1 ping statistics ---
ackets transmitted, 0 received, 100% packet loss, time 0ms

       Retry         : Re-run the last command that failed               
       DietPi-Config : Edit network, APT/NTP mirror settings etc
       Send report   : Uploads bugreport containing system info to DietPi

                    <Ok>                            <Exit>

"

Joulinar, 2020-03-25:
@teamallnighter 

if you are not able to `ping 1.1.1.1` you would have local issues as it's part of Cloudflares DNS Server. It is the globally available IP address for Cloudflares 1.1.1.1 DNS resolver:

https://www.dnswatch.info/dns/dnslookup?la=en&host=1.1.1.1&submit=Resolve

Anyway if you like to change this, you can do it as follow. 

Do `nano /DietPi/dietpi.txt` and search for the following 2 lines:

```
# General connection and DNS testing
# - IP to ping when checking network connectivity. Default: 1.1.1.1 (Cloudflare DNS, should be very fast world-wide)
CONFIG_CHECK_CONNECTION_IP=1.1.1.1
# - Domain to ping when checking DNS resolver. Default: one.one.one.one (Cloudflare DNS domain, see above)
CONFIG_CHECK_DNS_DOMAIN=one.one.one.one
```

you can change it to `8.8.8.8` and `google.com`

It's a similar issue like this #3337  or #3369

Joulinar, 2021-05-27:
going to close this, feel free to reopen if needed

closed 2021-05-27 by Joulinar