MichaIng/DietPi #3441: VNC won't display decktop

K92Pi, 2020-03-28:
### Creating a bug report/issue
Hello!

I'm new at DietPi and I'd like to know why I can't make VNC work properly. 
I would like to give a chance to this OS, but I already have issues at the beginning. 
For me it's very important to being able to connect through VNC, because my RPi 4 doesn't have a monitor, in other words, I'm running headless. 
I'm using RealVNC client and RealVNC server on the Pi. 

#### Required Information
- DietPi version
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=28
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
-bash: or: command not found
bustercat /etc/debian_version

- Kernel version
Linux DietPiServer 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux

- SBC device 
RPi 4 Model B (armv7l)

- Power supply used | (EG: 5V 1A RAVpower)

- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | VNC
- Was the software title installed freshly or updated/migrated?
Installed freshly

- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | reference code: 2bea0156-316a-4703-946b-a8f8f643c89f

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- Show desktop in VNC client

#### Actual behaviour

- VNC client says: Cannot currently show the desktop

#### Extra details

- ...

Joulinar, 2020-03-28:
Hi,

How do you connect to the VNC server? Do you simply use the IP address? If yes, pls add a :1 at the end.

MichaIng, 2020-03-28:
Yes use virtual VNC mode (default), do not enable desktop autostart and connect to default display ID 1 as suggested by Joulinar.

K92Pi, 2020-03-29:
> Hi,
> 
> How do you connect to the VNC server? Do you simply use the IP address? If yes, pls add a :1 at the end.

I use the IP address. If I add :1 it won't connect.

MichaIng, 2020-03-29:
@K92Pi 
That is strange, since it is a basically part of the VNC protocol to choose the screen index. Which VNC client do you use? Probably you find some settings where you can choose the screen index/ID to connect to.

GvY85, 2020-04-10:
Do you actually have a desktop running? If your in the default CLI mode: VNC does not support that so youl have to connect trough SSH first and enable an actual desktop

Joulinar, 2020-04-10:
well you could install / operate RealVNC without a specific desktop.

MichaIng, 2020-04-16:
@Joulinar 
Is that so? Never tried but what does the VNC client show when connecting to a RealVNC server without any desktop (or default xstart application) installed?

Joulinar, 2020-04-16:
Ahh I guess I was a little bit unspecific on my replay. What I was trying to say, you don't need to select anything special on DietPi. Just go to install RealVNC using `dietpi-software` and magic happen. I meant that there is no need to specify or install anything in addition. DietPi will take care on dependencies. As well it is not needed to put a desktop into autostart.

MichaIng, 2020-04-21:
@Joulinar 
Ah yes that is true, LXDE will be installed automatically if no other desktop is installed or selected.

Joulinar, 2020-05-11:
@K92Pi 
Does your issue still persist?

MichaIng, 2020-05-18:
Marking as closed due to missing reply. Feel free to reopen if required.

closed 2020-05-18 by MichaIng