MichaIng/DietPi #3445: Failed to start lighttpd service after running letenscrypt

saytinh, 2020-03-30:
Dear Alls,

My Pi3 installed with newest dietpi and configured it with Owncloud. I want to implement https for owncloud so I did a dietpi-letsencrypt.

The problem is that after running dietpi-letsencrypt, the service lightttpd always got failed like below. So my way to come back work is removing lighttp and install it again. 

Really looking for advice for how to implement letencypt for owncloud. Thanks

> sudo systemctl status lighttpd.service
● lighttpd.service - Lighttpd Daemon
   Loaded: loaded (/lib/systemd/system/lighttpd.service; disabled; vendor
 preset: enabled)
   Active: failed (Result: exit-code) since Mon 2020-03-30 16:
21:58 BST; 3min 31s ago
  Process: 1103 ExecStartPre=/usr/sbin/lighttpd -tt -f /etc/lighttpd/ligh
ttpd.conf (code=exited, status=255)

MichaIng, 2020-03-30:
@saytinh 
Many thanks for your report. Could you paste the output of the following commands:
```
journalctl -u lighttpd
cat /var/log/lighttpd/error.log
```

saytinh, 2020-03-30:
Great. Here it is for first command. For the second one I found nothing in error.log after run it.

> dietpi@DietPi:/var/log/lighttpd$ journalctl -u lighttpd
-- Logs begin at Thu 2016-11-03 17:16:42 GMT, end at Mon 2020-03-30 16:38:39 BST. --
Mar 30 16:21:56 DietPi systemd[1]: Starting Lighttpd Daemon...
Mar 30 16:21:56 DietPi lighttpd[1052]: 2020-03-30 16:21:56: (network.c.598) SSL: BIO_read_filename('/etc/letsencrypt/live/thongtanxa.tienic.net/combined.pem') failed
Mar 30 16:21:56 DietPi systemd[1]: lighttpd.service: Control proc
ess exited, code=exited status=255
Mar 30 16:21:56 DietPi systemd[1]: Failed to start Lighttpd Daemo
n.
Mar 30 16:21:56 DietPi systemd[1]: lighttpd.service: Unit entered
 failed state.
Mar 30 16:21:56 DietPi systemd[1]: lighttpd.service: Failed with
result 'exit-code'.
Mar 30 16:21:56 DietPi systemd[1]: lighttpd.service: Service hold-off time over, scheduling restart.
Mar 30 16:21:56 DietPi systemd[1]: Stopped Lighttpd Daemon.
Mar 30 16:21:56 DietPi systemd[1]: Starting Lighttpd Daemon...
Mar 30 16:21:56 DietPi lighttpd[1076]: 2020-03-30 16:21:56: (network.c.598) SSL: BIO_read_filename('/etc/letsencrypt/live/thongtanxa.tienic.net/combined.pem') failed
Mar 30 16:21:56 DietPi systemd[1]: lighttpd.service: Control proc
ess exited, code=exited status=255
Mar 30 16:21:56 DietPi systemd[1]: Failed to start Lighttpd Daemo
n.
Mar 30 16:21:56 DietPi systemd[1]: lighttpd.service: Unit entered
 failed state.
Mar 30 16:21:56 DietPi systemd[1]: lighttpd.service: Failed with
result 'exit-code'.
Mar 30 16:21:57 DietPi systemd[1]: lighttpd.service: Service hold-off time over, scheduling restart.
Mar 30 16:21:57 DietPi systemd[1]: Stopped Lighttpd Daemon.
Mar 30 16:21:57 DietPi systemd[1]: Starting Lighttpd Daemon...
Mar 30 16:21:57 DietPi lighttpd[1085]: 2020-03-30 16:21:57: (network.c.598) SSL: BIO_read_filename('/etc/letsencrypt/live/thongtanxa.tienic.net/combined.pem') failed
Mar 30 16:21:57 DietPi systemd[1]: lighttpd.service: Control process exited, code=exited status=255
Mar 30 16:21:57 DietPi systemd[1]: Failed to start Lighttpd Daemon.
Mar 30 16:21:57 DietPi systemd[1]: lighttpd.service: Unit entered failed state.
Mar 30 16:21:57 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
Mar 30 16:21:57 DietPi systemd[1]: lighttpd.service: Service hold-off time over, scheduling restart.
Mar 30 16:21:57 DietPi systemd[1]: Stopped Lighttpd Daemon.
Mar 30 16:21:57 DietPi systemd[1]: Starting Lighttpd Daemon...
Mar 30 16:21:57 DietPi lighttpd[1094]: 2020-03-30 16:21:57: (network.c.598) SSL: BIO_read_filename('/etc/letsencrypt/live/thongtanxa.tienic.net/combined.pem') failed
Mar 30 16:21:57 DietPi systemd[1]: lighttpd.service: Control process exited, code=exited status=255
Mar 30 16:21:57 DietPi systemd[1]: Failed to start Lighttpd Daemon.
Mar 30 16:21:57 DietPi systemd[1]: lighttpd.service: Unit entered failed state.
Mar 30 16:21:57 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
Mar 30 16:21:58 DietPi systemd[1]: lighttpd.service: Service hold-off time over, scheduling restart.
Mar 30 16:21:58 DietPi systemd[1]: Stopped Lighttpd Daemon.
Mar 30 16:21:58 DietPi systemd[1]: Starting Lighttpd Daemon...
Mar 30 16:21:58 DietPi lighttpd[1103]: 2020-03-30 16:21:58: (network.c.598) SSL: BIO_read_filename('/etc/letsencrypt/live/thongtanxa.tienic.net/combined.pem') failed
Mar 30 16:21:58 DietPi systemd[1]: lighttpd.service: Control process exited, code=exited status=255
Mar 30 16:21:58 DietPi systemd[1]: Failed to start Lighttpd Daemon.
Mar 30 16:21:58 DietPi systemd[1]: lighttpd.service: Unit entered failed state.
Mar 30 16:21:58 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
Mar 30 16:21:58 DietPi systemd[1]: lighttpd.service: Service hold-off time over, scheduling restart.
Mar 30 16:21:58 DietPi systemd[1]: Stopped Lighttpd Daemon.
Mar 30 16:21:58 DietPi systemd[1]: lighttpd.service: Start request repeated too quickly.
Mar 30 16:21:58 DietPi systemd[1]: Failed to start Lighttpd Daemon.
Mar 30 16:21:58 DietPi systemd[1]: lighttpd.service: Unit entered failed state.
Mar 30 16:21:58 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.

MichaIng, 2020-03-30:
@saytinh 
I replaced your domain, as you might not want to make it public here 😉. Okay it failed to load the combined certificate + key file. Could you paste:
```
ls -al /etc/letsencrypt/live/my.domain.org/
```
Again you might want to replace your actual domain name.

saytinh, 2020-03-30:
Stupid me. I already deleted directory /etc/letsencrypt before getting your reply. Now I tried to uninstall and install lightttpd webserver again but I still failed to get webserver started. Below is output for journalctl -xe

> Unit lighttpd.service has begun starting up.
Mar 30 16:58:33 DietPi lighttpd[2433]: 2020-03-30 16:58:33: (network.c.598) SSL: BIO_read_filename('/etc/letsencrypt/live/thongtanxa.tienic.net/combined.pem') failed
Mar 30 16:58:33 DietPi systemd[1]: lighttpd.service: Control process exited, code=exited status=255
Mar 30 16:58:33 DietPi systemd[1]: Failed to start Lighttpd Daemon.
-- Subject: Unit lighttpd.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Unit lighttpd.service has failed.

Joulinar, 2020-03-30:
Hi,

as you deleted `/etc/letsencrypt`, you would need to recreated your certificates or remove SSL certificate configuration from lighttpd. Currently lighttpd is still trying to load the certificate files. And for sure this is failing as they did not exist anymore.

saytinh, 2020-03-30:
Hi @Joulinar 

Thanks for suggesting. But my question is how I do it?

Joulinar, 2020-03-30:
you could simply go to `/etc/lighttpd/conf-enabled` and remove the following files if exist

```
letsencrypt.conf
redirect.conf
```

once done try to restart the webserver `systemctl restart lighttpd.service`

saytinh, 2020-03-30:
Thanks @Joulinar. I got it worked now.

Joulinar, 2020-03-30:
@saytinh 
But you did not run https now. So if you like to have https enabled, you would need to recreate certificates again

saytinh, 2020-03-30:
> @saytinh
> I replaced your domain, as you might not want to make it public here 😉. Okay it failed to load the combined certificate + key file. Could you paste:
> 
> ```
> ls -alL /etc/letsencrypt/live/my.domain.org/
> ```
> 
> Again you might want to replace your actual domain name.

Hi @MichaIng 

Here is output for something you mentioned for journalctl -u lighttpd and ls -alL /etc/letsencrypt/live/my.domain.org/. What should I do next? Thanks

> Mar 30 18:21:58 DietPi systemd[1]: Starting Lighttpd Daemon...
Mar 30 18:21:58 DietPi lighttpd[1091]: 2020-03-30 18:21:58: (network.c.598) SSL: BIO_read_filename('/etc/letsencrypt/live/my.domain.org/combined.pem') failed
Mar 30 18:21:58 DietPi systemd[1]: lighttpd.service: Control process exited, code=exited sta
tus=255
Mar 30 18:21:58 DietPi systemd[1]: Failed to start Lighttpd Daemon.
Mar 30 18:21:58 DietPi systemd[1]: lighttpd.service: Unit entered failed state.
Mar 30 18:21:58 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
Mar 30 18:21:58 DietPi systemd[1]: lighttpd.service: Service hold-off time over, scheduling restart.Mar 30 18:21:58 DietPi systemd[1]: Stopped Lighttpd Daemon.
Mar 30 18:21:58 DietPi systemd[1]: Starting Lighttpd Daemon...
Mar 30 18:21:58 DietPi lighttpd[1108]: 2020-03-30 18:21:58: (network.c.598) SSL: BIO_read_filename('/etc/letsencrypt/live/my.domain.org/combined.pem') failed
Mar 30 18:21:58 DietPi systemd[1]: lighttpd.service: Control process exited, code=exited sta
tus=255

> total 36
drwxr-xr-x 2 root root 4096 Mar 30 18:25 .
drwx------ 3 root root 4096 Mar 30 18:25 ..
-rw-r--r-- 1 root root 2273 Mar 30 18:25 cert.pem
-rw-r--r-- 1 root root 1647 Mar 30 18:25 chain.pem
-rw-r--r-- 1 root root 5545 Mar 30 18:25 combined.pem
-rw-r--r-- 1 root root 3920 Mar 30 18:25 fullchain.pem
-rw-r--r-- 1 root root 3272 Mar 30 18:25 privkey.pem
-rw-r--r-- 1 root root  692 Mar 30 18:25 README

Joulinar, 2020-03-30:
@saytinh 
do you still have the issue?

saytinh, 2020-03-30:
> @saytinh
> do you still have the issue?

Yep. I still got issue as lighttpd failed to start. You can see the output in my previous post.

Joulinar, 2020-03-30:
@MichaIng 
Seems similar to this old issue. Permission was not the problem. More reading the `combined.pem`. Could it be corrupted? 

#1544

https://dietpi.com/phpbb/viewtopic.php?t=2861

saytinh, 2020-03-30:
> @MichaIng
> Seems similar to this old issue. Permission was not the problem. More reading the `combined.pem` file. Could it be corrupted?
> 
> #1544
> 
> https://dietpi.com/phpbb/viewtopic.php?t=2861

saytinh, 2020-03-30:
This output below might help you to investigate

> Mar 30 19:09:51 DietPi systemd[1]: Started Lighttpd Daemon.
-- Subject: Unit lighttpd.service has finished start-up
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Unit lighttpd.service has finished starting up.
--
-- The start-up result is done.
Mar 30 19:09:52 DietPi lighttpd[2329]: 2020-03-30 19:09:51: (network.c.464) can't bind to port:  443 Address already in use
Mar 30 19:09:52 DietPi systemd[1]: lighttpd.service: Main process exited, code=exited, status=255/n/a
Mar 30 19:09:52 DietPi systemd[1]: lighttpd.service: Unit entered failed state.
Mar 30 19:09:52 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
Mar 30 19:09:52 DietPi systemd[1]: lighttpd.service: Service hold-off time over, scheduling restart.
Mar 30 19:09:52 DietPi systemd[1]: Stopped Lighttpd Daemon.
-- Subject: Unit lighttpd.service has finished shutting down
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Unit lighttpd.service has finished shutting down.

Joulinar, 2020-03-30:
this is an interesting message

` can't bind to port: 443`

pls can you `stop` lighttpd.service and `disable` it. pls do a `reboot` afterwads

saytinh, 2020-03-30:
Hear it is for lsof -i -P -n | grep LISTEN

> openvpn    334 nobody    6u  IPv4   8860      0t0  TCP *:443 (LISTEN)
dropbear   806   root    3u  IPv4  11731      0t0  TCP *:22 (LISTEN)
dropbear   806   root    4u  IPv6  11732      0t0  TCP *:22 (LISTEN)
smbd       860   root   30u  IPv6  12591      0t0  TCP *:445 (LISTEN)
smbd       860   root   31u  IPv6  12592      0t0  TCP *:139 (LISTEN)
smbd       860   root   32u  IPv4  12593      0t0  TCP *:445 (LISTEN)
smbd       860   root   33u  IPv4  12594      0t0  TCP *:139 (LISTEN)
redis-ser  870  redis    5u  IPv4  12600      0t0  TCP 127.0.0.1:6379 (LISTEN)
mysqld     985  mysql   17u  IPv4  14346      0t0  TCP 127.0.0.1:3306 (LISTEN)

Joulinar, 2020-03-30:
well you have OpenVPN running on HTTPS port 443. That's a good reason why lighttpd is not able to start. Was there a reason to configure OpenVPN that way? Because the port 443 can be used only once. So you need to decide which application should use it.

saytinh, 2020-03-30:
> well you have OpenVPN running on HTTPS port 443. That's a good reason why lighttpd is not able to start. Was there a reason to configure OpenVPN that way? Because the port 443 can be used only once. So you need to decide which application should use it.

I really got it worked. What on earth a simple solution. Thanks all @Joulinar and @MichaIng

Joulinar, 2020-03-30:
But now you would need to reconfigure OpenVPN to don't use port 443 again

saytinh, 2020-03-30:
> But now you would need to reconfigure OpenVPN to don't use port 443 again

Understood. It doesn't matter. Appreciate for your help @Joulinar

MichaIng, 2020-03-30:
@Joulinar 
Great, thanks for figuring out the issue. Would have not guessed VPN on that port 👍. @saytinh I thought about corrupted combined.pem as well, but your output showed that at least the file sizes fit 100%:
```
-rw-r--r-- 1 root root 2273 Mar 30 18:25 cert.pem
-rw-r--r-- 1 root root 1647 Mar 30 18:25 chain.pem
-rw-r--r-- 1 root root 5545 Mar 30 18:25 combined.pem
-rw-r--r-- 1 root root 3920 Mar 30 18:25 fullchain.pem
-rw-r--r-- 1 root root 3272 Mar 30 18:25 privkey.pem
```
combined.pem = cert.pem + privkey.pem
5545 = 2273 + 3272 matches.

saytinh, 2020-03-31:
> combined.pem = cert.pem + privkey.pem

That's absolute superb thing. Cheers

closed 2020-03-31 by saytinh

Cjkeenan, 2020-09-22:
I am having a similar issue but it only popped up after reconfiguring Pi-Hole. This is one of the outputs from `journalctl -u lighttpd`:
```
Sep 22 02:23:28 BH-3B systemd[1]: lighttpd.service: Scheduled restart job, restart counter is at 4.                                                                                             
Sep 22 02:23:28 BH-3B systemd[1]: Stopped Lighttpd Daemon.                                                                                                                                      
Sep 22 02:23:28 BH-3B systemd[1]: Starting Lighttpd Daemon...                                                                                                                                   
Sep 22 02:23:28 BH-3B lighttpd[3316]: Duplicate config variable in conditional 0 global: server.error-handler-404                                                                               
Sep 22 02:23:28 BH-3B lighttpd[3316]: 2020-09-22 02:23:28: (configfile.c.1289) source: /etc/lighttpd/conf-enabled/99-dietpi-pihole.conf line: 31 pos: 1 parser failed somehow near here: (EOL)  
Sep 22 02:23:28 BH-3B lighttpd[3316]: 2020-09-22 02:23:28: (configfile.c.1289) source: find /etc/lighttpd/conf-enabled -name '*.conf' -a ! -name 'letsencrypt.conf' -printf 'include "%p"\n' 2>/
dev/null line: 7 pos: 8 parser failed somehow near here: (EOL)                                                                                                                                  
Sep 22 02:23:28 BH-3B lighttpd[3316]: 2020-09-22 02:23:28: (configfile.c.1289) source: /etc/lighttpd/lighttpd.conf line: 92 pos: 1 parser failed somehow near here: (EOL)                       
Sep 22 02:23:28 BH-3B systemd[1]: lighttpd.service: Control process exited, code=exited, status=255/EXCEPTION                                                                                   
Sep 22 02:23:28 BH-3B systemd[1]: lighttpd.service: Failed with result 'exit-code'.                                                                                                             
Sep 22 02:23:28 BH-3B systemd[1]: Failed to start Lighttpd Daemon.                                                                                                                              
Sep 22 02:23:28 BH-3B systemd[1]: lighttpd.service: Service RestartSec=100ms expired, scheduling restart.                                                                                       
Sep 22 02:23:28 BH-3B systemd[1]: lighttpd.service: Scheduled restart job, restart counter is at 5.                                                                                             
Sep 22 02:23:28 BH-3B systemd[1]: Stopped Lighttpd Daemon.                                                                                                                                      
Sep 22 02:23:28 BH-3B systemd[1]: lighttpd.service: Start request repeated too quickly.                                                                                                         
Sep 22 02:23:28 BH-3B systemd[1]: lighttpd.service: Failed with result 'exit-code'.                                                                                                             
Sep 22 02:23:28 BH-3B systemd[1]: Failed to start Lighttpd Daemon.
```

saytinh, 2020-09-22:
Are you using the same 443 port for more than one service? I had the same problem before and get it solved by preventing that. Good luck.

MichaIng, 2020-09-22:
Did you tell Pi-hole to install/configure the webserver? Pi-hole sets up Lighttpd in a very different way, with an own hardcoded config that ignores the letsencrypt config actively, has a different web root, no FastCGI PHP etc etc, so that is only reasonable for little Pi-hole only machines where nothing else is running. On DietPi I recommend to install it through `dietpi-software` which sets up any desired webserver and PHP the "regular" way, i.e. like you'd usual do with standalone PHP-FPM server, compatible with parallel websites (aside of Pi-hole admin panel) and such.

Let me see how to restore the original Lighttpd config.

Cjkeenan, 2020-09-22:
I installed it originally with dietpi-software, but wanted to reconfigure so yes I let it set up the web server. Is there a way to reinstall via dietpi-software?

Joulinar, 2020-09-22:
> Is there a way to reinstall via dietpi-software?

yes, usually you could use `dietpi-software reinstall 93`

But this will not help in your case atm as there are other configurations broken now.
You have the following issue `Duplicate config variable in conditional 0 global: server.error-handler-404   `

To fix Lighttpd, run: `sed -i '/^[[:blank:]]*server.error-handler-404/d' /etc/lighttpd/lighttpd.conf`
- Note that this removes any custom 404 handler page. Pi-hole itself uses the 404 handler to show the blocking page on blocked requests.
- If you want to disable the blocking page instead, run the above command with `/etc/lighttpd/conf-available/99-dietpi-pihole.conf` as target.
- Finally (re)start Lighttpd: `systemctl restart lighttpd`

MichaIng, 2020-09-22:
First restore the original Lighttpd config and drop purge PHP-CGI:
`/etc/lighttpd/lighttpd.conf`:
```
server.modules = (
	"mod_indexfile",
	"mod_access",
	"mod_alias",
 	"mod_redirect",
)

server.document-root        = "/var/www/html"
server.upload-dirs          = ( "/var/cache/lighttpd/uploads" )
server.errorlog             = "/var/log/lighttpd/error.log"
server.pid-file             = "/var/run/lighttpd.pid"
server.username             = "www-data"
server.groupname            = "www-data"
server.port                 = 80

# strict parsing and normalization of URL for consistency and security
# https://redmine.lighttpd.net/projects/lighttpd/wiki/Server_http-parseoptsDetails
# (might need to explicitly set "url-path-2f-decode" = "disable"
#  if a specific application is encoding URLs inside url-path)
server.http-parseopts = (
  "header-strict"           => "enable",# default
  "host-strict"             => "enable",# default
  "host-normalize"          => "enable",# default
  "url-normalize-unreserved"=> "enable",# recommended highly
  "url-normalize-required"  => "enable",# recommended
  "url-ctrls-reject"        => "enable",# recommended
  "url-path-2f-decode"      => "enable",# recommended highly (unless breaks app)
 #"url-path-2f-reject"      => "enable",
  "url-path-dotseg-remove"  => "enable",# recommended highly (unless breaks app)
 #"url-path-dotseg-reject"  => "enable",
 #"url-query-20-plus"       => "enable",# consistency in query string
)

index-file.names            = ( "index.php", "index.html" )
url.access-deny             = ( "~", ".inc" )
static-file.exclude-extensions = ( ".php", ".pl", ".fcgi" )

compress.cache-dir          = "/var/cache/lighttpd/compress/"
compress.filetype           = ( "application/javascript", "text/css", "text/html", "text/plain" )

# default listening port for IPv6 falls back to the IPv4 port
include_shell "/usr/share/lighttpd/use-ipv6.pl " + server.port
include_shell "/usr/share/lighttpd/create-mime.conf.pl"
include "/etc/lighttpd/conf-enabled/*.conf"

#server.compat-module-load   = "disable"
server.modules += (
	"mod_compress",
	"mod_dirlisting",
	"mod_staticfile",
)
```
+
```
apt purge php7.3-cgi
```
Then:
```
dietpi-software reinstall 84 93
```
That reconfigures Lighttpd as well.

Was there a specific reason why you chose to "reconfigure" instead of just "repair" or update? That is basically a completely fresh install from ground up including all config choices.

Cjkeenan, 2020-09-22:
Thank you guys so much for the help. So i replaced the entire `lighttpd.conf` file with what you provided, then purged, then went to reinstall and the reinstall is failing whenever the `Restarting lighttpd service...` command is issued. Any ideas?

Joulinar, 2020-09-22:
can you post the service log `journalctl -u lighttpd` to see what the issue is about

Cjkeenan, 2020-09-22:
It looks to be the same error to me no?

Joulinar, 2020-09-22:
ah ok you updated your post. Still the issue with the `Duplicate config variable in conditional 0 global: server.error-handler-404`

You could run `grep -r 'server.error-handler-404' /etc/lighttpd/` to find the duplicate entry. And follow this https://github.com/MichaIng/DietPi/issues/3445#issuecomment-696618530 to have it solved

Cjkeenan, 2020-09-22:
```
root@BH-3B:/etc/php/7.3/cgi/conf.d# grep -r 'server.error-handler-404' /etc/lighttpd/                                                                                                           
/etc/lighttpd/lighttpd.conf.orig:server.error-handler-404    = "/pihole/index.php"                                                                                                              
/etc/lighttpd/conf-available/99-dietpi-pihole.conf:server.error-handler-404 = "/html/pihole/index.php"   
```

Joulinar, 2020-09-22:
```
sed -i '/^[[:blank:]]*server.error-handler-404/d' /etc/lighttpd/lighttpd.conf
systemctl restart lighttpd
```

Cjkeenan, 2020-09-22:
Seems to be working now, checking with a reboot to see if it starts properly and everything.

Cjkeenan, 2020-09-22:
Also in regards to services and who controls them, systemd vs dietpi, what exactly is the difference?

MichaIng, 2020-09-22:
While the setting was present in lighttpd.conf.orig only,  removing it from lighttpd.conf solved the issue? :smiley: However, good that it's working now.

The service control, if you mean the systemd units, it's basically about what starts the service on boot:
- systemd (the default OS init system and service handler on Debian) starts them at boot, if they are "enabled" as fast as their Before=/After= directives are satisfied. But since those can never cover all in-fact dependencies, especially when custom services come into play which are not installed via DEB packages from official Debian repository, ugly error messages or even service startups can happen.
- Prominent example is ownCloud/Nextcloud, where a cron job handles maintenance/cleanup tasks every 5 minutes which is not possible before the database ia running, same is for requests from PHP e.g. through webserver to any cloud dav backend. And and PHP request from the webserver fails if the PHP-FPM server is not yet running. While this does not really cause issues in this case, at least it messes your logs with error messages that distract when you want to look for real issues. So it basically makes sense to start services in a specific order where you know that such service dependencies exist.
- That is what dietpi-services does at the end of the boot process when those have not yet been started by systemd before.
- The other thing is stopping services during installs, updates and backups to prevent data loss or other issues when tasks interfere, but that is a different switch (exclude/include).

Joulinar, 2020-09-22:
ah, have overlooked that it was a `lighttpd.conf.orig`. Indeed strange that it is solved than.

Cjkeenan, 2020-09-22:
> ah, have overlooked that it was a `lighttpd.conf.orig`. Indeed strange that it is solved than.

Is there something else I should do?

> The service control, if you mean the systemd units, it's basically about what starts the service on boot:

So having everything on DietPi controlled would be fine? I noticed that if I have dropbear controlled by dietpi vs systemd, it does not seem to auto start.

Also is there any system that can auto-restart after a service goes down for whatever reason or does it already do that?

MichaIng, 2020-09-22:
Just to be sure, could you paste:
```
cat /etc/lighttpd/lighttpd.conf
```

Ah good find with Dropbear. dietpi-services does not handle any service by default that could be required for the current shell session or network, otherwise it would kill itself. This includes SSH and VNC servers and as well Pi-hole (as the system could use it for DNS resolution). The dietpi-services UI should show you "excluded" while it would need to be set to "included". But do not do that for SSH as it makes it unusable for many DietPi tasks. We should show enabled/disabled instead of DietPi/systemd controlled when the service is not included for DietPi handling. And print another warning when attempting to include known SSH and VPN servers at least.

Cjkeenan, 2020-09-22:
> Just to be sure, could you paste:
> 
> ```
> cat /etc/lighttpd/lighttpd.conf
> ```

It seems to heavily truncate the file, but this is what was outputted:
```
    "text/plain",                                                                                                                                                                                                                                                              
    "text/xml"                                                                                                                                                                                                                                                                 
)                                                                                                                                                                                                                                                                              
                                                                                                                                                                                                                                                                               
mimetype.assign = (                                                                                                                                                                                                                                                            
    ".ico"   => "image/x-icon",                                                                                                                                                                                                                                                
    ".jpeg"  => "image/jpeg",                                                                                                                                                                                                                                                  
    ".jpg"   => "image/jpeg",                                                                                                                                                                                                                                                  
    ".png"   => "image/png",                                                                                                                                                                                                                                                   
    ".svg"   => "image/svg+xml",                                                                                                                                                                                                                                               
    ".css"   => "text/css; charset=utf-8",                                                                                                                                                                                                                                     
    ".html"  => "text/html; charset=utf-8",                                                                                                                                                                                                                                    
    ".js"    => "text/javascript; charset=utf-8",                                                                                                                                                                                                                              
    ".json"  => "application/json; charset=utf-8",                                                                                                                                                                                                                             
    ".map"   => "application/json; charset=utf-8",                                                                                                                                                                                                                             
compress.cache-dir = "/var/cache/lighttpd/compress/"                                                                                                                                                                                                                           
compress.filetype  = (                                                                                                                                                                                                                                                         
    "application/json",                                                                                                                                                                                                                                                        
    "application/vnd.ms-fontobject",                                                                                                                                                                                                                                           
[... cut ...]                                                                                                                                                                                   
# This uses include_shell instead of an include wildcard for compatibility                                                                                                                                                                                                     include_shell "cat external.conf 2>/dev/null" 
```

MichaIng, 2020-09-22:
Dammit, that is again the Pi-hole config. Strange that the 404 handler was not found inside there but inside the original. I'm confused.

It seems like the Pi-hole installer ignores cmd arguments when the setupVars already exist (it wad installed before). So that needs to be edited manually then.

However, as it works for now and I'm on mobile phone currently, let me come back with a manual revert this evening.

Cjkeenan, 2020-09-22:
Whatever you need me to do I can help. The backup system for Pi-Hole is pretty great so nbd there. The greater issue is its 5:24am for me haha, so we can get back to this when you have time.

MichaIng, 2020-09-22:
Okay, I just checked the Pi-hole installer code and indeed on reinstall `/etc/pihole/setupVars.conf` overrides even command options. So:
```
G_CONFIG_INJECT 'INSTALL_WEB_SERVER=' 'INSTALL_WEB_SERVER=false' /etc/pihole/setupVars.conf
mv /etc/lighttpd/lighttpd.conf{.orig,}
rm /etc/lighttpd/external.conf
G_CONFIG_INJECT 'server.document-root' 'server.document-root = "/var/www"' /etc/lighttpd/lighttpd.conf
sed -i '/"url-normalize-required"/s/"enable"/"disable"/g' /etc/lighttpd/lighttpd.conf
G_CONFIG_INJECT '"mod_setenv",' '	"mod_setenv",' /etc/lighttpd/lighttpd.conf '"mod_.+",'
G_AGP php7.3-cgi
```

Cjkeenan, 2020-09-22:
I ran all of that, how do I verify it had the intended effect?

MichaIng, 2020-09-22:
Good question 😄. If the Pi-hole web UI is reachable fine, now that `php7.3-cgi` has been purged, then `php7.3-fpm` is used as intended now and it nothing conflicts. In all other aspects there is no real noticeable difference between both configs.

Cjkeenan, 2020-09-22:
I think it is fine then question mark? :smile: Thank you guys so much for all of your help. It is quite refreshing actually having devs that are this responsive.