MichaIng/DietPi #3446: Add tmpfs to Kodi temp directory

glogiotatidis, 2020-03-30:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
Kodi saves temporary files, including `kodi.log` under `~/.kodi/temp` which it occasionally cleans up. To avoid SD card corruption we could mount `~/.kodi/temp` with tmpfs

#### Describe the solution you'd like:
- Mount `~/.kodi/temp` with tmpfs

MichaIng, 2020-03-30:
@glogiotatidis 
Many thanks for your request, great idea.

What I was actually thinking of is to switch our starter to a systemd service with a fixed `kodi` user. This allows us to grant the process specific permissions and have a central config and data dir, as we have it for most other software services. I guess it would be possible as well then to statically move the Kodi tmp dir to e.g. `/run/kodi` or similar. What is Kodi actually storing there, how large needs the tmpfs to be?

glogiotatidis, 2020-03-30:
+1 on using systemd.

Kodi uses `.kodi/temp` for `special://logpath` and `special://temp`  by default. According to [Kodi doc](https://kodi.wiki/view/Special_protocol) they are for 

*special://temp*
 >  Kodi's temporary directory. This path is used to cache various data during normal usage. Unless you need the log, nothing in this directory is detrimental to Kodi's operation. Normally special://home/temp 

*special://logpath*
 >  This path points to the path where the log file is saved. 

I queried 3 dietpi installs i have and it ranges from 6Mb to 90Mb. I'd say 200Mb should be enough (maybe even 100 if 200 is a stretch)

MichaIng, 2020-03-30:
@glogiotatidis 
Okay, I think `/tmp/kodi/` should work well for this. We can even configure the systemd unit to use a `PrivateTmp=true`. systemd then creates a `/tmp/systemd-private-<string>-kodi.service-<string>` directory that is readable to the Kodi process only and is removed automatically on service stop. Only for the logs this is suboptimal, but probably those can be redirected to `systemd-journald` or to `/var/log/kodi/`, hence the RAMlog, if enabled.

Harvie, 2020-06-03:
I personaly use tmpfs for /tmp
Then i use /etc/tmpfiles.d to create /tmp/kodi and symlink that to ~/.kodi/tmp

MichaIng, 2020-06-03:
Yes that is an alternative. However instead of using a symlink it is indeed cleaner to tell Kodi directly which directory to use.