MichaIng/DietPi #3447: Error during Installation of RPI-Monitor

Gill-Bates, 2020-03-30:
```
 Mode: Installing RPi-Monitor: web interface system stats
[  OK  ] DietPi-Software | Connection test: https://github.com/XavierBerger/RPi-Monitor-deb/raw/master/packages/rpimonitor_2.12-r0_all.deb
[  OK  ] DietPi-Software | wget https://github.com/XavierBerger/RPi-Monitor-deb/raw/master/packages/rpimonitor_2.12-r0_all.deb -O 66.deb
[FAILED] DietPi-Software | dpkg --force-hold,confdef,confold -i 66.deb
[ INFO ] DietPi-Software | Trying automated APT fix
[ INFO ] DietPi-Software | APT fix, please wait...
```
─────────
 - Device model : RPi 4 Model B (armv7l)
 DietPi v6.28.0

Joulinar, 2020-03-30:
Hi,

for me the installation failed as well at this stage but after 1-2 minutes it continued and RPi Monitor was applied. After a reboot the service was up. Did you wait for the installation to finish or did you cancelled it at this stage?

MichaIng, 2020-03-30:
@Gill-Bates 
Many thanks for your report.

Yes guys this is expected, not an error. Background:
- `dpkg -i package.deb` does not install dependencies automatically, only `apt-get` does this.
- If dpkg "failed" because of missing dependencies, the package is installed regardless but marked.
- `apt-get -f install` checks for missing dependencies and pulls/installs them automatically then, finishing the install of the originating package by this.

So in case of packages with lots of dependencies, which vary, based on distro version/architecture, or change regularly, instead of trying to keep track of all those, we let APT do the job for us.
___________
But actually I changed the behaviour with DietPi v6.29 and we use `apt-get install /path/to/package.deb` now. I didn't know that this is possible as all APT man pages and docs clearly say that it only allows to install packages by name and not from file. I found this working by chance and now `dietpi-software` will do this, which should reduce confusion and overhead.

Gill-Bates, 2020-03-30:
@MichaIng What a fast feedback! Thanks for picking this up so fast. Yes, the installation finished successfully, but to see a [FAILED] fears me for a moment. :-)

closed 2020-05-03 by MichaIng