MichaIng/DietPi #3448: dietpi-banner

Gill-Bates, 2020-03-30:
With the Service https://ipinfo.io/ you can add an additional banner for "Provider Information".

The script has to catch the public IP from the local Gateway and to throw it against the service above;

Example:
`curl ipinfo.io/8.8.8.8
`

Result:
`{
  "ip": "8.8.8.8",
  "hostname": "dns.google",
  "city": "Mountain View",
  "region": "California",
  "country": "US",
  "loc": "37.3860,-122.0838",
  "org": "AS15169 Google LLC",
  "postal": "94035",
  "timezone": "America/Los_Angeles",
  "readme": "https://ipinfo.io/missingauth"
`
Missing Auth means, that this request was made without having a commerical token. So the response is limited to some core parameters. But still useful :-)

MichaIng, 2020-03-30:
@Gill-Bates 
We added city + country info for the WAN IP to the banner with v6.29:
```
 ─────────────────────────────────────────────────────
 DietPi v6.29.0 (dev) : 23:10 - Mon 03/30/20
 ─────────────────────────────────────────────────────
 - Uptime : up 4 weeks, 4 days, 21 hours
 - CPU temp : 36'C : 96'F (Cool runnings)
 - LAN IP : <my.lan.ip> (eth0)
 - WAN IP : <my.wan.ip> Hamburg-Germany
 - Freespace (RootFS) : 58G
 - Freespace (userdata) : 2.5T
 - MOTD : Folding@home takes up the fight against 2019 Corona virus:
          -> https://foldingathome.org/2020/02/27/
          On x86_64 system you can install it via dietpi-software and contribute.
 ─────────────────────────────────────────────────────
```
We use this sevice:
```
2020-03-30 23:12:57 root@micha:/var/log# curl -sSfLm 2 https://ifconfig.co/json
{"ip":"<my.wan.ip>","ip_decimal":<my.wan.ip>"country":"Germany","country_eu":true,"country_iso":"DE","city":"Hamburg","hostname":"<public.hostname>","latitude":53.5496,"longitude":9.9355,"asn":"AS15943","asn_org":"<myISP>","user_agent":{"product":"curl","version":"7.68.0","raw_value":"curl/7.68.0"}}
```
However the timezone, as of your other post, is missing.

Gill-Bates, 2020-03-30:
@MichaIng Thats right. https://ifconfig.co does not deliver the Timezone. https://ipinfo.io/ will do it.

By the way: https://ifconfig.co is DNS-Protected by Cloudflare. If Cloudflare blocks your request (e.g. with a Captcha), the request will fail.

MichaIng, 2020-03-30:
@Gill-Bates 
> By the way: https://ifconfig.co is DNS-Protected by Cloudflare. If Cloudflare blocks your request (e.g. with a Captcha), the request will fail.

That is actually a good reason to switch 🤔. The same is true for dietpi.com, hence our own tiny service: https://dietpi.com/myip.php (WAN IP only, no location/provider info)

Hmm https://ipinfo.io/ gives me a wrong info about my location:
```
 city: "Norderstedt"
 region: "Schleswig-Holstein"
```
It is close to the border but https://ifconfig.co/ has it correct.

Joulinar, 2020-03-30:
btw `https://ipinfo.io/` id blocked by a blacklist that can be used by PiHole
```
 Match found in https://gitlab.com/quidsup/notrack-blocklists/raw/master/notrack-blocklist.txt:
   ipinfo.io
```
and further more, `https://ifconfig.co/` place me at "city": "Hagen", just 500km far away from my real location 😄

Gill-Bates, 2020-03-30:
@MichaIng What https://ifconfig.co does is a lookup at a geolocation provider. This provider has stored the exit nodes of all providers. I assume the exit node of your ISP is Norderstedt. Such a query will never determine your exact location. But it should be accurate enough for the time zone.

You could also do such a lookup for the survey-module to improve the statistics of your users.

MichaIng, 2020-03-30:
The city generally is a minor info anyway. The main purpose was to add the country as this is a major interesting info when using a VPN, which again is a major reason to check ones WAN IP as well.

Long-term best thing would be to host an own service for those infos and use a sub domain that bypasses Cloudflare. But I don't have the time currently to set this up.

Okay so for now both public providers have some issue in rare cases, ad blocklist or Cloudflare captcha, I think we have to live with this for now. One can always enable the custom banner entry and add an own curl, if required 😉.