MichaIng/DietPi #3449: Set Timezone based on Public IP

Gill-Bates, 2020-03-30:
What if, when the Setup Assistant tries to use the Public Gateway IP address to determinate the local timezone?

Example: `curl ipinfo.io/8.8.8.8`
Voilá: ` "timezone": "America/Los_Angeles",`

Suggestion for an additional sequence during the first-time setup assistance:

> Dietpi has automatically detected the following time zone: **America/Los_Angeles**
> Do you want to set this time zone as default now?

MichaIng, 2020-03-30:
@Gill-Bates 
Interesting idea. We could add a related `dietpi.txt` entry "auto". The only thing is that we configure this on early boot process before network is configured and this service requires network. So it would need another loop during later firstrun setup.

mr-bolle, 2020-08-01:
> Example: `curl ipinfo.io/8.8.8.8/timezone`
> Voilá: ` "timezone": "America/Los_Angeles",`

good idear :+1: 
the IP from the Google DNS Server are not correct, i get return "America/Los_Angeles", and my correct one is "Europa/Berlin". 
The following command uses the IP of the current host.

`curl https://ipinfo.io/timezone`