MichaIng/DietPi #3450: buster-odroid c2

drieksje, 2020-03-31:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

drieksje, 2020-03-31:
hi,
installed the new dietpi buster version(odroid c2) , so the only thing i can do is install the minimal version of dietpi--NO software!!!!!
[image]

MichaIng, 2020-03-31:
@drieksje 
Okay I found it. Actually `aml-libs-odroid` is only required for Kodi (simply not yet available for C2 Buster), not for Xserver in general. I'll adjust this for v6.29.

So simply remove this package from dietpi-software completely for now, since it is not available for your distro version currently.
```
sed -i 's/aml-libs-odroid//' /DietPi/dietpi/dietpi-software
```
I'll patch it back in again once it is (basically Kodi) available for arm64 Odroids on Buster.

closed 2020-05-03 by MichaIng