MichaIng/DietPi #3456: Bluealsa not registering with Dietpi

TROFER, 2020-04-03:
I've recently been trying to connect a Bluetooth speaker to my Pi 3A+ running the latest image of dietpi, after reinstalling bluealsa / enabling and disabling Bluetooth & audio repeatedly it appears that dietpi doesn't seem to detect Bluetooth devices as audio outputs even if they are fully connected.

I would be happy to provide any more information you require

Thanks, Tristan

MichaIng, 2020-04-03:
@TROFER 
First of all are the Bluetooth devices detected correctly? `bluetoothctl`

Do you mean bluez-alsa? https://github.com/Arkq/bluez-alsa
A I see now it is called bluealsa indeed nowadays.

And can you paste the output of: `aplay -l`

TROFER, 2020-04-03:
Thanks for your response, 

I have tried to use a different speaker and this seems to have no effect. 

Here is the output of ```aplay -1```
```**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 IEC958/HDMI [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: ALSA [bcm2835 ALSA], device 2: bcm2835 IEC958/HDMI1 [bcm2835 IEC958/HDMI1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
Thanks, Tristan

MichaIng, 2020-04-03:
@TROFER 
Okay indeed ALSA only detects the onboard sound card. I am not familiar with bluealsa, is it probably required to run a command to create a certain audio device from a certain bluetooth device?

And did you install all dependencies? I.e. "sbc" is required as Bluetooth audio codec, not sure if the lib only or the tools as well:
```
apt install libsbc1
```
and in case
```
apt install sbc-tools
```

TROFER, 2020-04-03:
Thanks,

I have just installed the dependencies you mentioned and it appears that libsbc1 is already installed, you may be right about bluealsa needing a command or configuration. There is an article which talks about ```~/.asoundrc``` could this be the cause of the problem?

Thanks, Tristan

MichaIng, 2020-04-03:
@TROFER 
Exactly. Please try the following:
```
cat << _EOF_ > /etc/asound.conf
defaults.bluealsa.service "org.bluealsa"
defaults.bluealsa.device "XX:XX:XX:XX:XX:XX"
defaults.bluealsa.profile "a2dp"
defaults.bluealsa.delay 10000
_EOF_
```
Replace XX:XX:XX:XX:XX:XX with the MAC address of the connected Bluetooth audio device, as shown by `bluetoothctl`.

TROFER, 2020-04-03:
Thanks, that fixed the issue

MichaIng, 2020-04-03:
@TROFER 
Great. It should be btw possible to configure it as default audio device: https://github.com/Arkq/bluez-alsa/issues/251
The following additional `/etc/asound.conf` entry could be tried:
```
pcm.!default bluealsa
ctl.!default bluealsa
```
So you don't need to add `-D bluealsa` to your aplay command and all audio players should use Bluetooth audio by defaut.

MichaIng, 2020-06-26:
Ah there is actually a bluealsa package available on the RPi repo:
```
# apt show bluealsa
Package: bluealsa
Version: 0.13
Priority: optional
Section: sound
Maintainer: Simon Long <simon@raspberrypi.org>
Installed-Size: 244 kB
Depends: libasound2 (>= 1.0.16), libbluetooth3 (>= 4.91), libc6 (>= 2.28), libdbus-1-3 (>= 1.9.14), libglib2.0-0 (>= 2.30.0), libsbc1, bluez
Homepage: https://github.com/Arkq/bluez-alsa
Download-Size: 57.8 kB
APT-Sources: https://archive.raspberrypi.org/debian buster/main armhf Packages
Description: Bluetooth ALSA Audio backend
 Packaging of the bluez-alsa audio backend to add ALSA support to
 BlueZ version 5.
```
So no need to compile oneself and it ships with a default ALSA (drop-in) config, systemd unit and all this. Even available for all other archs, so we could use this and implement Bluetooth audio into `dietpi-config` natively for all SBC/PC models: http://archive.raspberrypi.org/debian/pool/main/b/bluealsa/
udev activation, systemd unit, all is there, very nice!

MichaIng, 2020-07-15:
I reopen this issue and make a feature request out of it as most info around it is accumulated here.

shoulih, 2022-08-22:
any update for this feature request?  
BTW, if I want to pair a Bluetooth speaker to dietpi, what is the procedure now?

Thanks

MichaIng, 2022-09-18:
Try to enable Bluetooth via `dietpi-config` > Advanced Options, then:
```sh
apt install bluealsa
```
Try to bind the speakers via `bluetoothctl`, then:
```sh
alsamixer -D bluealsa # check or tune mixer settings
speaker-test -D bluealsa # test
```

shoulih, 2022-09-20:
Not able to install bluealsa with the latest dietpi 8.8 for some reason:

root@DietPi:~# apt install bluealsa
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package bluealsa is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'bluealsa' has no installation candidate
root@DietPi:~#

Joulinar, 2022-09-20:
@MichaIng a Debian package `bluealsa` did not exist.

@shoulih to enable bluetooth: `dietpi-config` > `4  : Advanced Options` >  `Bluetooth       : [On]`

MichaIng, 2022-09-20:
Just checked it, indeed it got removed with Bullseye (while it still appears as recommendation of some other packages), still present on Buster. Note that it was only present on RPi anyway: https://archive.raspberrypi.org/debian/pool/main/b/bluealsa/

It is actively developed, so looks like one better compiles it from source then installing the old packages: https://github.com/Arkq/bluez-alsa

Joulinar, 2022-09-20:
looks like it will become available with Bookworm https://packages.debian.org/bookworm/bluez-alsa-utils

MichaIng, 2022-09-20:
Great find. I wonder why they use the old name "bluez-alsa", while the project is now called "BlueALSA" 🤔. The contained binary is of course called `bluealsa` as of source code, but the config and service files are called "bluez-alsa". Confusing inconsistency 😅.

However, great to see this!

KevinGaudry, 2024-02-13:
Any idea on how to set it up as default ? (speaker is working and connected, it plays some test sound)

lupa18, 2024-09-06:
> ```shell
> echo -e 'pcm.!default bluealsa\nctl.!default bluealsa' > /etc/asound.conf
> ```

Hi ! I have a RPI5 and am trying to get this working. Using this I got it connected and playing sound but awful.  Any hint to let it working?

Also another question, what should be configuration for mpd with this approach?

Thanks in advance !

MichaIng, 2024-10-18:
With this command, BlueALSA ia setup as default audio device, which is also used by default by MPD.

About aweful audio: Can you describe a bit more what the issue is? Probably it helps wrap it into some conversion plugin, if this is not the case already.