MichaIng/DietPi #3458: NanoPi R1S-H3

ckuethe, 2020-04-04:
### Creating an image request

#### Formal device information
- Device name | NanoPi R1S-H3
- Official product URL | https://www.friendlyarm.com/index.php?route=product/product&product_id=274
- GitHub resource URL (if available) | http://wiki.friendlyarm.com/wiki/index.php/NanoPi_R1
- Image download URL | https://drive.google.com/drive/folders/19Eu--we0rmXUAtA4T0qGzz3GwyizRdct

#### Is the SBC officially supported by the Debian installer?
- Not sure

#### If not, is a reliable 3rd party Debian image available for this SBC?
- https://dietpi.com/downloads/testing/DietPi_NanoPiM1Plus-ARMv7-Buster.7z

#### If not, are there install instructions for Debian available?
http://wiki.friendlyarm.com/wiki/index.php/Building_U-boot_and_Linux_for_H5/H3/H2%2B

The M1plus DietPi image boots, wifi and serial console work, so it shouldn't be a big lift to 

#### Vote for this image on FeatHub: https://feathub.com/MichaIng/DietPi/

ckuethe, 2020-04-04:
dmesg. As you can see, it thinks it's an M1 Plus

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.28-sunxi (root@builder) (gcc version 8.3.0 (GNU Toolchain for the A-profile Architecture 8.3-2019.03 (arm-rel-8.36))) #20.02.7 SMP Sat Mar 28 14:32:15 CET 2020
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=50c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: FriendlyArm NanoPi M1 Plus
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 128 MiB at 0x55c00000
[    0.000000] On node 0 totalpages: 131072
[    0.000000]   Normal zone: 1152 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 131072 pages, LIFO batch:31
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: Using PSCI v0.1 Function IDs from DT
[    0.000000] percpu: Embedded 20 pages/cpu s49868 r8192 d23860 u81920
[    0.000000] pcpu-alloc: s49868 r8192 d23860 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129920
[    0.000000] Kernel command line: root=UUID=51464e5d-3ef8-4fd1-9612-6948befb1945 rootwait rootfstype=ext4 console=ttyS0,115200 console=tty1 hdmi.audio=EDID:0 disp.screen0_output_mode=1920x1080p60 consoleblank=0 loglevel=7 ubootpart=8945af7f-01 ubootsource=mmc usb-storage.quirks=0x2537:0x1066:u,0x2537:0x1068:u   sunxi_ve_mem_reserve=0 sunxi_g2d_mem_reserve=0 sunxi_fb_mem_reserve=16 cgroup_enable=memory swapaccount=1
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] allocated 524288 bytes of page_ext
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 365076K/524288K available (9216K kernel code, 1053K rwdata, 2880K rodata, 1024K init, 316K bss, 28140K reserved, 131072K cma-reserved, 0K highmem)
[    0.000000] random: get_random_u32 called from __kmem_cache_create+0x2b/0x3ac with crng_init=0
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 41301 entries in 81 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000005] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.000017] Switching to timer-based delay loop, resolution 41ns
[    0.000209] Console: colour dummy device 80x30
[    0.000608] printk: console [tty1] enabled
[    0.000662] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000)
[    0.000690] pid_max: default: 32768 minimum: 301
[    0.000895] LSM: Security Framework initializing
[    0.000991] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001014] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001107] *** VALIDATE tmpfs ***
[    0.001753] *** VALIDATE proc ***
[    0.002069] *** VALIDATE cgroup1 ***
[    0.002090] *** VALIDATE cgroup2 ***
[    0.002163] CPU: Testing write buffer coherency: ok
[    0.002578] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.003098] Setting up static identity map for 0x40100000 - 0x40100054
[    0.003237] rcu: Hierarchical SRCU implementation.
[    0.003788] smp: Bringing up secondary CPUs ...
[    0.014624] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.025449] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.036207] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.036308] smp: Brought up 1 node, 4 CPUs
[    0.036366] SMP: Total of 4 processors activated (192.00 BogoMIPS).
[    0.036381] CPU: All CPU(s) started in HYP mode.
[    0.036392] CPU: Virtualization extensions available.
[    0.037151] devtmpfs: initialized
[    0.043667] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.043960] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.044060] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.048903] xor: measuring software checksum speed
[    0.088155]    arm4regs  :  1255.000 MB/sec
[... 318 lines cut ...]
[    9.034846] platform regulatory.0: Falling back to sysfs fallback for: regulatory.db
[    9.046431] usb 2-1: reset high-speed USB device number 2 using ehci-platform
[    9.059424] lima 1c40000.gpu: bus rate = 200000000
[    9.064444] lima 1c40000.gpu: mod rate = 384000000
[    9.073899] [drm] Initialized lima 1.0.0 20190217 for 1c40000.gpu on minor 1
[... cut ...]
[    9.203364] sunxi-ir 1f02000.ir: initialized sunXi IR driver
[    9.205524] input: dw_hdmi as /devices/platform/soc/1ee0000.hdmi/rc/rc0/input1
[    9.400884] asoc-simple-card sound: i2s-hifi <-> 1c22800.i2s mapping ok
[    9.452780] cfg80211: failed to load regulatory.db
[   10.189009] random: crng init done
[   10.192464] random: 7 urandom warning(s) missed due to ratelimiting
[   14.140861] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   35.826297] vcc3v0: disabling
[   35.826309] vcc5v0: disabling
```

MichaIng, 2021-05-19:
I'm gonna implement NanoPi R1 support now: #4017
Again same SoC (H3) but as well two Ethernet ports. Image should work fine on both boards even that R1S-H3 has no eMMC and only 1 USB port.

There is an R1S-H5 as well, but I cannot even find it in the shop, so it doesn't seem to be relevant. That would require a new image of course, different SoC, even different architecture (ARMv6/aarch64). I mark this issue as closed in favour of the R1 request.

closed 2021-05-19 by MichaIng

MichaIng, 2021-06-01:
@ckuethe 
If it is still a topic for you, it would be great if you could test the NanoPi R1 image on your R1S-H3: https://dietpi.com/downloads/images/DietPi_NanoPiR1-ARMv7-Buster.7z

ckuethe, 2021-06-01:
Trying it now...

Observations: serial console doesn't work; I know my cable is plugged in the right way because the M1 image prints the expected boot messages. Nothing from uboot or the kernel with the R1 image.

Something is happening though. After a few seconds the red light starts doing the heartbeat pattern. If power it off and check the microsd in my laptop, the filesystem has been resized from 610MB to 32GB (same 32GB card that worked as an M1)

It's not connecting to my wifi, though it did with the M1 image.

I'll have to play with it more after work.

MichaIng, 2021-06-01:
Many thanks for testing.

Did you use UART0, the debug console, if I'm not mistaken? That should be enabled by default, but of course it's possible that the required device tree definitions of R1 and R1S are different, so that it does not work with R1 device tree on R1S. There are three overlays to enable UART1 - UART3. Okay I just see R1S has only the 3-pins UART0 connector, not UART1, right? The R1 has UART1 as well, and none of both have UART2 or UART3 exposed. Not sure if it's possible, but probably they can be swapped. Just to rule it our, in `/boot/armbianEnv.txt`, you could append ` uart1 uart2 uart3` to the overlays setting to have all four theoretically existing UART devices enabled.

You did enable WiFi in `/bootdietpi.txt` and set SSID + WPA key accordingly in `/bootdietpi-wifi.txt`? Else connecting it via Ethernet to get SSH up should work to check what's wrong with WiFi and/or the serial device/console.

Btw, with the M1 Plus image, does the second Ethernet adapter work?

ckuethe, 2021-06-02:
That's correct, the R1S has only the single 3-pin UART. This uart works with no fuss using the M1Plus image.

First difference: plugging in a stock M1Plus image starts emitting u-boot output immediately. The R1 image emits no output all from uboot.

I've added the suggested overlays to armbianEnv.txt; no change in serial console.

So I just didn't wait long enough for wifi to connect - after a while I saw it finally connect to my network.

Here's the dmesg from both M1Plus and R1; https://gist.github.com/ckuethe/97f77fdb47670921f3fd3acdd3f51d93

MichaIng, 2021-06-02:
Okay good, then the firstrun setup steps simply took a little longer.

I think I found the problem. The "console=serial" setting in `armbianEnv.txt` triggers boot messages to be printed to `console=ttyS1`, which most likely is UART1 (as present on R1), while UART0 would then be ttyS0. Those IDs do not necessarily need to match, sometimes UART0 can be ttyS2 or so, but in most cases it does match.

But there is a way to add custom console outputs. Please try:
```
C_GONFIG_INJECT 'console=' 'console=none' /boot/armbianEnv.txt
C_GONFIG_INJECT 'consoleargs=' 'consoleargs=ttyS0,115200' /boot/armbianEnv.txt
```
This disables output to the non-existent ttyS1 and enables it on ttyS0. There is also an `earlycon` flag, which sounds like it would give earlier boot messages. To enable it:
```
C_GONFIG_INJECT 'earlycon=' 'earlycon=on' /boot/armbianEnv.txt
```

Ah, lol thanks for the diff, which basically shows the ttyS0 assumption to be correct.
_________
Strange is that with the M1+ image, the memory write speeds are nearly doubled. But this test probably doesn't mean something: https://forum.armbian.com/topic/5635-raid6-numbers-meaning/
Would be interesting though if there are actual memory benchmark result differences between both images.

Also interesting is that the available system memory is the same, even a tiny mit more with the M1+ image, right? `free`
I through there might be some reservation for the GPU with the M1+ image, since that one has an HDMI port.

When you run `ip l` with the M1+ image, do you see both Ethernet interfaces?

ckuethe, 2021-06-03:
Both the M1Plus and R1S images see two onboard ethernets and a wifi. Haven't tried plugging any wired network in yet, but the interfaces identify and don't complain when I `up` them.

After setting  console and consoleargs, boot messages and serial console work, even without the uart overlays. Enabling earlycon causes the boot to hang with no output... maybe in uboot somewhere. Dunno. u-boot isn't emitting anything.

MichaIng, 2021-06-03:
Okay, many thanks for testing. Since the M1 Plus image requires no boot config changes, it seems to be the better choice then. I'll check if there maybe is an R1S-H3 device tree, or if we create an image where at least DietPi detects it as R1S, or if we merge M1 Plus and R1S-H3 device IDs.

ckuethe, 2021-06-04:
Another interesting difference: the R1 image activates a CDC-ACM gadget, giving me a usb port over serial (yes I can log in to that tty), which the M1 image doesn't do.

MichaIng, 2021-06-04:
Indeed interesting. But can the port then still be used for regular USB devices? If I'm not wrong, e.g. on RPi, the gadget mode is either-or.

Is this module loaded? `lsmod | grep g_cdc`

ckuethe, 2021-06-05:
Indeed the presence of that does prevent normal usb port operation, which is definitely not helpful for my use case. :) I was hoping that maybe only the microusb port would go into gadget mode, but it seems to happen to the regular USB-A port too. I do have a bunch of these boards so it's easy enough to test R1S mode on one while running the others in M1Plus mode.

## R1S
```
# lsmod | sort
8189es               1187840  0
Module                  Size  Used by
autofs4                36864  2
cfg80211              491520  1 8189es
cpufreq_dt             20480  0
evdev                  20480  0
g_serial               16384  0
gpio_keys              20480  0
industrialio           53248  1 sun4i_gpadc_iio
ip_tables              24576  0
libcomposite           45056  2 g_serial,usb_f_acm
phy_generic            20480  2 sunxi
pwrseq_simple          16384  1
r8152                 180224  0
realtek                20480  0
rfkill                 20480  3 cfg80211
sun4i_gpadc_iio        16384  0
sun8i_thermal          16384  0
sunxi                  16384  0
sunxi_cedrus           32768  0
u_serial               24576  3 usb_f_acm
uio                    16384  1 uio_pdrv_genirq
uio_pdrv_genirq        20480  0
usb_f_acm              20480  1
v4l2_mem2mem           20480  1 sunxi_cedrus
videobuf2_common       40960  3 sunxi_cedrus,v4l2_mem2mem,videobuf2_v4l2
videobuf2_dma_contig    20480  1 sunxi_cedrus
videobuf2_memops       20480  1 videobuf2_dma_contig
videobuf2_v4l2         20480  2 sunxi_cedrus,v4l2_mem2mem
x_tables               24576  1 ip_tables
```

## M1Plus
```
# lsmod |sort
8189es               1187840  0
Module                  Size  Used by
autofs4                36864  2
cdc_acm                32768  0
cfg80211              491520  1 8189es
cpufreq_dt             20480  0
display_connector      20480  0
dw_hdmi_cec            16384  0
dw_hdmi_i2s_audio      16384  0
evdev                  20480  0
g_serial               16384  0
gpio_keys              20480  0
gpu_sched              28672  1 lima
industrialio           53248  1 sun4i_gpadc_iio
ip_tables              24576  0
libcomposite           45056  1 g_serial
lima                   45056  0
pwrseq_simple          16384  1
r8152                 180224  0
realtek                20480  0
rfkill                 20480  3 cfg80211
snd_soc_hdmi_codec     20480  0
sun4i_gpadc_iio        16384  0
sun8i_thermal          16384  0
sunxi_cedrus           32768  0
sunxi_cir              20480  0
uio                    16384  1 uio_pdrv_genirq
uio_pdrv_genirq        20480  0
v4l2_mem2mem           20480  1 sunxi_cedrus
videobuf2_common       40960  3 sunxi_cedrus,v4l2_mem2mem,videobuf2_v4l2
videobuf2_dma_contig    20480  1 sunxi_cedrus
videobuf2_memops       20480  1 videobuf2_dma_contig
videobuf2_v4l2         20480  2 sunxi_cedrus,v4l2_mem2mem
x_tables               24576  1 ip_tables
```

`cdc_acm` is present because I have a USB GPS connected. It's detected by the M1 personality but not the R1, even when plugged in at boot time.

MichaIng, 2021-06-06:
With a single USB port that is a problem indeed. On R1 there are two USB ports, so having one by default in gadget mode is okay. Does `modprobe -r usb_f_acm g_serial` work to restore the regular USB port?

Another argument for M1 Plus image on R1S-H3 is that the second Ethernet port uses the USB bus, hence the single USB A port only, and hence the reason that second Ethernet port does not require any special device tree entry or so. On R1 it is differently attached, and due to SoC limitations only 100 Mbps, but it allows that SBC to have two USB A ports. So while it looks similar at first, R1 and R1S are quire different boards and another single Ethernet H3 image (and hence device tree) fits better for R1S.