MichaIng/DietPi #3459: Dietpi 6.28 cannot complete initial installation

razkal, 2020-04-05:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=28
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | root@DietPi:~# cat /etc/debian_version
10.3

- Kernel version | Linux DietPi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux
- SBC device | RPi 3 Model B (armv7l)
- Power supply used | 5V 2.5A 
- SDcard used | Samsung 32gb

#### Additional Information (if applicable)
#### Details:
- Date           | Sun  5 Apr 15:12:16 BST 2020
- Bug report     | N/A
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux DietPi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. Boot Rpi
2. Go to "dietpi-software-Install" and start the Dietpi installation

#### Expected behaviour:

- Dietpi should complete the initial installation after the first boot.

#### Actual behaviour:

- Dietpi cannot complete the initial installation after the first boot.

#### Extra details:

- I manually ran 'dpkg --configure -a' and I started again the installation process (failed again).

#### Additional logs:
```
E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.
```

Joulinar, 2020-04-05:
Hi,

usually if you do a fresh install/flash of DietPi to a SD Card, Initial installation should run automatically. If this is a fresh install, can you download dietpi image once again and do a fresh copy to SD card pls, just to avoid any corruption during imaging the SD card. Maybe you can use a different SD card as well.

razkal, 2020-04-05:
hi,

i redownloaded and flashed the img with balena etcher 1.5.80 but i still get the same errors...
i am pasting the last lines from the "APT update" task in case you find them useful:

Adding 'diversion of /boot/overlays/uart2.dtbo to /usr/share/rpikernelhack/overlays/uart2.dtbo by rpikernelhack'
dpkg-divert: error: unable to sync file '/usr/share/rpikernelhack/overlays/uart2.dtbo.dpkg-divert.tmp': Input/output error
tee: /var/tmp/dietpi/logs/dietpi-firstrun-setup.log: Read-only file system
                                                                          tee: /var/tmp/dietpi/logs/dietpi-software_apt.log: Read-only file system
                                                                                                                                                  dpkg: error processing archive /tmp/apt-dpkg-install-AHpeYo/10-raspberrypi-kernel_1.20200212-1_armhf.deb (--unpack):
 new raspberrypi-kernel package pre-installation script subprocess returned error exit status 2
dpkg: unrecoverable fatal error, aborting:
 unable to flush updated status of 'raspberrypi-kernel': Read-only file system
E: Sub-process /usr/bin/dpkg returned an error code (2)
/DietPi/dietpi/func/dietpi-globals: line 977: /usr/bin/mawk: Input/output error
/DietPi/dietpi/func/dietpi-globals: line 979: /usr/bin/mawk: Input/output error
/DietPi/dietpi/func/dietpi-globals: line 984: /usr/bin/tail: Input/output error
[FAILED] DietPi-Software | G_AGUG
[ INFO ] DietPi-Software | If problems persist, please report at https://github.com/MichaIng/DietPi/issues for investigation.
/DietPi/dietpi/func/dietpi-globals: line 1069: /bin/date: Input/output error
/DietPi/dietpi/func/dietpi-globals: line 1069: /bin/uname: Input/output error
/DietPi/dietpi/func/dietpi-globals: line 1132: /bin/cat: Input/output error
/DietPi/dietpi/func/dietpi-globals: line 1133: /bin/rm: Input/output error
[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.
/DietPi/dietpi/func/dietpi-globals: line 945: /bin/rm: Input/output error
/DietPi/dietpi/func/dietpi-globals: line 184: /bin/rm: Input/output error
[FAILED] DietPi-Software | Failed to remove scripts working directory: /tmp/DietPi-Software
/DietPi/dietpi/dietpi-login: line 109:  5059 Bus error               /DietPi/dietpi/func/dietpi-banner 0
/DietPi/dietpi/func/dietpi-globals: line 561: /usr/bin/mawk: Input/output error
/DietPi/dietpi/func/dietpi-globals: line 568: /bin/stty: Input/output error
/DietPi/dietpi/func/dietpi-globals: line 570: ((: <= 120 : syntax error: operand expected (error token is "<= 120 ")
/DietPi/dietpi/func/dietpi-globals: line 576: ((: >= 7 : syntax error: operand expected (error token is ">= 7 ")
/DietPi/dietpi/func/dietpi-globals: line 748: /usr/bin/whiptail: Input/output error
/DietPi/dietpi/func/dietpi-globals: line 561: /usr/bin/mawk: Input/output error
/DietPi/dietpi/func/dietpi-globals: line 568: /bin/stty: Input/output error
/DietPi/dietpi/func/dietpi-globals: line 570: ((: <= 120 : syntax error: operand expected (error token is "<= 120 ")
/DietPi/dietpi/func/dietpi-globals: line 576: ((: >= 7 : syntax error: operand expected (error token is ">= 7 ")
/DietPi/dietpi/func/dietpi-globals: line 805: /usr/bin/whiptail: Input/output error
/DietPi/dietpi/dietpi-login: line 1: /bin/rm: Input/output error
-bash: /etc/profile.d/at-dbus-fix.sh: Input/output error
-bash: /etc/profile.d/bash_completion.sh: Input/output error
-bash: /root/.profile: Input/output error
root@DietPi:~#

Joulinar, 2020-04-05:
Hmm your system seems to be mounted `read-only`. Do you have a different SD Card to try?

MichaIng, 2020-04-05:
@razkal 
Many thanks for your report. I am afraid based on your various error messages and read-only fs mount, your SDcard is broken. So I suggest as well to try a different SDcard.

You can verify this by running: `dmesg`
I am pretty sure it will show red I/O error lines.

razkal, 2020-04-05:
i just found an older sd card and it runs ok so far (i am installing now some software from dietpi-software menu).

Joulinar, 2020-04-05:
That was my thought as well. Good that is is working with a different SD card now.

closed 2020-04-05 by razkal