MichaIng/DietPi #3462: Add C.H.I.P (chip) 9$ computer

RajashekarY, 2020-04-08:
#### Details:
- Date           | Wed  8 Apr 08:33:01 UTC 2020
- Bug report     | N/A
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | N/A
- Pre-image      | N/A
- SBC device     | NULL (ID=22)
- Kernel version | Linux chip 4.4.13-ntc-mlc #1 SMP Tue Dec 6 21:38:00 UTC 2016 armv7l GNU/Linux
- Distro         | jessie (ID=3)
- Command        | G_AGI linux-image-4.4.13-ntc-mlc
- Exit code      | 100
- Software title | DietPi-PREP

#### Steps to reproduce:

1. I was trying to upgrade the NTC (Next Thing Co.) repos sources with Dietpi `PREP SYSTEM.sh`
it exited with the above error

#### Expected behaviour:

- I was assuming that this should upgrade system with latest Buster sources and security patches

#### Actual behaviour:

- command line tool raises a error like the attached image below

#### Extra details:

- It is a armhf board with built in 4GB NAND flash bluetooth wifi 80 i/o pins
I Have 3 of these boards and I have also seen this [issue](https://github.com/MichaIng/DietPi/issues/770) to getone to provide a guide
[image]

#### Additional logs:
```
E: Command line option --allow-change-held-packages is not understood
```

Joulinar, 2020-04-09:
hmm I see you are still using Debian Jessie. Best to my knowledge Jessie is not supported anymore by DietPi. But I guess @MichaIng could be more specific in this.

MichaIng, 2020-04-14:
@RajashekarY
Many thanks for your report. Hmm there are basically two issues in your case.
1. Actually we still wanted to support Jessie pre-images with DietPi-PREP, although forcing a Stretch upgrade then. However indeed we run apt-get with the `--allow-change-held-packages` option that is not yet supported on Jessie. Luckily DietPi v6.29 allows to easily workaround this. I'll implement the required change.
2. The picture you posted contains an empty `apt-mark auto` call. Could you please paste the output of: `apt-mark showmanual`

MichaIng, 2020-04-26:
**Done: https://github.com/MichaIng/DietPi/commit/8193ba9c140b12cfc11d953950612423f01354f3**

closed 2020-05-03 by MichaIng