MichaIng/DietPi #3463: First Boot Fail on Raspberry Pi 3 Model A+

dsnyder0pc, 2020-04-08:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
```
#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=28
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `buster or 10.3`
- Kernel version | `Linux cdripper 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux`
- SBC device | `RPi 3 Model A+ (armv7l) or (EG: RPi3)`
- Power supply used | `Original Apple iPod charger 5 Watts (5 volts at 1000 mA)`
- SDcard used | `SanDisk Ultra 32GB MicroSDHC UHS-I Card`
#### Additional Information (if applicable)
- Software title | `first boot of DietPi`
- Was the software title installed freshly or updated/migrated? `fresh install`
- Can this issue be replicated on a fresh installation of DietPi? `yes`

#### Steps to reproduce

1. Flash DietPi_RPi-ARMv6-Buster.img from January 26th to a fresh microSD card using Etcher
2. Pre-configure dietpi-txt with these changes from the defaults:
```
AUTO_SETUP_LOCALE=en_US.UTF-8
AUTO_SETUP_KEYBOARD_LAYOUT=us
AUTO_SETUP_TIMEZONE=America/Los_Angeles
AUTO_SETUP_NET_ETHERNET_ENABLED=0
AUTO_SETUP_NET_WIFI_ENABLED=1
AUTO_SETUP_NET_HOSTNAME=cdripper
AUTO_SETUP_AUTOMATED=1
AUTO_SETUP_INSTALL_SOFTWARE_ID=5
AUTO_SETUP_INSTALL_SOFTWARE_ID=121
CONFIG_BOOT_WAIT_FOR_NETWORK=2
CONFIG_WIFI_COUNTRY_CODE=US
CONFIG_SERIAL_CONSOLE_ENABLE=0
CONFIG_SOUNDCARD=usb-dac
```
3. Set the WIFI_SSID and WIFI_KEY in dietpi-wifi.txt

#### Expected behaviour
- The RPi should boot up, go through the resize process, reboot, connect to Wi-Fi, install updates and specified packages, reboot again, and then be reachable on the network with no human interaction after connecting power.

#### Actual behaviour
- The device is crashing with this message:
```
---------------------------------------------------------------------
- DietPi has encounted an error, and, is unable to continue         -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste the BLUE lines below, into the ticket              -
---------------------------------------------------------------------

#### Details:
- Date           | Wed  8 Apr 14:49:16 PDT 2020
- Bug report     | N/A
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model A+ (armv7l) (ID=3)
- Kernel version | Linux DietPi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | ping -c 1 -W 5 1.1.1.1
- Exit code      | 2
- Software title | DietPi-Software
```

#### Extra details
- Video of what I'm seeing on the console (boot starts six seconds in): https://youtu.be/ePAwlxNo6WA?t=6
- Complete log files and dietpi.txt/dietpi-wifi.txt: [dietpi-RPi3a-error.tar](https://www.dsnyder.ws-e.com/dietpi-RPi3a-error.tar)
- I do realize that the power supply that I'm using may be a little under-spec'd. I'll try with a 5V 3A supply and let you know if the problem goes away.
- I am only able to reproduce this problem with an HDMI cable connected, which is really strange. Frustrating because I can't see what happens when it succeeds.  🤷‍♂

Joulinar, 2020-04-08:
@lhaclove 
many thanks for your report.

Hi, if you are not able to `ping 1.1.1.1` you would have local issues as it's part of Cloudflares DNS Server. It is the globally available IP address for Cloudflares 1.1.1.1 DNS resolver:

https://www.dnswatch.info/dns/dnslookup?la=en&host=1.1.1.1&submit=Resolve

Anyway if you like to change this, you can do it as follow. 

Do `nano /DietPi/dietpi.txt` and search for the following 2 lines:

```
# General connection and DNS testing
# - IP to ping when checking network connectivity. Default: 1.1.1.1 (Cloudflare DNS, should be very fast world-wide)
CONFIG_CHECK_CONNECTION_IP=1.1.1.1
# - Domain to ping when checking DNS resolver. Default: one.one.one.one (Cloudflare DNS domain, see above)
CONFIG_CHECK_DNS_DOMAIN=one.one.one.one
```

you can change it to `8.8.8.8` and `google.com`

It's a similar issue like this #3337

dsnyder0pc, 2020-04-08:
Thanks for the quick reply. I'm not having problems ping'ing 1.1.1.1 from other RPi's on my Wi-Fi network, so I don't think I have a general problem with reaching Cloudflare. But, I'm wondering if there's just not enough juice in that old 5V 1A iPod charger to run the GPU and Wi-Fi chip in the A+. Perhaps it's going under-voltage just enough to drop off when the first-boot processes reaches the ping check. Trying again and will reply soon. Thanks!

Joulinar, 2020-04-08:
well than you have 2 option. Change the ping check or change the charger

dsnyder0pc, 2020-04-08:
I tried changing the charger: same result.

However, I immediately connected a USB keyboard and ran:
```
root@DietPi:~# ping -c 1 -W 5 1.1.1.1
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.
64 bytes from 1.1.1.1: icmp_seq=1 ttl=55 time=6.39 ms

--- 1.1.1.1 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 6.392/6.392/6.392/0.000 ms
```
So, perhaps it's some kind of first-boot race condition?

Trying changing the ping check next.

Joulinar, 2020-04-08:
well normally not, for my systems it'S working on first boot always

dsnyder0pc, 2020-04-09:
Yeah, intermittent failure under certain circumstances is a common race-condition symptom. Good news though. Switching to Google for the check as you suggested worked for my most recent test. I'll do it a few more times to confirm.

My new working set of dietpi.txt changes from the defaults are as follows:
```
         CONFIG_CHECK_CONNECTION_IP = 8.8.8.8
       CONFIG_BOOT_WAIT_FOR_NETWORK = 2
         AUTO_SETUP_KEYBOARD_LAYOUT = us
    AUTO_SETUP_NET_ETHERNET_ENABLED = 0
     AUTO_SETUP_INSTALL_SOFTWARE_ID = 5,121
           CONFIG_WIFI_COUNTRY_CODE = US
            CONFIG_CHECK_DNS_DOMAIN = google.com
        AUTO_SETUP_NET_WIFI_ENABLED = 1
            AUTO_SETUP_NET_HOSTNAME = cdripper
       CONFIG_SERIAL_CONSOLE_ENABLE = 0
                  AUTO_SETUP_LOCALE = en_US.UTF-8
               AUTO_SETUP_AUTOMATED = 1
                   CONFIG_SOUNDCARD = usb-dac
                AUTO_SETUP_TIMEZONE = America/Los_Angeles
```

I've hacked together a little script that updates `dietpi.txt` and `dietpi-wifi.txt` on a freshly flashed microSD card for parameters that are useful for making a Raspberry Pi configure itself automatically as a Roon endpoint to drive a USB DAC. It's not beautiful, with the Google changes, it's working again. So thanks!

https://github.com/dsnyder0pc/DietPi/blob/master/dietpi-setup

Here's the output of a successful boot: https://youtu.be/8uOV4iBN4KM

Connect a USB DAC, fire up Roon, and the DAC shows up without any additional configuration. Well, that's idea. I'm happy again.

Joulinar, 2020-04-09:
if you don't like google, you can change it to Quad9 as well. Feel free.

dsnyder0pc, 2020-04-09:
Well, on my 2nd attempt with the Google settings, it failed again. As before, if I connect a USB cable, I can run the ping command with no problems. The first-boot process must be trying to run the ping command too early or not waiting long enough for the first packet to come back. IMHO, specifying 5ms for ping's waittime is optimistic for Wi-Fi. Any objections to bumping that up to something like 500ms or even 5000ms?

Joulinar, 2020-04-09:
Within `dietpi.txt` you have the possibility to set a value to force the system to wait until a valid network connection was established and an IP address assigned.

dsnyder0pc, 2020-04-09:
Wait. Just found this:
```
CONFIG_G_CHECK_URL_TIMEOUT=5
```

dsnyder0pc, 2020-04-09:
I thought that this value forced the system to wait until a valid network connection was established and an IP address assigned:
```
# Delay boot until network is established: 0=disabled | 1=10 second wait max (default) | 2=infinite wait
CONFIG_BOOT_WAIT_FOR_NETWORK = 2
```
I'm using that already. Trying the bigger timeout now.

dsnyder0pc, 2020-04-09:
Ugh. Same result. Even with this:
```
ping -c 1 -W 5000 8.8.8.8
```
Exit code is 2
*The transmission was successful but no responses were received.*

Frustrating…

dsnyder0pc, 2020-04-09:
Looking at the console output, the system definitely has an IP address and has successfully completed NTPD network time sync. I don't understand why this ping test is failing. No reason for it to do so at this point.

dsnyder0pc, 2020-04-09:
I just tried sending more packets. Same result. Ugh!
```
ping -c 5 -W 5000 8.8.8.8
```
But, after more diggin, it looks like the largest waittime value ping will accept is 2147ms. Trying again with `ping -c 1 -W 2000`

dsnyder0pc, 2020-04-09:
Okay. Final set of parameters that seem to be working for me:

```
       CONFIG_BOOT_WAIT_FOR_NETWORK = 2
                   CONFIG_SOUNDCARD = usb-dac
               AUTO_SETUP_AUTOMATED = 1
         AUTO_SETUP_KEYBOARD_LAYOUT = us
                  AUTO_SETUP_LOCALE = en_US.UTF-8
         CONFIG_CHECK_CONNECTION_IP = 8.8.8.8
         CONFIG_G_CHECK_URL_TIMEOUT = 2000
            AUTO_SETUP_NET_HOSTNAME = roon-usb
                AUTO_SETUP_TIMEZONE = America/Los_Angeles
            CONFIG_CHECK_DNS_DOMAIN = google.com
       CONFIG_SERIAL_CONSOLE_ENABLE = 0
     AUTO_SETUP_INSTALL_SOFTWARE_ID = 5,121
        AUTO_SETUP_NET_WIFI_ENABLED = 1
           CONFIG_WIFI_COUNTRY_CODE = US
    AUTO_SETUP_NET_ETHERNET_ENABLED = 0
```

I'll run a bunch more tests before this weekend. I'm hosting an event during which several dozen folks from around the world will be following along at home while I walk them through the process of making Roon outputs from Rasberry Pi's. Kind-of need this to work from everywhere. LOL. Wish me luck!

Thanks for your help, @Joulinar 
Closing this for now.

dsnyder0pc, 2020-04-09:
I had a friend in Denver try this process to help me verify that we're all good. Unfortunately, seeing exactly the same thing, even with my adjusted waittime and "Google" changes:

## Crash message:

[image]

Here are the changes made from the defaults in dietpi.txt and dietpi-wifi.txt:

[image]

@Joulinar - Your thoughts would be most appreciated. My event is just two days away.  :)
I wonder if this hack would help:
```
CONFIG_CHECK_CONNECTION_IP=127.0.0.1
```
I don't understand why this check has no re-tries and borks the entire install. It's important that this process be resilient and hands-off…at least for my use case; probably others too. Thanks.

dsnyder0pc, 2020-04-09:
Hold on…I may know what's wrong…could be user error (on my part, not Neil's)

Joulinar, 2020-04-09:
@dsnyder0pc 
basically the aim of the connection test is exactly to verify if internet access as well as DNS resolution are working well before going to download software 😃 

Anyway you could completely bypass the test be setting the check values to localhost

```
CONFIG_CHECK_CONNECTION_IP = 127.0.0.1
CONFIG_CHECK_DNS_DOMAIN = localhost
```

dsnyder0pc, 2020-04-09:
No, that was not it. I just had two back-to-back failures using the same configs that worked late last night. Could be more Wi-Fi interference during the day? Still, this needs to be a lot more robust. Suggestions still needed. Thanks.

[image]

dsnyder0pc, 2020-04-09:
I found a place to bump up the retries after line 1169 in `dietpi-globals`:

```
1163         G_RUN_CMD(){
1164
1165                 G_CHECK_ROOT_USER 1
1166
1167                 G_ERROR_HANDLER_COMMAND=$@
1168
1169                 G_ERROR_HANDLER_RETRY=1
1170                 if echo "$@" | grep -q ping; then G_ERROR_HANDLER_RETRY=5; fi
1171                 while (( $G_ERROR_HANDLER_RETRY ))
1172                 do
1173
```

Results look promising so far.

dsnyder0pc, 2020-04-09:
Looks like `G_ERROR_HANDLER_RETRY` is being treated as a boolean. It's not used to specify a number of retries.  :-\

I'll try the localhost thing.

dsnyder0pc, 2020-04-09:
@Joulinar - disabling the connection test (by using localhost) seems to work, although I'm not very satisfied with that as a general solution. The test seems like it's good to have, given the high rate of intermittent failure and lack of retries, I'm out of options at the moment.

I hope you will consider adding a `G_RUN_CMD_RETRY()` function that accepts a number of retries. It seems like that might help a lot in this case…especially if other DietPi users are encountering this issue too.

dsnyder0pc, 2020-04-09:
I'm going with this fix for now. Please let me know if you find a way to retry the pings or run them later in the first-boot process. Thanks again.

https://github.com/dsnyder0pc/DietPi/blob/master/dietpi-setup#L192-L194

Joulinar, 2020-04-09:
the ping check is nothing related to the first boot process. it will be executed as prerequisite every time you try to install software.

MichaIng, 2020-04-15:
@dsnyder0pc 
> I hope you will consider adding a G_RUN_CMD_RETRY() function that accepts a number of retries. It seems like that might help a lot in this case…especially if other DietPi users are encountering this issue too.

Just as you mentioned it, I found the exactly same problem and merged the error handler and command wrapper and error handler and added exactly this option: https://github.com/MichaIng/DietPi/blob/dev/dietpi/func/dietpi-globals#L895
I'll now implement native G_CHECK_CONNECTION and G_CHECK_DNS functions which use dietpi.txt settings for timeout and automated retries. In case of interactive firstrun setup, the error handler allows to retry the whole command anyway and now one can even edit it or open a subshell for investigation (among other options).

dsnyder0pc, 2020-05-08:
Looks like my problem is back:

dsnyder0pc, 2020-05-08:
[image]

Just selecting "Cancel" and typing reboot got things going again. But most of the folks I'm helping with this project will not have a display or keyboard connected to the RPi when they do this. The NanoPi NEO2 does not even have a way to connect a display.

@MichaIng - Would you mind sharing the settings that you've added to dietpi.txt that provide parameters to your new error handlers? Sorry, but it's not obvious to me after a minute or two of looking at your update what those might be. Thanks.

dsnyder0pc, 2020-05-08:
Grr...looks like I need to find a way to bypass the survey with a dietpi.txt setting also.  :-\

[image]

dsnyder0pc, 2020-05-08:
Found it.

```
SURVEY_OPTED_IN=0
```

MichaIng, 2021-06-02:
I mark this as closed again. The survey prompt does not show up when automated install is enabled via dietpi.txt. And without it, you'd need to manually enter username + password to login, hence have a keyboard and display available to select the choice.

closed 2021-06-02 by MichaIng