MichaIng/DietPi #3464: Can't open sudo dietpi-software

IBOYHK, 2020-04-11:
#### Details:
- Date           | Sat 11 Apr 21:38:08 BST 2020
- Bug report     | 8ede33d1-45a2-4245-9297-28824b7379c0
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux DietPi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | ping -c 1 -W 5 one.one.one.one
- Exit code      | 1
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
PING one.one.one.one(one.one.one.one (2606:4700:4700::1001)) 56 data bytes

--- one.one.one.one ping statistics ---
1 packets transmitted, 0 received, 100% packet loss, time 0ms
```

Joulinar, 2020-04-11:
Hi,

I see you have enabled IPv6. Is it really used or could you have it deactivate?

IBOYHK, 2020-04-12:
it just the router DNS address

Joulinar, 2020-04-12:
This is not your router DNS. This is Cloudflare global IPv6 for `one.one.one.one (2606:4700:4700::1001)`

ljford7, 2020-04-15:
I am having the exact same issue as well. I can ping 1.1.1.1 from the network connectivity test, but it always fails when I run dietpi-software.

[image]

Joulinar, 2020-04-15:
Well it's not the same issue as you can not ping the IP where this topic was open for not able to ping the DNS name which might be a problem due to IPv6. 

Pls can run the ping from DietPi OS level and let it run for a minute to see how connectivity is. Furthermore you can change the ping check to another IP address.

ljford7, 2020-04-15:
It is failing every time from the DietPi OS level. I tried google instead of cloudfare and the same failure.

Joulinar, 2020-04-15:
So `ping 1.1.1.1` is failing as well as `ping 8.8.8.8`?

ljford7, 2020-04-15:
Yes (I have two separate pi's running DietPi and they both are failing with ping).

Joulinar, 2020-04-15:
Can you ping any other internet address?

ljford7, 2020-04-15:
Joulinar thanks for your help, but I believe it is a problem with my router blocking icmp requests. I have a EdgeRouter Lite that captures all DNS requests and forces them to use my pihole. Unfortunately it seems to stop icmp (ping) requests from working as well. Too bad the dietpi-software couldn't use another way to validate internet in lieu of ping.

Joulinar, 2020-04-15:
Well ICMP has nothing to do with DNS requests. These are 2 totally different things. 

How about to ping Cloudflare or Google from a different computer in your network like a workstation or laptop? Is this working?

ljford7, 2020-04-15:
I get that, but I think the hairpin NAT is capturing the requests and not letting them go through (my firewall rules are setup correctly from what I have read). 

I can't ping from any computer on my network. I can ping directly from my router and it works. I ran a trace route from my PC and it is stopping at my router.  What is really strange is that dietpi-software used to work just fine (installed wireguard from it), but now I can't get to it due to the ping issue.

Joulinar, 2020-04-15:
Well depends what time you installed software, as the ping check was introduced in one of the last release. Anyway you could completely bypass ping as well as DNS check by changing settings to point to `127.0.0.1` and `localhost`, which in your case is the only chance to continue as long as your router is blocking outgoing ICMP requests.

ljford7, 2020-04-15:
Just wanted to close the loop on this. This was a issue with my router. Finally got it resolved by setting up another masquerade that allows icmp requests through (lots of trial and error, but it works). The masquerade for my pihole was putting the icmp requests in a black hole.

MichaIng, 2020-04-16:
@IBOYHK
If it is still an issue in your case, can you please try to edit `dietpi.txt` and change the setting `CONFIG_CHECK_DNS_DOMAIN` from `one.one.one.one` to `dns9.quad9.net`.

Disabling IPv6 as suggested by Jourlinar is worth a try, there are cases where local networks/routers do not handle those well.

closed 2020-04-16 by IBOYHK