MichaIng/DietPi #3465: Nextcloud install fail with Mariadb issue

befelton, 2020-04-12:
- Date Sun 12 Apr 10:51:27 PDT 2020
- Bug report 13046a4b-489a-49db-a98b-7e4afef48c09
- DietPi version v6.28.0 (MichaIng/master)
- Image creator DietPi Core Team
- Pre-image Raspbian Lite
- SBC device RPi 4 Model B (armv7l) (ID=4)
- Kernel version Linux Glory 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux
- Distro buster (ID=5)
- Command systemctl restart mariadb
- Exit code 1
- Software title DietPi-Software

#### Steps to reproduce:
1. Use dietpi-drive_manager to move user data to external drive
2. Use dietpi-software to install nextcloud

#### Expected behaviour:
It works!

#### Actual behaviour:
It crashes when it cannot restart mariadb

#### Extra details:
I've tried this three times with a new image, trying to find a sequence that would work. It reliably fails, and always in the same way. The install will work if the user data is left in its initial location on the SD card.

#### Additional logs:
Job for mariadb.service failed because the control process exited with error code.
See "systemctl status mariadb.service" and "journalctl -xe" for details.

# systemctl status mariadb.service
● mariadb.service - MariaDB 10.3.22 database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2020-04-12 10:53:55 PDT; 30min a
go
     Docs: man:mysqld(8)
           https://mariadb.com/kb/en/library/systemd/
  Process: 17724 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/run/mysqld (
code=exited, status=0/SUCCESS)
  Process: 17725 ExecStartPre=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION 
(code=exited, status=0/SUCCESS)
  Process: 17728 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery ] && VAR= ||   VAR
=`/usr/bin/galera_recovery`; [ $? -eq 0 ]   && systemctl set-environment _WSREP_START_POSIT
ION=$VAR || exit 1 (code=exited, status=0/SUCCESS)
  Process: 17785 ExecStart=/usr/sbin/mysqld $MYSQLD_OPTS $_WSREP_NEW_CLUSTER $_WSREP_START_
POSITION (code=exited, status=1/FAILURE)
 Main PID: 17785 (code=exited, status=1/FAILURE)
   Status: "MariaDB server is down"

Apr 12 10:53:54 Glory systemd[1]: Starting MariaDB 10.3.22 database server...
Apr 12 10:53:55 Glory mysqld[17785]: 2020-04-12 10:53:55 0 [Note] /usr/sbin/mysqld (mysqld 
10.3.22-MariaDB-0+deb10u1) starting as process 17785 ...
Apr 12 10:53:55 Glory mysqld[17785]: 2020-04-12 10:53:55 0 [Warning] Can't create test file
 /var/lib/mysql/Glory.lower-test
Apr 12 10:53:55 Glory mysqld[17785]: [90B blob data]
Apr 12 10:53:55 Glory mysqld[17785]: 2020-04-12 10:53:55 0 [ERROR] Aborting
Apr 12 10:53:55 Glory systemd[1]: mariadb.service: Main process exi
ted, code=exited, status=1/FAILURE
Apr 12 10:53:55 Glory systemd[1]: mariadb.service: Failed with resu
lt 'exit-code'.
Apr 12 10:53:55 Glory systemd[1]: Failed to start MariaDB 10.3.22 d
atabase server.

Joulinar, 2020-04-12:
Hi,

I guess /mnt/ was not copied correctly to your external drive. Pls can you check if the following directory is existing `/mnt/dietpi_userdata/mysql`

So probably this could be a solution for you as well > https://dietpi.com/phpbb/viewtopic.php?p=22717#p22717

befelton, 2020-04-12:
/mnt/dietpi_userdata/mysql exists, as does /mnt/[my external drive]/dietpi_userdata/mysql
The contents of the two directories seem identical.

When I tried your command, I got the same result you did:

root@Glory:/var/lib# ls -la|grep mysql
lrwxrwxrwx  1 root   root     26 Apr 12 11:46 mysql -> /mnt/dietpi_userdata/mysql

...but shouldn't the above show a link to /mnt/[my external drive]/dietpi_userdata/mysql instead, since I relocated the user files to the external drive before I started the nextcloud install?

Joulinar, 2020-04-12:
you could have a check on `/mnt/`. There you should see that `/dietpi_userdata/` should have been linked to your external drive. 

could you have a look to `/mnt/dietpi_userdata/` who is owning `/mysql`?

root@DietPi3:# ls -la /mnt/dietpi_userdata/
total 36
drwxrwxr-x 9 dietpi   dietpi   4096 Apr 12 22:22 .
drwxr-xr-x 6 root     root     4096 Jan  7 23:35 ..
drwxrwxr-x 2 dietpi   dietpi   4096 Apr 11 12:25 downloads
drwxrwxr-x 2 dietpi   dietpi   4096 Apr 11 12:25 Music
drwxr-xr-x 5 mysql    mysql    4096 Apr 12 22:26 mysql
drwxrwx--- 5 www-data www-data 4096 Apr 12 21:35 nextcloud_data
drwxrwxr-x 2 dietpi   dietpi   4096 Apr 11 12:25 Pictures
drwxr-xr-x 2 www-data www-data 4096 Apr 12 22:22 rpicam
drwxrwxr-x 2 dietpi   dietpi   4096 Apr 11 12:25 Video
root@DietPi3:#

befelton, 2020-04-12:
Confirmed, /mnt/dietpi_userdata is indeed a link to my external drive.
The owner of /mnt/dietpi_userdata/mysql:
drwxr-xr-x 3 mysql  mysql  4096 Apr 12 11:50 mysql

Joulinar, 2020-04-12:
Owner is `mysql`, which is correct. Quite strange that MariaDB is not able to create the test file

`Can't create test file /var/lib/mysql/Glory.lower-test`

Can you have a look if the file still exist? To verify the access to the directory, you could do following to create a demo file that is owned by `mysql user/group`. So it should be similar to what MariaDB is trying during start process.

`sudo -u mysql -g mysql touch /var/lib/mysql/demo.file`

befelton, 2020-04-12:
root@Glory:/# sudo -u mysql -g mysql touch /var/lib/mysql/demo.file
touch: cannot touch '/var/lib/mysql/demo.file': Permission denied

???

Joulinar, 2020-04-12:
that's strange. Can you go to `/mnt/dietpi_userdata/` and do the following `chown mysql:mysql mysql`. Once done try to create the demo file again.

```
sudo -u mysql -g mysql touch /var/lib/mysql/demo.file
sudo -u mysql -g mysql touch /mnt/dietpi_userdata/mysql/demo.file
```

befelton, 2020-04-12:
root@Glory:/mnt/dietpi_userdata# chown mysql:mysql mysql
root@Glory:/# ls -al /mnt/dietpi_userdata/
total 28
drwxrwxr-x 7 dietpi dietpi 4096 Apr 12 10:46 .
drwx------ 4 dietpi dietpi 4096 Apr 12 10:28 ..
drwxrwxr-x 2 dietpi dietpi 4096 Apr 12 10:22 downloads
drwxrwxr-x 2 dietpi dietpi 4096 Apr 12 10:22 Music
drwxr-xr-x 3 mysql  mysql  4096 Apr 12 11:50 mysql
drwxrwxr-x 2 dietpi dietpi 4096 Apr 12 10:22 Pictures
drwxrwxr-x 2 dietpi dietpi 4096 Apr 12 10:22 Video

root@Glory:/# sudo -u mysql -g mysql touch /var/lib/mysql/demo.file
touch: cannot touch '/var/lib/mysql/demo.file': Permission denied

root@Glory:/# sudo -u mysql -g mysql touch /mnt/dietpi_userdata/mysql/demo.file2
touch: cannot touch '/mnt/dietpi_userdata/mysql/demo.file2': Permission denied

Joulinar, 2020-04-12:
what file system format is your external device using? `ext4`?

befelton, 2020-04-13:
Yes, ext4

befelton, 2020-04-13:
The dietpi user can write to /mnt/dietpi_userdata/, and it shows up on the external drive; so it isn't a general failure.

Joulinar, 2020-04-13:
Can you give 777 permission to the mysql folder and try again.

befelton, 2020-04-13:
After the 777:

root@Glory:/# ls -al /mnt/dietpi_userdata/
total 28
drwxrwxr-x 7 dietpi dietpi 4096 Apr 12 10:46 .
drwx------ 4 dietpi dietpi 4096 Apr 12 10:28 ..
drwxrwxr-x 2 dietpi dietpi 4096 Apr 12 17:11 downloads
drwxrwxr-x 2 dietpi dietpi 4096 Apr 12 10:22 Music
drwxrwxrwx 3 mysql  mysql  4096 Apr 12 11:50 mysql
drwxrwxr-x 2 dietpi dietpi 4096 Apr 12 10:22 Pictures
drwxrwxr-x 2 dietpi dietpi 4096 Apr 12 10:22 Video

root@Glory:/# sudo -u mysql -g mysql touch /var/lib/mysql/demo.file
touch: cannot touch '/var/lib/mysql/demo.file': Permission denied

root@Glory:/# sudo -u mysql -g mysql touch /mnt/dietpi_userdata/mysql/demo.file
touch: cannot touch '/mnt/dietpi_userdata/mysql/demo.file': Permission denied

I'm stumped. I'm tempted to fix it with a hammer...but I'll sleep on it first.

Joulinar, 2020-04-13:
This is weird. I did a test on my RPi3 and connected a USB stick, MariaDB was still working fine after transfering `dietpi_userdata`. Can you try to create test folder on `/mnt/dietpi_userdata`

```
mkdir /mnt/dietpi_userdata/demo
chown mysql:mysql /mnt/dietpi_userdata/demo
chmod 755 /mnt/dietpi_userdata/demo
sudo -u mysql -g mysql touch /mnt/dietpi_userdata/demo/demo.file
```

befelton, 2020-04-13:
root@Glory:/# mkdir /mnt/dietpi_userdata/demo
root@Glory:/# chown mysql:mysql /mnt/dietpi_userdata/demo
root@Glory:/# chmod 755 /mnt/dietpi_userdata/demo
root@Glory:/# sudo -u mysql -g mysql touch /mnt/dietpi_userdata/demo/demo.file
touch: cannot touch '/mnt/dietpi_userdata/demo/demo.file': Permission denied
root@Glory:/# ls -al /mnt/dietpi_userdata/
total 32
drwxrwxr-x 8 dietpi dietpi 4096 Apr 13 07:01 .
drwx------ 4 dietpi dietpi 4096 Apr 12 10:28 ..
drwxr-xr-x 2 mysql  mysql  4096 Apr 13 07:01 demo
drwxrwxr-x 2 dietpi dietpi 4096 Apr 12 17:11 downloads
drwxrwxr-x 2 dietpi dietpi 4096 Apr 12 10:22 Music
drwxrwxrwx 3 mysql  mysql  4096 Apr 12 11:50 mysql
drwxrwxr-x 2 dietpi dietpi 4096 Apr 12 10:22 Pictures
drwxrwxr-x 2 dietpi dietpi 4096 Apr 12 10:22 Video

I encountered this problem each time I tried to install Nextcloud after moving the userdata to the external drive. I haven't tried mariadb by itself.

Joulinar, 2020-04-13:
This has nothing to do with MariaDB or NextCloud. This is a simple user permission issue you have on your external device. As you can see, even creating a simple directory will not allow `mysql` user to write to that directory. And this is quite strange. Do you have another external device like USB stick to try how it behaves there? As well you might be able to test different USB Ports on your device. And you could try to create the demo directory somewhere on temp directory that is not located on the external device. Just to see how it behaves.

befelton, 2020-04-13:
Following your recommendations, I found that mysql could write to the root filesystem (on the SD card). I attached a 2nd external drive, formatted it with dietpi-drive_manager, and mysql could write to it. I moved my user data to the 2nd drive, reformatted the 1st external drive, and then mysql could write to it too. I moved the userdata back to external drive 1 and tried to install Nextcloud. Mariadb still crashed: but with a different error this time.

At that point I started over with a fresh dietpi image, and everything is now working as it should. The 1st external drive that I moved the user files to, and to which the mysql user was unable to write, had initially been formatted ext4 on another system. The dietpi user was always able to write to it, before and after the local formatting. I don't know whether there is a bug there, or if it is just a feature that I don't understand.

In any case, thanks very much for all the help, Joulinar!

closed 2020-04-13 by befelton