MichaIng/DietPi #3466: [suggestion] install Wireguard from Buster-Backports, not from Bullseye

GvY85, 2020-04-13:
First of all, thax for DietPi, this is a wonderfull distro so thanks for all the work put in.

Regarding Wireguard: I noticed that Dietpi-Software uses the Bullseye repository  to install Wireguard.
This works fine at the moment but perhaps in the future it will lead to issues and could potentially even lead to a FrankenDebian when the dependencies for Wireguard Bullseye are different than Wireguard Buster.
Since some time Wireguard is now also in Buster-Backports so parhaps now is a good time to switch over froom Bullseye to Buster-Backports?

Nice small fix for 6.29 ;).

Joulinar, 2020-04-14:
just noticed that `buster-backports` is lacking behind `bullseye`

```
Package wireguard-dkms
buster-backports (kernel): fast, modern, secure kernel VPN tunnel (DKMS version)
- 0.0.20200318-1~bpo10+1: all
bullseye (testing) (net): fast, modern, secure kernel VPN tunnel (DKMS version)
- 1.0.20200401-1: all
sid (unstable) (net): fast, modern, secure kernel VPN tunnel (DKMS version)
- 1.0.20200401-1: all
```

https://packages.debian.org/search?suite=all&section=all&arch=any&lang=en&searchon=names&keywords=wireguard-dkms

GvY85, 2020-04-15:
Lol, you could argue that 'way behind' is just one version that will probably be updated in the next 5 days or so. At the time i wrote my suggestion sid+bullseye+backports where all on 318 ;).

In my opinion its bad practice for DietPi to selectively grab packages outside Stable and Backports. If users want them, they can always install them from Sid or (at there own risk) upgrade there whole dist to Bullseye.

Again, there was I time Wireguard wasnt in Backports but now that it is, lets use it :).

GvY85, 2020-04-15:
> just noticed that `buster-backports` is lacking way behind `bullseye`
> 
> ```
> Package wireguard-dkms
> buster-backports (kernel): fast, modern, secure kernel VPN tunnel (DKMS version)
> - 0.0.20200318-1~bpo10+1: all
> bullseye (testing) (net): fast, modern, secure kernel VPN tunnel (DKMS version)
> - 1.0.20200401-1: all
> sid (unstable) (net): fast, modern, secure kernel VPN tunnel (DKMS version)
> - 1.0.20200401-1: all
> ```
> 
> https://packages.debian.org/search?suite=all&section=all&arch=any&lang=en&searchon=names&keywords=wireguard-dkms
> 
> we already have users who get nervous that `bullseye` package is not updated same time like `sid` 😃
> 
> https://dietpi.com/phpbb/viewtopic.php?f=11&t=7228
> 
> what will happen if DietPi will switch to `buster-backports` 😉

MichaIng, 2020-04-16:
@GvY85 
Many thanks for your suggestion. I didn't know that WireGuard has been backported to Buster meanwhile.

Yeah the version will lack behind a bid, but currently from what I see they are identical to Bullseye and Sid, so delay should be a few days only. Big advantage that we can skip adding an additional repo + pin priorities etc. BUT: This only works on non-RPi Buster, since Raspbian has no backports repo and on Stretch I don't see a reason to prefer Buster backports over Bullseye, since in both cases repo + pins must be added and the packages do not pull any other dependencies anyway.

Joulinar, 2020-04-16:
Well on wireguard.com the linked package is `wireguard-dkms`. And there the version are different between `buster-backports` and `bullseye`. 

```
Package wireguard-dkms
buster-backports (kernel): fast, modern, secure kernel VPN tunnel (DKMS version)
0.0.20200318-1~bpo10+1: all
bullseye (testing) (net): fast, modern, secure kernel VPN tunnel (DKMS version)
1.0.20200401-1: all
sid (unstable) (net): fast, modern, secure kernel VPN tunnel (DKMS version)
1.0.20200413-2: all
```

There was as a new version released by Wireguard and it already arrived at `sid`. Let's see how long it will take into `buster-backports` and `bullseye`.

GvY85, 2020-04-17:
> Well on wireguard.com the linked package is `wireguard-dkms`. And there the version is different between `buster-backports` and `bullseye`.
> 
> ```
> Package wireguard-dkms
> buster-backports (kernel): 0.0.20200318-1~bpo10+1
> bullseye (testing) (net): 1.0.20200401-1
> sid (unstable) (net): 1.0.20200413-2
> ```
> 
> There was as a new version released by Wireguard on 13th of April and it already arrived at `sid`. Let's see how long it will take into `buster-backports` and `bullseye`.

If you want bleeding edge, just dit-upgrade to bullseye or sid :)

MichaIng, 2020-04-21:
Jep let's track the time it takes: https://packages.debian.org/wireguard-dkms
Currently still the same:
```
buster-backports (kernel): fast, modern, secure kernel VPN tunnel (DKMS version)
0.0.20200318-1~bpo10+1: all
bullseye (testing) (net): fast, modern, secure kernel VPN tunnel (DKMS version)
1.0.20200401-1: all
sid (unstable) (net): fast, modern, secure kernel VPN tunnel (DKMS version)
1.0.20200413-2: all
```

Joulinar, 2020-05-04:
for the records. `backports` has been updated to same version as `sid` while `bullseye` is still on old version.

```
buster-backports (kernel): fast, modern, secure kernel VPN tunnel (DKMS version)
1.0.20200429-1~bpo10+1: all
bullseye (testing) (net): fast, modern, secure kernel VPN tunnel (DKMS version)
1.0.20200401-1: all
sid (unstable) (net): fast, modern, secure kernel VPN tunnel (DKMS version)
1.0.20200429-2: all
```

as well there is a message at wireguard.com recommending `backports` now 
`>> Users with Debian releases older than Bullseye should enable backports.`

MichaIng, 2020-05-04:
@Joulinar 
Very strange to ship an update to backports before testing suite... However this basically means that there is no reason to use the Bullseye branch on Buster systems. Since there are no Raspbian backports this means:
- Debian Stretch + Raspbian Stretch + Raspbian Buster => Pull WireGuard from Bullseye
- Debian Buster => Pull WireGuard from Buster backports, which is then all that need to be changed

Joulinar, 2020-05-09:
Small update. `bullseye` again ahead of `backports`

```
Package wireguard-dkms
buster-backports (kernel): fast, modern, secure kernel VPN tunnel (DKMS version)
1.0.20200429-1~bpo10+1: all
bullseye (testing) (net): fast, modern, secure kernel VPN tunnel (DKMS version)
1.0.20200429-2: all
sid (unstable) (net): fast, modern, secure kernel VPN tunnel (DKMS version)
1.0.20200429-2: all
```

MichaIng, 2020-05-10:
@Joulinar 
Although not the upstream version, only a Debian/maintainer patch more.

MichaIng, 2020-05-31:
Done: https://github.com/MichaIng/DietPi/commit/d51514a68bbc9efeef36a32d3102546daea26f7b
On Debian Buster, backports are used (nothing to add), on Stretch and Raspbian Buster, Bullseye is still required. Adding Buster backports to Stretch would be theoretically possible as well, but, honestly, much too complicated 😉.

closed 2020-06-30 by MichaIng