MichaIng/DietPi #3467: DietPi-Config | systemd-timesyncd fails

LexiconCode, 2020-04-14:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.28
- Distro version | debian_version 10.3
- Kernel version | Linux DietPi 4.19.0-8-amd64 
- SBC device | VPS

#### Extra details
`NTP mirror : Change the NTP mirror used `
Successfully change to any mirror however systemd-timesyncd still fails. Most likely related to an issue with the VPS however I little guidance to troubleshoot.

Joulinar, 2020-04-14:
Hi,

I'm a little bit lost what your issue now is? You can change NTP server to `ntp.multacom.com` if this is working for you.

LexiconCode, 2020-04-14:
> I'm a little bit lost what your issue now is? You can change NTP server to `ntp.multacom.com` if this is working for you.

Greetings, yes that fixes my original problem. However `one.one.one.one` refusing to resolve is still an issue. The DNS is resolves other in `sudo apt-get update`.

Joulinar, 2020-04-14:
Can you try the following

```
ping -4 -c 1 -W 10 one.one.one.one
ping -c 1 -W 10 1.1.1.1
```

LexiconCode, 2020-04-14:
```
root@DietPi:~# ping -4 -c 1 -W 10 one.one.one.one
PING one.one.one.one (1.1.1.1) 56(84) bytes of data.
64 bytes from one.one.one.one (1.1.1.1): icmp_seq=1 ttl=61 time=0.924 ms

--- one.one.one.one ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 0.924/0.924/0.924/0.000 ms
```

```
root@DietPi:~# ping -c 1 -W 10 1.1.1.1
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.
64 bytes from 1.1.1.1: icmp_seq=1 ttl=61 time=1.03 ms

--- 1.1.1.1 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 1.032/1.032/1.032/0.000 ms

```

Joulinar, 2020-04-14:
so you are able to `ping one.one.one.one` using IPv4. Do you have IPV6 activated? If not needed pls can you disable IPv6 on `dietpi-config` network settings

LexiconCode, 2020-04-14:
Thank you very much @Joulinar that worked.

closed 2020-04-14 by LexiconCode

Joulinar, 2020-04-14:
ok looks like that IPv6 is not fully working on that network. So better to have it deactivated if not needed.