MichaIng/DietPi #3468: USBridge w Raspotify crashes regularly

dac2800, 2020-04-14:
I'm hoping someone can help me. With apologies in advance for the lack of detail (I'm not that code proficient), I am having the following issue. I'm running Sparky/USBridge with a current version of DietPi and Raspotify (both downloaded 2 days ago) through the Spotify Connect app on a current iOS device. Every 20-45 minutes the Sparky/USBridge stops streaming, and ~30 seconds later Spotify no longer recognizes it. Doing a reboot through the Allo web interface caused the device to show up again ~15 minutes later. I reset the Soundcard from usb-dac to usb-dac-1.1 per a thread, but this had no impact. Any thoughts/suggestions would be appreciated.

FYI I reviewed this thread https://github.com/MichaIng/DietPi/issues/2427, but in all honesty don't really understand what it is suggesting.

MichaIng, 2020-04-16:
@dac2800 
Many thanks for your report.

Is this on an RPi or Sparky SBC or oder SBC?
If it's Sparky SBC, can you please paste the output of: `cat /DietPi/uEnv.txt`

dac2800, 2020-04-16:
Thank you for getting back to me. It is a Sparky SBC with a USBridge board, current (as of Sunday) versions of DietPi and Raspotify installed. I am happy to paste the output, but honestly have no idea how to copy it. Currently, I access the Sparky by plugging my monitor and keyboard into it (it is plugged into my network). I also am signed into the Sparky through a browser. Is there a way to access it from my Mac or my iPad to copy the contents of that file?

-David

Joulinar, 2020-04-16:
You could use SSH to login to OS layer

dac2800, 2020-04-16:
My apologies - I don’t know what that means. 

You could use SSH to login to OS layer

Joulinar, 2020-04-16:
SSH is a way/protocol/method to connect to a server/system without physical keyboard and screen attached.

You would need a SSH Client on your computer. This will allow you to connect using SSH protocol. As possible client you could use `Putty`. 

https://www.ssh.com/ssh/putty/mac/

MichaIng, 2020-09-27:
I mark this as closed. Feel free to reopen if issue persists.

closed 2020-09-27 by MichaIng