MichaIng/DietPi #3469: Update Pydio 8 to Pydio Cell

chunjiw, 2020-04-15:
Pydio 8 is now at End of Life. It says [here](https://pydio.com/en/features/pydio-8-vs-pydio-cells) that arm compilation is on the roadmap 2020 Q1. 

Just want to open an issue here to keep track. Thanks for the awesome work!

MichaIng, 2020-04-16:
@chunjiw 
Many thanks for heads up. Hmm they mention ARM 64bit support explicitly, I hope ARMv6/7 will be supported as well. Having webserver and everything included has the huge disadvantage that binaries need to be compiled at all instead of simply downloading html/PHP code to run on any desired webserver. But of course we cannot support officially EOL marked web application software for long.

MichaIng, 2022-01-16:
Pre-compiled binaries are available for `x86_64` only 😞: https://download.pydio.com/pub/cells/release/3.0.3/
Latest dev and nightly builds are however available for `aarch64` as well (since 2022), so probably this is true for next release as well: https://download.pydio.com/pub/cells/nightly/2022-01-16/

Still not sure what to do about ARMv6/7 then, of course compiling can be tested. Since especially on ARMv6 this will likely take a lot of time, we may provide pre-compiled binaries on our own server.

MichaIng, 2022-01-16:
Pydio Cells seems to have an import tool to import from a running Pydio 8 instance: https://pydio.com/en/pydio-cells/why-migrating-pydio-8-cells

[image]

So we can provide migration steps.

chunjiw, 2022-01-16:
Awesome! Thank you!

Joulinar, 2024-01-22:
@MichaIng I guess we need to remove Pydia (ID 48) from our repository as it is not working on Bookworm anymore. Last version has been released 3 years ago. If you agree, I will open PR to remove Pydio.

MichaIng, 2024-01-22:
How does it not work anymore? Does it not support PHP 8.2?
... this one seems to indicate the same: #5171

I wonder whether we should implement Pydio Cells as direct replacement and offer a migration for all users.

Joulinar, 2024-01-22:
based on a forum post, I tested all 3 Webserver. On Apache our old config file is leading to a service failure. For the other 2 web server, I received a HTTP 500.

MichaIng, 2024-01-22:
Oh right, that `php_admin_value` works with `mod_php` only, not with PHP-FPM. So we would need to add a PHP config instead to set "output_buffering Off". However, the internal server error indicates an issue with PHP. Could be checked with
```sh
journalctl -u php8.2-fpm
```
I hope errors are passed through.

Joulinar, 2024-01-23:
there is nothing within `php8.2-fpm` service log, but I found something within Nginx [error.log](https://github.com/MichaIng/DietPi/files/14029457/error.log)