MichaIng/DietPi #3472: Super slow external HDD

praptoumache, 2020-04-17:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
```
#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=28
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
`10.3`
- Kernel version | `uname -a`
`Linux pi.hole 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux`
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
`RPi 4 Model B (armv7l)`
- Power supply used | (EG: 5V 1A RAVpower)
`Original Pi4 USB-C PowerSupply`

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
`Installed pihole and pivpn`
`Installed eeprom and tweaked CPU Perfomance with dietpi-config`
`HDD is freshly partitioned and formatted ext3 using fdisk and mkfs.ext3`
- Was the software title installed freshly or updated/migrated?
`fresh install`
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Expected behaviour
write speed of ext hdd with 60MB/s and higher
- ...

#### Actual behaviour
```
as i was trying to copy files over the network to the hdd with rsync i get write speeds <1mb
mostly just 300kb/s
used the dietpi benchmark tool and get this
```
[image]

Joulinar, 2020-04-17:
Hi,

do you use an external power supply for your HDD?? Any reason why using ext3 and not ext4?

Maybe this link could be interesting for you. It's related to bad performance on RPi4 using USB3 + SSD https://www.raspberrypi.org/forums/viewtopic.php?t=245931

MichaIng, 2020-04-21:
@praptoumache 
Yes, please retry with ext4 if you have no strong reason for ext3. Assure that you use the USB3 port, since AFAIK RPi4 has both, USB2 and USB3. A dedicated power supply is required in every case for stability reasons, although, if it runs, it should then not make performance differences.

Did you use any special mount options?

MichaIng, 2020-05-18:
Marking as closed due to missing reply. Feel free to reopen if required.

closed 2020-05-18 by MichaIng