MichaIng/DietPi #3473: Missing armbian.list in /etc/apt/sources.list.d/ at M3, Fire3, ZeroPi Buster images

StephanStS, 2020-04-17:
The file armbian.list is missing in the buster images of M3, Fire3 and ZeroPi.

- https://dietpi.com/downloads/images/DietPi_NanoPiM3-ARMv8-Buster.7z
- https://dietpi.com/downloads/images/DietPi_NanoPiFire3-ARMv8-Buster.7z
- https://dietpi.com/downloads/images/DietPi_NanoPiZeroPi-ARMv7-Buster.7z

The result is, that the kernel is not updated any more.

I have already generated corrected images and did first, short tests, how can I upload them?

StephanStS, 2020-04-19:
FYI: I did generate the three images (M3, Fire3, ZeroPi) which are downloadable in 11/2019. MichaIng uploaded them to the download area.

Joulinar, 2020-04-19:
So topic can be closed?

StephanStS, 2020-04-19:
Sorry, my description was a bit misunderstandable.
I generated in 11/2019 the images which had the mistake with the missing file.
I actually have generated three new images which need to be uploaded.

Joulinar, 2020-04-19:
@MichaIng 
Your turn :smiley:

MichaIng, 2020-04-21:
@StephanStS 
Found some time today to go through open issues and will upload your new images now.

MichaIng, 2020-04-21:
**Done: https://dietpi.com/downloads/images/**

StephanStS, 2020-05-02:
Tested the downloadable image it with ZeroPi, assume that it is also valid for Fire3 and M3 (not explicitly tested).

closed 2020-05-02 by StephanStS