MichaIng/DietPi #3474: Octoprint installation fails with AttributeError: 'module' object has no attribute '__version__'

zell-mbc, 2020-04-18:
#### Details:                                                                                           
- Date           | Sat 18 Apr 08:03:16 BST 2020                                                         
- Bug report     | e815918b-67eb-4c60-b5fd-d2dbbf6d1db7                                                 
- DietPi version | v6.28.0 (MichaIng/master)                                                            
- Image creator  | DietPi Core Team                                                                     
- Pre-image      | Raspbian Lite                                                                        
- SBC device     | RPi Zero W (armv6l) (ID=1)                                                           
- Kernel version | Linux 3DPRINTER 4.19.97+ #1294 Thu Jan 30 13:10:54 GMT 2020 armv6l GNU/Linux         
- Distro         | buster (ID=5)                                                                        
- Command        | python setup.py install                                                              
- Exit code      | 1                                                                                    
- Software title | DietPi-Software                                                                      

#### Steps to reproduce:                                                                                
                                                             
1. Create brand new DietPi image
2. Wait for all OS updates to finish
3. Run sudo dietpi-software                                                                                          
4. Select OctoPrint from "Software Optimised"
5. Select "Install"                                                                         

#### Expected behaviour:                                                                                
                                                                            
- Octoprint should instal without error                                                                                                   

#### Actual behaviour:                                                                                  
                                                                             
- Installation stops with error below 

#### Extra details:                                                                                     
                                  
- This might be the same issue as described here: https://github.com/OctoPrint/OctoPrint/issues/2468                                                                                              

#### Additional logs:                                                                                   
```                                                                                                     
    dists = self.install_eggs(spec, download, tmpdir)                                                   
  File "/usr/local/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 896, in install_eggs                                                                                                   
    return self.build_and_install(setup_script, setup_base)                                             
  File "/usr/local/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 1164, in build_and_install                                                                                             
    self.run_setup(setup_script, setup_base, args)                                                      
  File "/usr/local/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 1150, in run_setup                                                                                                     
    run_setup(setup_script, args)                                                                       
  File "/usr/local/lib/python2.7/dist-packages/setuptools/sandbox.py", line 253, in run_setup           
    raise                                                                                               
  File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__                                         
    self.gen.throw(type, value, traceback)                                                              
  File "/usr/local/lib/python2.7/dist-packages/setuptools/sandbox.py", line 195, in setup_context       
    yield
  File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__
    self.gen.throw(type, value, traceback)
  File "/usr/local/lib/python2.7/dist-packages/setuptools/sandbox.py", line 166, in save_modules
    saved_exc.resume()
  File "/usr/local/lib/python2.7/dist-packages/setuptools/sandbox.py", line 141, in resume
    six.reraise(type, exc, self._tb)
  File "/usr/local/lib/python2.7/dist-packages/setuptools/sandbox.py", line 154, in save_modules
    yield saved
  File "/usr/local/lib/python2.7/dist-packages/setuptools/sandbox.py", line 195, in setup_context
    yield
  File "/usr/local/lib/python2.7/dist-packages/setuptools/sandbox.py", line 250, in run_setup
    _execfile(setup_script, ns)
  File "/usr/local/lib/python2.7/dist-packages/setuptools/sandbox.py", line 45, in _execfile
    exec(code, globals, locals)
  File "/tmp/easy_install-uuW5EW/MarkupSafe-2.0.0a1/setup.py", line 56, in <module>
    "semantic_version>=2.8,<2.9",
  File "/tmp/easy_install-uuW5EW/MarkupSafe-2.0.0a1/setup.py", line 43, in run_setup
    "werkzeug>=0.16,<0.17",
  File "/usr/local/lib/python2.7/dist-packages/setuptools/__init__.py", line 145, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python2.7/distutils/core.py", line 124, in setup
    dist.parse_config_files()
  File "/usr/local/lib/python2.7/dist-packages/setuptools/dist.py", line 702, in parse_config_files
    ignore_option_errors=ignore_option_errors)
  File "/usr/local/lib/python2.7/dist-packages/setuptools/config.py", line 121, in parse_configuration
    meta.parse()
  File "/usr/local/lib/python2.7/dist-packages/setuptools/config.py", line 426, in parse
    section_parser_method(section_options)
  File "/usr/local/lib/python2.7/dist-packages/setuptools/config.py", line 399, in parse_section
    self[name] = value
  File "/usr/local/lib/python2.7/dist-packages/setuptools/config.py", line 184, in __setitem__
    value = parser(value)
  File "/usr/local/lib/python2.7/dist-packages/setuptools/config.py", line 515, in _parse_version
    version = self._parse_attr(value, self.package_dir)
  File "/usr/local/lib/python2.7/dist-packages/setuptools/config.py", line 350, in _parse_attr
    value = getattr(module, attr_name)
AttributeError: 'module' object has no attribute '__version__'

Joulinar, 2020-04-18:
Hi,

many thanks for your report. Are you sure you linked the correct topic? becasue the linked one is talking about

`AttributeError: 'module' object has no attribute 'Timeout'`

while your issue seems to be

`AttributeError: 'module' object has no attribute '__version__'`

Next to that there is no DietPi own package. As far as I can see, DietPi will clone the Octoprint Git. So it's the latest up-to-date version available. 

https://github.com/MichaIng/DietPi/blob/eed0d2662a545750e65f8af6d95e5727ee04497d/dietpi/dietpi-software#L5586

zell-mbc, 2020-04-18:
No, got the wrong link, I updated my initial post accordingly. Here's the correct one: https://github.com/encode/django-rest-framework/issues/6203 
It's for a different project but given that the error is the same I was wondering if the solution is maybe something similar. A long shot, I know.

But given that the package is pulled down from upstream, I guess I need to check with the Octoprint team anyway?

Joulinar, 2020-04-18:
wait a second. I will play around with install process on my test system. Might be it will work. Will report back soon.

Joulinar, 2020-04-18:
ok it's working that way. Basically 2 simple adjustment'S needed on DietPi scripts

`nano /DietPi/dietpi/dietpi-software`

change the following lines
```
#                       INSTALL_URL_ADDRESS='https://github.com/foosel/OctoPrint.git'
                        INSTALL_URL_ADDRESS='https://github.com/OctoPrint/OctoPrint.git'
```

and 
```
#                       G_RUN_CMD python setup.py install
                        G_RUN_CMD pip install .
```

Than installation should finish successfully. Sevice is running after reboot and web interface is reachable 

```
root@DietPiVM1:~# systemctl status octoprint.service
● octoprint.service - OctoPrint (DietPi)
   Loaded: loaded (/etc/systemd/system/octoprint.service; disabled; vendor preset: enabled)
   Active: active (running) since Sat 2020-04-18 11:29:49 CEST; 4min 56s ago
 Main PID: 620 (octoprint)
    Tasks: 11 (limit: 1171)
   Memory: 131.8M
   CGroup: /system.slice/octoprint.service
           └─620 /usr/bin/python /usr/local/bin/octoprint serve --iknowwhatimdoing
```
 
@MichaIng 
Still I'm lacking how to open PR. Otherwise I would do changes myself 😄

zell-mbc, 2020-04-18:
Cool! I can confirm the installation is working for me as well. The service is up and running. Thank you very much for the quick turn around.

I can't get to the web interface yet, but that's likely because the web server isn't part of the Octoprint install and has to be installed and configured separately. I'll look into this.

closed 2020-04-18 by zell-mbc

Joulinar, 2020-04-18:
No web server is running already. Octoprint has it's own web server. Just have a look to DietPi user guide how to access

https://dietpi.com/phpbb/viewtopic.php?p=7958#p7958

Should be `http://yourIP:5000/`

zell-mbc, 2020-04-18:
Indeed, it's the port I missed. All up and running now.

MichaIng, 2020-04-21:
@zell-mbc @Joulinar 
Hey guys, many thanks for report and solution. Indeed OctoPrint has been made a GitHub organuzation now. While the old link still works and is redirected to the new one, we should use the new one diretly of course. Also the install method has changed, so yes, needs to be changed in our script. I'll open a PR.