MichaIng/DietPi #3475: Systemd-Timesyncd Service Failed

prettiboifb, 2020-04-18:
Hey!

Im new to this thing and I have no clue what i'm doing but I would love some assistance. I had noticed that my CouchPotato & Sonarr isn't loading the Web UI, however Transmission still works and when I try to uninstall either of the two I get an error saying;

Team                                                  - Pre-image: Raspbian Lite                                                         og file content:                                                                   ob for systemd-timesyncd.service failed because of unavailable resources or anotherystem error.                                                                       ee "systemctl status systemd-timesyncd.service" and "journalctl -xe" for details

Also when I ran the command Systemd-Timesyncd Service status I got the below response;

root@DietPiMedia:~# sudo systemctl status systemd-timesyncd
● systemd-timesyncd.service - Network Time Synchronization
   Loaded: loaded (/lib/systemd/system/systemd-timesyncd.service; disabled; vendor preset: enabled)
  Drop-In: /lib/systemd/system/systemd-timesyncd.service.d
           └─disable-with-time-daemon.conf
   Active: failed (Result: resources) since Sat 2020-04-18 13:07:45 BST;
 13min ago
     Docs: man:systemd-timesyncd.service(8)

Apr 18 13:07:45 DietPiMedia systemd[1]: systemd-timesyncd.service: Failed t
o run 'start' task: No space left on device
Apr 18 13:07:45 DietPiMedia systemd[1]: Failed to start Network Time Synchr
onization.
Apr 18 13:07:45 DietPiMedia systemd[1]: systemd-timesyncd.service: Unit ent
ered failed state.
Apr 18 13:07:45 DietPiMedia systemd[1]: systemd-timesyncd.service: Failed w
ith result 'resources'.
Apr 18 13:07:45 DietPiMedia systemd[1]: systemd-timesyncd.service: Service has no hold-off time, scheduling restart.
Apr 18 13:07:45 DietPiMedia systemd[1]: Stopped Network Time Synchronization.
Apr 18 13:07:45 DietPiMedia systemd[1]: systemd-timesyncd.service: Start re
quest repeated too quickly.
Apr 18 13:07:45 DietPiMedia systemd[1]: Failed to start Network Time Synchr
onization.
Apr 18 13:07:45 DietPiMedia systemd[1]: systemd-timesyncd.service: Unit ent
ered failed state.
Apr 18 13:07:45 DietPiMedia systemd[1]: systemd-timesyncd.service: Failed w
ith result 'resources'.
root@DietPiMedia:~#

Does anyone know how to get this fixed?

Joulinar, 2020-04-18:
Hi,

Any thanks for your report.

Looks like your file system is full as you have the following message 

`No space left on device`

Pls can you post output of filling command 

`df -h`

prettiboifb, 2020-04-18:
> df -h

Ok no prob see the below;

root@DietPiMedia:~# df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       7.3G  7.3G     0 100% /
devtmpfs        484M     0  484M   0% /dev
tmpfs           488M  4.0K  488M   1% /dev/shm
tmpfs           488M   14M  475M   3% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           488M     0  488M   0% /sys/fs/cgroup
tmpfs          1023M   12K 1023M   1% /tmp
tmpfs            50M     0   50M   0% /var/log
tmpfs            10M  1.4M  8.7M  14% /DietPi
/dev/mmcblk0p1   42M   24M   18M  58% /boot

prettiboifb, 2020-04-18:
> Hi,
> 
> Any thanks for your report.
> 
> Looks like your file system is full as you have the following message
> 
> `No space left on device`
> 
> Pls can you post output of following command
> 
> `df -h`

Also this is the first time this has happen, the only changes made was the installation of Sonarr yesterday. Prior to the installation Couchpotato and Transmission worked fine. 

However after installing Sonarr and testing my first download, I was trying to figure out where in the web ui I should put the downloaded TV shows that will post process before moving to final TV folder.

What I realized was that after it downloaded two series, the others was aborted as a lack of space, so I realised it was saving to the Pi memory card in the diet-pi_userdata section in the downloads folder.

After much research I think I figured out where in the web ui i should change the directory and direct the downloads. However before I could test, both Sonarr and Couch no longer loaded.

Then the rest is history.. SIGH

Joulinar, 2020-04-18:
Hi,

I guess your are using a 8GB SD card and it's completely used

`/dev/root 7.3G 7.3G 0 100% /`

you could do the following to find the larges files. Be patient, it could take a while

`du -a / | sort -n -r | head -n 20`

prettiboifb, 2020-04-18:
> Hi,
> 
> I guess your are using a 8GB SD card and it's completely used
> 
> `/dev/root 7.3G 7.3G 0 100% /`
> 
> I cleanup would be needed. You could do the following to find the larges files. Be patient, it could take a while
> 
> `du -a / | sort -n -r | head -n 20`

Ok thank you! I will try and let you know how it worked out.

MichaIng, 2020-06-01:
I mark this as closed due to outstanding reply. Feel free to reopen if issue is still present.

closed 2020-06-01 by MichaIng