MichaIng/DietPi #3476: qBittorrent log files not on dietpi-ramlog

GvY85, 2020-04-18:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.28, 6.29 beta
- Distro version | Buster`
- SBC device | amd64

#### Additional Information (if applicable)
- Software title | qBittorrent

#### Steps to reproduce
Install qBittorrent via Dietpi-Software

#### Expected behaviour
Qbittorrent log files should be in dietpi-ramlog to save write cycles on SD or eMMC

#### Actual behaviour
qBittorrent logs are stored in` ~/.local/share/data/qBittorrent/`

#### Extra details
Tried moving the logs to `/var/log` by adding the following in `/home/qbittorrent/.config/qBittorrent/qBittorrent.conf`:

`[Application]
FileLogger\Enabled=true
FileLogger\Age=6
FileLogger\DeleteOld=true
FileLogger\Backup=true
FileLogger\AgeType=1
FileLogger\Path=/var/log
FileLogger\MaxSize=10
`
based on [this post](https://github.com/qbittorrent/qBittorrent/issues/6033) but no change after a `service qbittorrent restart`

question is if this needs to be solved by qBittorrent or if , as a workaround we could include ` ~/.local/share/data/qBittorrent/` in the ramlog?

MichaIng, 2020-04-21:
@GvY85 
Many thanks for your report.

Indeed it would be great to have those logs in RAM. If there is a way to disable file logging completely and instead use system log (systemd-journald/journalctl), this would be even better IMO. I'll have a closer look for v6.30.

MichaIng, 2020-07-15:
No chance to log to stdout or syslog currently: https://github.com/qbittorrent/qBittorrent/issues/10077
`/var/log` RAMlog is best option for now. We'll leave internal rotation option active. If DietPi-RAMlog is disabled, this is reasonable, however it requires to log to a subdir with proper permissions.

WolfganP, 2020-07-15:
I pushed for a similar change with samba last year (documented at https://github.com/MichaIng/DietPi/issues/2396)
@GvY85 just for the sake of convenience on this discussion, check the "writing to SD" monitoring commands I used. Tweaking here and there you can mostly avoid any writing to SD/eMMC.

As @MichaIng said, permissions on /var/log are usually the issue with some apps running on normal users ids, just create a subfolder with proper permissions via sudo inside, redirect your log file in the apps setting and everything should work.

MichaIng, 2020-07-15:
Done: https://github.com/MichaIng/DietPi/commit/30e6371756ac43f13b334d230325096be8853d77
Effective for new installs only.

WolfganP, 2020-07-15:
@MichaIng I don't know of any doc for the .conf file, but I always use https://github.com/qbittorrent/qBittorrent/blob/master/src/base/preferences.h as reference when trying to check something in my headless setup as an starting point, and then... the search box :-)

MichaIng, 2020-07-15:
Jep, I used the application source for this `[Application]` settings. Works somehow but a bid tedious to check and track where/how the values are used in the end, and I wanted to find out about `FileLogger\MaxSize` and `FileLogger\MaxSizeBytes`, the first seems to be deprecated/removed and replaced be the second. Since we need to support Stretch until Buster, we either need to know if the first still works, is e.g. translated somewhere else or otherwise alter this setting based on qbittorrent version. However I didn't add it for now to let the default take place.

WolfganP, 2020-07-15:
What if you include both in the config file? I don't think they will collide/interact in the app settings, and that way you'll support both versions easily (add a note on `Cleanup needed in the future` and that will be it)

MichaIng, 2020-07-15:
You're right, invalid settings are simply ignored. But not needed to put those logging options inside, since it anyway has no effect with RAMdisk and hourly clean (files will not become that large). If someone has RAMdisk disabled, the defaults are probably fine anyway.

Best would be to reduce the default config to a minimum and give a link to a config overview, it it exists. But anyway you can as well configure most things via web UI.

closed 2020-08-31 by MichaIng