MichaIng/DietPi #3477: Sonarr Won't Delete Files

prettiboifb, 2020-04-19:
Hey!

Not sure if anyone can assist wit this issue I'm having. I just recently installed Sonarr but whenever a download is completed in the downloads folder, when it is finish renaming the file the moving it to the completed series, the files are not deleted in the downloads folder.

Is there any permission that needs to be executed so that Sonarr can delete the  files after the post processing process is completed?

MichaIng, 2020-04-21:
@prettiboifb 
Many thanks for your report. Can you please paste the following:
```
ls -al /mnt/dietpi_userdata/downloads
ls -al /mnt/dietpi_userdata/Video
getent passwd sonarr
systemctl car sonarr
```
Remove/anonymize any contained download file/dir names and such of course, I only want to assure the parent and download/movie permissions are correct.

MichaIng, 2020-05-18:
Marking as closed due to missing reply. Feel free to reopen if required.

closed 2020-05-18 by MichaIng