MichaIng/DietPi #3478: RockPi problems

optimaldoe, 2020-04-19:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`  6.9
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version` RockPi
- Kernel version | `uname -a`
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)  RockPi 4Gbytes
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra) eMMC 32 GBytes

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. ...It will not instal anything. The image is on the MMc and can bott: I can selct the cofig and I can selcet the menu for new optimized software. However it fails completely will not and says
"The certificate of 'ssh.dietpi.com' is not trusted! 185.101.93.93:443
2. ...I have tried this withe Raspberry Pi 3b a number of times and it works beatifully with the SD card. So I was chocked to experience this.

#### Expected behaviour

- ...

#### Actual behaviour

- ...DietPi canot install anything

#### Extra details

- ...

Joulinar, 2020-04-19:
Hi,

many thanks for your report. Pls can you check how certificate looks like if your execute the following 

`openssl s_client -showcerts -connect ssh.dietpi.com:443`

MichaIng, 2020-05-18:
Marking as closed due to missing reply. Feel free to reopen if required.

closed 2020-05-18 by MichaIng

optimaldoe, 2020-05-18:
Well Now I have forgotten this issue and have used a Rasbian image instead.

Your efforts are much appreciated Linux can do a lot.
However I still find the configuartion very difficult. I find always a tip, but they often don't work (issue, date, outdated etc)
configuration control and version is a problem, with so many free releases.
Just now I fighting the buster std image on a Raspberry Py Zero. The WiFi is important since there is no ethernet connector. But I just cannot find a solution trying wpa_supplicant but   
debug info:
Failed to add interface wlan0
wlan0: Request to deauthenticate - bssid=00:00:00:00:00:00 pending_bssid=00:00:00:00:00:00 reason=3 state=DISCONNECTED
TDLS: Tear down peers
nl80211: Data frame filter flags=0x0
wlan0: State: DISCONNECTED -> DISCONNECTED
nl80211: Set wlan0 operstate 0->0 (DORMANT)
netlink: Operstate: ifindex=3 linkmode=-1 (no change), operstate=5 (IF_OPER_DORMANT)
EAPOL: External notification - portEnabled=0
EAPOL: External notification - portValid=0
wlan0: WPA: Clear old PMK and PTK
wlan0: Cancelling delayed sched scan
wlan0: Cancelling scan request
wlan0: Cancelling authentication timeout
Off-channel: Clear pending Action frame TX (pending_action_tx=(nil)
nl80211: Data frame filter flags=0x0
Off-channel: Action frame sequence done notification: pending_action_tx=(nil) drv_offchan_tx=1 action_tx_wait_time=0 off_channel_freq=0 roc_waiting_drv_freq=0
Remove interface wlan0 from radio phy0
Remove radio phy0
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211: Remove monitor interface: refcount=0
netlink: Operstate: ifindex=3 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
nl80211: Set mode ifindex 3 iftype 2 (STATION)
nl80211: Unsubscribe mgmt frames handle 0x88573721 (mode change)

MichaIng, 2020-05-20:
@optimaldoe 
Okay so this is not a DietPi issue anymore then. However it depends on how the network in general is set up. Raspbian by default comes with dhcpcd, AFAIK while DietPi uses isc-dhcp-client which integrates better with ifupdown.

A very basic IPv4 WiFi + DHCP setup would look like this:
#### /etc/network/interfaces
```
# Drop-in configs
source interfaces.d/*

# Local
auto lo
iface lo inet loopback

# WiFi
allow-hotplug wlan0
iface wlan0 inet dhcp # Or when dhcpcd is used: "iface wlan0 inet manual"
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
```
#### /etc/wpa_supplicant/wpa_supplicant.conf
```
network={
	ssid="your_SSID"
	scan_ssid=1
	key_mgmt=WPA-PSK
	psk="your_WiFi_password"
}
```
```
ifup wlan0
```