MichaIng/DietPi #3479: Odroid C1 dont boot

biagioDonato, 2020-04-21:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
- Distro version | 
- Kernel version | 
- SBC device | Odroid C1
- Power supply used | 5V 2A
- SDcard used | SanDisk ultra 16GB class 10

#### Additional Information (if applicable)
hash.txh
FILE:	DietPi_OdroidC1-ARMv7-Buster.img
DATE:	Sun Apr  5 16:46:04 CEST 2020
MD5:	bd8f0a9078a2966898088747d739af28
SHA1:	6f23d001a5d051e22caf9cd8109c01ed38202194
SHA256:	c9b5851b6f9c113d7040264aac5620fc2f625c1ca55a8227587dca7a0f0b5a80

When I turn on my odroid C1, monitor remain black.

The blue led status remain on, in C1 manual I found this

> The blue LED  • Is on (solid light) when the bootloader is running
> The blue LED  • Blinks slowly when the kernel is running, like a heartbeat
> The blue LED  • Blinks quickly when the kernel is in panic mode

 
When I flash the img I see only 1 partition in sd card, instead when I flash the stable version of dietpi or odroid official ubuntu the partition in SD are 2. Is possible that the SBC not found the boot partition?

Joulinar, 2020-04-21:
Hi,

many thanks for your report. I guess this forum post fit's your issue as well. You probably can have a look. 

https://dietpi.com/phpbb/viewtopic.php?f=11&t=7461

As far as I can see, there are still challenges with the image for Odroid C1

MichaIng, 2020-04-21:
@biagioDO 
Yes, sadly the issue is known and I lack the time currently to do further research. Basically the Armbian kernel/bootloader combination simply does not boot, and they do not take care about this anymore since Odroid C1 is EOL. There is a fix out there, but it did not work so far. If you have some time for research and tinkering (following the link of Joulinar), I would highly apprechiate. Otherwise I would give it a last try some days later, otherwise revert to another base image with original kernel that should boot.

MichaIng, 2020-08-29:
Latest image is working, marking as closed.

closed 2020-08-29 by MichaIng