MichaIng/DietPi #3480: Question: does installing raspi-config "harm" dietpi in any way?

BerndKohl, 2020-04-21:
Hi there,
I am running DietPi on several Raspberry Pis and I am really really happy with this.

I was looking for a way/tool to backup my SBCs while running and finally found [raspiBackup](https://www.linux-tips-and-tricks.de/de/raspberry/23-pi-erstellt-automatisch-backups-von-sich-selbst-pi-creates-automatic-backups-of-itself/). Basically it uses dd to create an image of the running Raspbian. Those are stored on a NAS being mounted via NFS.

In order to reduce image size (I am using 128GB SD cards) raspiBackup can be configured to use [piShring](https://github.com/Drewsif/PiShrink) to smartly reduce image size (not only compressing them but resizing the image's filesystem beforehand).

This uses raspi-config to resize the filesystem on the initial boot (after a restore from the image on a fresh SD card was made - in case of a hardware/software failure).

And here comes my problem: dietpi does not provide raspi-config.

So I am wondering if simply installing it via apt-get install (no idea if this even works on diepi) would solve the issue or how to mitigate this without losing the possibility of reducing the backup image size.

Any ideas are very much appreciated.

Joulinar, 2020-04-21:
Hi,

yep `apt install raspi-config` will work on DietPi. I just give it a try on my RPi3B+ demo system. On first call you would need to specify a user, who will be used by `raspi-config`. You can go for user `dietpi`. I really just did a small check and changed my host name. This was working. But I'm not sure if there are other side effects. I guess the one who finally could answer it, will be @MichaIng 

so in meantime you could try it out 😄 
Do a full backup of your SD Card and play around with `raspi-config`. If it's not working your could easily restore your SD card.

MichaIng, 2020-04-21:
`raspi-config` does no harm, it doubles some `dietpi-config` features but AFAIK earlier incompatibilities, e.g. where `dietpi-config` needed certain settings to be present, commented when disabled, have been fixed.

BerndKohl, 2020-04-22:
As usual, very good support!

Thank you so much.

closed 2020-04-22 by BerndKohl