MichaIng/DietPi #3482: Koel doesn't install on Pi 1, Pi Zero, Pi 3

C-Fu, 2020-04-21:
#### Details:
- Date           | Tue 21 Apr 18:45:05 +08 2020
- Bug report     | fad200b7-9f69-4aeb-9403-b2196d4d3306
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux pihole-fu 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | Connection test: https://github.com/phanan/koel/archive/.tar.gz
- Exit code      | 8
- Software title | DietPi-Software

#### Steps to reproduce:

1. Install Koel
2. Script tries to get https://github.com/phanan/koel/archive/.tar.gz instead of https://github.com/phanan/koel/archive/v4.2.2.tar.gz or https://github.com/phanan/koel/archive/$version.tar.gz

#### Expected behaviour:

- get v4.2.2.tar.gz

#### Actual behaviour:

- get  .tar.gz which obviously doesn't exist even when these two are in /DietPi/dietpi/dietpi-software
                        local fallback_url='https://github.com/phanan/koel/archive/v4.2.2.tar.gz'
                        Download_Install "https://github.com/phanan/koel/archive/$version.tar.gz"

#### Extra details:

- ...

#### Additional logs:
```
Spider mode enabled. Check if remote file exists.
--2020-04-21 18:41:47--  https://github.com/phanan/koel/archive/.tar.gz
Resolving github.com (github.com)... 192.30.255.113
Connecting to github.com (github.com)|192.30.255.113|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
Remote file does not exist -- broken link!!!
```

Joulinar, 2020-04-21:
Hi,

many thanks for your report. I found the issue. The repository is not reachable anymore. Means that `dietpi-software` will not get correct version

root@DietPi3:~# curl -s https://api.github.com/repos/phanan/koel/releases/latest
{
  "message": "Moved Permanently",
  "url": "https://api.github.com/repositories/47495360/releases/latest",
  "documentation_url": "https://developer.github.com/v3/#http-redirects"
}

So I guess `dietpi-software` needs to be updated to point to `https://api.github.com/repositories/47495360/releases/latest`

https://github.com/MichaIng/DietPi/blob/eed0d2662a545750e65f8af6d95e5727ee04497d/dietpi/dietpi-software#L5369

C-Fu, 2020-04-21:
> Hi,
> 
> many thanks for your report. I found the issue. The repository is not reachable anymore. Means that `dietpi-software` will not get correct version

I see, thanks! I was going mad the past few days, I thought it was me :D

Joulinar, 2020-04-21:
Not sure if you already fixed `dietpi-software` yourself. If not, this maybe helpful to update the URL

`sed -i 's/https\:\/\/api\.github\.com\/repos\/phanan\/koel\/releases\/latest/https\:\/\/api\.github\.com\/repositories\/47495360\/releases\/latest/' /DietPi/dietpi/dietpi-software`

be careful it's just 1 command to be put into 1 command line 😉

MichaIng, 2020-04-21:
@C-Fu @Joulinar 
Many thanks for reporting. I was afraid that GitHub added some redirect for this API in general but I tested some other cases and it worked fine there. I also checked the Koel repo itself, if owner or branch changed, but everything is as it was: https://github.com/koel/koel
No idea why the API has a redirect there, but the easiest solution is to use `curl -sL` instead, so it follows redirects. This is also safer probably since we cannot be sure if this repo ID stays the same.

MichaIng, 2020-04-21:
Fixed with: https://github.com/MichaIng/DietPi/commit/15404f5ee6a333c285aa31c744c30124d94b6437

closed 2020-05-03 by MichaIng