MichaIng/DietPi #3485: Mopidy python2 and Iris

lupa18, 2020-04-22:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
When installing mopidy, you are using python2 and several plugins, like Iris, Spotify and other seems not to be working installing with v2. 

#### Describe the solution you'd like:
Is it possible to install/use mopidy with pyhton3?

lupa18, 2020-04-22:
And also having this warning:
"DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020...."

MichaIng, 2020-04-22:
@lupa18 
Many thanks for your report. Indeed I just checked and we install extensions with `pip` command instead of `pip3`. I see now this is new since Mopidy 3. Hmm, the problem is on ARMv8 we pull the Debian repo packages, since official Mopidy repo does not ship ARMv8 packages, hence we need to use the correct pip/pip3 based on arch and distro version 🤔.

MichaIng, 2020-04-22:
**Fixed with: #3486**

closed 2020-05-03 by MichaIng