MichaIng/DietPi #3487: Transmission download folder without rwx (---) permission

sealteam666, 2020-04-23:
So i am downloading movies to  /mnt/mydisk/transmission but movies are downloading with drwxrwx--- i want it to download with drwxrwxrwx 
Goal is that jellyfin can read new movie download with transmission i am new to dietpi i dont know where to post also i dont permission and writes in linux..please help

Joulinar, 2020-04-23:
Hi,

many thanks for your report. basically you would need to set umask value in `transmission `config file. Do `nano /etc/transmission-daemon/settings.json` and change `umask` value to `0`. Once done you would need to reload transmission service.

https://github.com/transmission/transmission/wiki/Editing-Configuration-Files

sealteam666, 2020-04-24:
Thank you so much man it worked .

closed 2020-04-24 by sealteam666