MichaIng/DietPi #3489: USB Keyboard emulation on RPi0W

danergo, 2020-04-24:
There are a lot of infos around to use RPI0W as a standard USB keyboard via g_hid.
I.e.: https://github.com/raspberrypisig/pizero-usb-hid-keyboard
or: https://www.aidanwoods.com/blog/building-a-wifi-enabled-usb-rubber-ducky

And there are others too (all relates to the same method).

Is there any chance that DietPi truncates this feature so I can not use this by default?

MichaIng, 2020-04-25:
@danergo 
Since DietPi for RPi is based on official Raspbian with it's repo, kernel etc, it should work fine. The dtoverlays and kernel modules are all available. The guides or installers might expect a certain command/tool that is not installed by default on DietPi, e.g. the first link install.sh uses `git`, which you can install via `dietpi-software` or simply `apt install git`.

danergo, 2020-04-26:
@MichaIng: I have reinstalled DietPi at least a dozens of times, and it did not worked.
Then I installed Raspbian Lite and worked for the very first trial.

With DietPi, `/dev/hidg0` is not available after running those commands in the manual.
One thing might be the problem: the original guide needs an rc.local script to work and somewhere I read that the hidg0 init shall be run before user logs in (so it's not good from ssh session).

But DietPi does not have rc.local and I did not tried with systemctl style autorun.
I tried however with DietPi's autorun system but it did not work.

Again: this shall be working with DietPi, but the single essence of the solution (which is bringing up `hidg0`) is not available.

I would still prefer using DietPi vs Raspbian Lite, but currently I don't have any option to see RPi0W's console, as I can only login with ssh over wifi.

MichaIng, 2020-04-26:
@danergo 
There is a script to create the rc.local and make it executable: https://github.com/raspberrypisig/pizero-usb-hid-keyboard/blob/master/enableHIDRCLocal.sh
It will then be called on each boot (independendly from user login) automatically.
So after running the install.sh this should be all done already.
You can check via: `journalctl -u rc-local`
And: `cat /etc/rc.local`

`lsmod` shows "dwc2"? ...

... ah I think I know the issue. Please do:
```
grep -q 'dtoverlay=dwc2' || echo 'dtoverlay=dwc2' | sudo tee -a /DietPi/config.txt
```
You should see the output, showing that the setting has been added and hence was not been present. `config.txt` on DietPi is on a RAMdisk, hence when you add something to /boot/config.txt, it gets overwritten by the RAMdisk on shutdown. So one needs to alter /DietPi/config.txt instead, as mentioned at the top of the file.
Great that we have choosen to remove it so from next version on all files can be edited from /boot, like on Raspbian: https://github.com/MichaIng/DietPi/blob/dev/CHANGELOG.txt#L19

danergo, 2020-04-26:
@MichaIng 

Sounds cool, thank you.
Let me ask one question which a bit unrelated. In the meanwhile as I mentioned I have installed Raspbian Lite, and it works without issues, and I also made it _readonly_.

Now I don't necessarily touch this device again unless I can be sure that DietPi can also work in fully readonly rootfs (this Pi is like a pendrive: can be plugged out from the computer, i.e. power break without correct shutdown procedure).

The procedure I made with Raspbian consisted some steps, like:

- disable swap
- replace log manager
- make fs readonly
- move some files to temp fs
- update random seed file

Now I know DietPi is kind of ligthweight and already use a nice log which does not load the SD card.
I'm also aware of the `dietpi-drive_manager`, but don't know if it is not deprecated, is it still working with the latest buster images, etc.

Could you please confirm this for me? Then I will test this USB HID thing and reply back here with my results.

MichaIng, 2020-04-26:
@danergo 
> I'm also aware of the `dietpi-drive_manager`, but don't know if it is not deprecated, is it still working with the latest buster images, etc.

Our scripts work with all Raspbian/Debian versions up to Bullseye (testing), since Buster is the new stable version and we provide RPi Buster images for a long time, of course they have been tested throughoutly.

Basically you can do everything with DietPi that you can do with Raspbian. DietPi is Raspbian made lightweight and with our own script/tool set on top, but with same APT repo, kernel/firmware/bootloader etc.
We have a RAMlog implementation, but it stores the log file meta data (names, permissions etc) to disk on shutdown and loads them into tmpfs on boot, since many services fail to start if their log files are not present or have wrong permissions.
Generally most DietPi scripts expect R/W access, most will prompt an error if this is not the case. Since You will see the one or the other error messages when going through boot log, but aside of that thinks will work just the same. Actually current beta/dev boot scripts (v6.29) contain only a single script that wants to write a single file to disk, which is only important when switching SDcard or moving root file system to a different drive, which requires R/W then anyway: https://github.com/MichaIng/DietPi/blob/dev/dietpi/preboot#L24

So basically, if you want to try it again with DietPi, I would wait for the next release, where RAMdisk has gone, which removes the need to write tmpfs files back to disk on shutdown, so more compatible with R/O systems. Else I don't see an issue.

danergo, 2020-04-26:
Okay, I'll wait for the next release, thanks.

Shall we close this issue for now and reopen after next version will come out?

MichaIng, 2020-04-26:
@danergo 
We can keep this issue open. It also reminds me of updating our Justboom remote control implementation. I found it using/creating config files which are unused since Raspbian/Debian Stretch.

MichaIng, 2020-09-28:
I mixed up the issues I think, as this had nothing to do with LIRC 😄, which was handled here instead: #3481
I'll close this issue now, feel free to reopen if there are still questions about the keyboard emulation topic or mounting a DietPi system R/O.

closed 2020-09-28 by MichaIng