MichaIng/DietPi #3490: Onboard bluetooth not working at NanoPi Neo2

ZakMcCrack, 2020-04-24:
With a fresh installation it was not possible to enable (onboard) bluetooth

#### Required Information
NanoPi Neo2
Linux DietPi 4.14.0 #82 SMP Fri Dec 8 14:33:14 CST 2017 aarch64
SanDisk ultra

#### Additional Information (if applicable)
fresh installed,

#### Steps to reproduce
installed,
updated / upgraded,
activated bluetooth via dietpi-config -> advanced options -> enable bluetooth
restart

#### Expected behaviour
bluetooth should work

#### Actual behaviour
no bluetooth device can be found
After restart
`systemctl status bluetooth` shows: loaded, but inactive.

```
root@DietPiNeo:~# systemctl status bluetooth
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: inactive (dead)
Condition: start condition failed at Fri 2020-04-24 17:37:38 CEST; 13min ago
     Docs: man:bluetoothd(8)
```

```
root@DietPiNeo:~# systemctl status hciuart
Unit hciuart.service could not be found.
```

```
root@DietPiNeo:~# lsmod
Module                  Size  Used by
bluetooth             331776  0
ir_lirc_codec          16384  0
lirc_dev               16384  1 ir_lirc_codec
sunxi_cir              16384  0
rc_core                32768  4 ir_lirc_codec,lirc_dev,sunxi_cir
snd_soc_simple_card    16384  0
snd_soc_simple_card_utils    16384  1 snd_soc_simple_card
rtc_ds1307             24576  0
g_mass_storage         16384  0
usb_f_mass_storage     36864  1 g_mass_storage
libcomposite           45056  2 g_mass_storage,usb_f_mass_storage
```

```
root@DietPiNeo:~# hcitool dev
Devices:

```

#### Extra details
I tried v6.0 and v6.14 image -> same behaviour

MichaIng, 2020-04-25:
@ZakMcCrack 
Many thanks for your report.

> Condition: start condition failed at Fri 2020-04-24 17:37:38 CEST; 13min ago

This means `/sys/class/bluetooth` is not available. I found a case where this is if it's blocked by rfkill.

Could you ckeck:
```
rfkill list
```
and if you find some Bluetooth-related entry, e.g.:
```
rfkill unblock bluetooth
```

If this doesn't help, probably something in kernel or device tree is missing. You could you test it with the new Buster-based image: https://dietpi.com/downloads/testing/DietPi_NanoPiNEO2-ARMv8-Buster.7z
It is based on Armbian with Linux 5.4. AFAIK, the current "stable" image is based on FriendlyARM image which is not available anymore (they do not provide Debian images anymore, only Ubuntu/FriendlyCore ones) and no kernel/firmware sources that I'm aware of.

StephanStS, 2020-04-26:
On the NEO2 there should be no Bluetooth at all.
See 

- http://wiki.friendlyarm.com/wiki/index.php/NanoPi_NEO2#Hardware_Spec or
- https://www.friendlyarm.com/index.php?route=product/product&product_id=180 

Typically the NanoPi boards which have WiFi have a WiFi/Bluetooth combo chip.
E.g. NEO4 or NEO Air do have Bluetooth. See 

- http://wiki.friendlyarm.com/wiki/index.php/NanoPi_NEO_Air#Hardware_Spec
- http://wiki.friendlyarm.com/wiki/index.php/NanoPi_NEO4#Hardware_Spec

So I assume you will be search for Bluetooth on your NEO2 in vain.  :-(

MichaIng, 2020-04-26:
@StephanStS 
That would be of course an easy explaination 😄. I never heard of an SBC with WiFi-only (not BT combo) chip, but since BT is not mentioned in specs...

The `dietpi-config` entry btw shows up in every case (besides virtual machine), which make sense since one can attach a USB BT adapter. It installs packages, enables services and kernel modules in a generic way and device tree overlays or special setup where known, so all that can succeed even without any BT adapter attached. We could add a hint about that, although it is probably too trivial that you need an adapter before BT can work?

StephanStS, 2020-04-26:
@MichaIng: I did not wanted to mention that the NEO2 has WiFi. It does not have one.

I could not test BT due to a missing BT USB adapter.
At least WiFi via an USB WiFi adpater works.

Joulinar, 2020-04-26:
yup there is no WiFi/ Bluetooth on a NanoPi NEO2  http://nanopi.io/nanopi-neo2.html
This is available on the Plus2 version only http://nanopi.io/nanopi-neo-plus2.html

MichaIng, 2020-04-26:
That's true, just remember this being one major difference between this two boards. So things are then true the other way round: Without onboard WiFi there is no onboard BT in all cases known to me 😉.

ZakMcCrack, 2020-04-27:
Sorry, you're absolutely right. I changed between RPi4, RPi Zero W and NanoPi Neo2 the whole day - all with DietPi on it.
I asked myself: what the hell is different? This might be the reason...

Nevertheless. Great work. Thanks.

closed 2020-04-27 by ZakMcCrack