MichaIng/DietPi #3492: VirtualHere does not install on DietPi

lhrique, 2020-04-28:
#### Details:
- Date           | Tue 28 Apr 16:59:16 -03 2020
- Bug report     | 9f8b81cc-c7ef-40fd-a395-1526d29541c6
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi  (armv6l) (ID=-1)
- Kernel version | Linux DietPi 4.19.66+ #1253 Thu Aug 15 11:37:30 BST 2019 armv6l GNU/Linux
- Distro         | stretch (ID=4)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/perl/libperl5.24_5.24.1-3+deb9u6_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/perl/perl_5.24.1-3+deb9u6_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/perl/perl-base_5.24.1-3+deb9u6_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/perl/perl-modules-5.24_5.24.1-3+deb9u6_all.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/libi/libidn/libidn11_1.33-1+deb9u1_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/e/e2fsprogs/libcom-err2_1.44.5-1+deb10u3_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/c/curl/curl_7.64.0-4+deb10u1_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/c/curl/libcurl4_7.64.0-4+deb10u1_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/c/curl/libcurl3-gnutls_7.52.1-5+deb9u10_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/f/freetype/libfreetype6_2.6.3-3.2+deb9u1_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/g/glib2.0/libglib2.0-0_2.50.3-2+deb9u2_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/libx/libxslt/libxslt1.1_1.1.29-2.1+deb9u2_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/php7.3/php7.3-zip_7.3.14-1~deb10u1_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/php7.3/php7.3-json_7.3.14-1~deb10u1_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/php7.3/php7.3-opcache_7.3.14-1~deb10u1_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/php7.3/php7.3-fpm_7.3.14-1~deb10u1_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/php7.3/php7.3-readline_7.3.14-1~deb10u1_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/php7.3/php7.3-gd_7.3.14-1~deb10u1_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/php7.3/php7.3-xml_7.3.14-1~deb10u1_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/php7.3/php7.3-curl_7.3.14-1~deb10u1_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/php7.3/php7.3-sqlite3_7.3.14-1~deb10u1_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/php7.3/php7.3-mbstring_7.3.14-1~deb10u1_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/php7.3/php7.3-cli_7.3.14-1~deb10u1_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/php7.3/php7.3-common_7.3.14-1~deb10u1_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch https://dtcooper.github.io/raspotify/pool/main/r/raspotify/raspotify_0.14.0~librespot.20200130T014147Z.3672214_armhf.deb  Could not resolve host: dtcooper.github.io
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/s/sudo/sudo_1.8.19p1-2.1+deb9u2_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

Joulinar, 2020-04-28:
Hi,

many thanks for report. Pls can you check your local internet connection and/or DNS server setting because you are not able to resolve host name for `raspbian.raspberrypi.org`. Do you use a local DNS server like PiHole?

lhrique, 2020-04-29:
DNS is working fine. Internet connection is ok too. I do not have PiHole and I can reach raspbian.raspberrypi.org form my router/gateway as well as form Raspebrry Pi itself (see messages below)

`PING mirrordirector.raspbian.org (93.93.128.193): 56 data bytes
64 bytes from 93.93.128.193: icmp_seq=0 ttl=47 time=227.089 ms
64 bytes from 93.93.128.193: icmp_seq=1 ttl=47 time=230.390 ms
64 bytes from 93.93.128.193: icmp_seq=2 ttl=47 time=238.287 ms

--- mirrordirector.raspbian.org ping statistics ---
3 packets transmitted, 3 packets received, 0.0% packet loss
round-trip min/avg/max/stddev = 227.089/231.922/238.287/4.698 ms`

`root@DietPi:~# nslookup raspbian.raspberrypi.org
Server:         10.71.1.1
Address:        10.71.1.1#53

Non-authoritative answer:
raspbian.raspberrypi.org        canonical name = mirrordirector.raspbian.org.
Name:   mirrordirector.raspbian.org
Address: 93.93.128.193

root@DietPi:~# ping raspbian.raspberrypi.org
PING mirrordirector.raspbian.org (93.93.128.193) 56(84) bytes of data.
64 bytes from 93.93.128.193 (93.93.128.193): icmp_seq=1 ttl=46 time=243 ms
64 bytes from 93.93.128.193 (93.93.128.193): icmp_seq=2 ttl=46 time=237 ms
64 bytes from 93.93.128.193 (93.93.128.193): icmp_seq=3 ttl=46 time=235 ms
64 bytes from 93.93.128.193 (93.93.128.193): icmp_seq=4 ttl=46 time=227 ms
64 bytes from 93.93.128.193 (93.93.128.193): icmp_seq=5 ttl=46 time=226 ms
64 bytes from 93.93.128.193 (93.93.128.193): icmp_seq=6 ttl=46 time=227 ms
^C
--- mirrordirector.raspbian.org ping statistics ---
6 packets transmitted, 6 received, 0% packet loss, time 5453ms
`

Joulinar, 2020-04-29:
pls try to run your software installation again. Maybe it was a temp issue to resolve `raspbian.raspberrypi.org`

lhrique, 2020-04-29:
I already did try (several trimes). Same result

Joulinar, 2020-04-29:
and you still get `Could not resolve 'raspbian.raspberrypi.org'` during `apt-get` ??

lhrique, 2020-04-29:
Sometimes I get:

`Err:6 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages
  Could not connect to raspbian.raspberrypi.org:80 (93.93.128.193), connection timed out`

Other times i get:
`Err:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages
  Cannot initiate the connection to raspbian.raspberrypi.org:80 (93.93.128.193). - connect (101: Network is unreachable)`

In both cases, immediately after the error I can ping raspbian.raspberrypi.org. As a matter of fact, during the installation attempt I have another PC in the same subnet pinging raspbian.raspberrypi.org continuously, with no flaws

Joulinar, 2020-04-29:
ok at least the issue with resolving the host name is gone. 

Do you have the possibility to connect your SBC to another network? Could be `ping` is working but you now need to connect on `http port 80` and this might be blocked somewhere in between.

lhrique, 2020-04-29:
I have two internet connections at my gateway. Here it is what I have done:
(1) I have forced traffic through a specifc interface (corresponing to a specific service provider). Running dietpi-software, I have got same results.
(2) I have completely bypassed my gateway and connected the SBC directly to one of the service provider's modem. Running dietpi-software, I have got same results.
(3) I have decided to run apt-get update/upgrade directly from CLI. Success!
(4) I have moved SBC to local subnet (behind gateway) and run dietpi-software once again in order to try to install virtualhere from it. Success!

MichaIng, 2020-04-29:
A test that closer matches APT btw is:
```
wget --spider http://raspbian.raspberrypi.org/raspbian/
```
or
```
curl -IL http://raspbian.raspberrypi.org/raspbian/
```

ping uses ICMP protocol while APT connects via HTTP. wget or curl can be used to test this connection/procotol explicitly.

Joulinar, 2020-04-29:
@lhrique 
So it depends where the SBC is located on your network? Could it be that some http traffic is blocked somewhere?

lhrique, 2020-04-29:
It is not the way I see it. dietpi-software has not been able to update/upgrade even outside local network

Joulinar, 2020-04-29:
Well dietpi is doing nothing else than calling apt-get same way as it would be running on CLI. But I guess @MichaIng could comment on this better than I

MichaIng, 2020-04-29:
@lhrique 
> (3) I have decided to run apt-get update/upgrade directly from CLI. Success!

That is strange, since as Joulinar said, `dietpi-software` calls `apt-get` as well, with some options but nothing that would affect the connection 🤔.

> (4) I have moved SBC to local subnet (behind gateway) and run dietpi-software once again in order to try to install virtualhere from it. Success!

So the initial setup was to have a local network gateway at one ISP and the SBC connected outside the local network directly to the gateway on the other ISP?

lhrique, 2020-04-30:
Gentlemen, I cannot explain why apt-get does not work well when called by dietpi-software. My setup comprises one local network attached to a gateway with two internet connections in load-balance/failover. I have only connected the SBC outside the gateway to try to implement a suggestion you have made. Now, it is back to where it should be (behind the gateway). Yesterday, I flashed dietpi again and started configuration from scratch. Once again, I experienced difficulties using dietpi-software to update/upgrade and install virtualhere -- tasks that I accomplished via CLI.
Anyway, thanks for your support.

MichaIng, 2020-05-01:
@lhrique 
Okay. Running `apt(-get)` update/upgrade/install manually still works fine? Could you please then try the following from console to use our APT function swappers that are also used in dietpi-software:
```
G_AGUP # apt-get update wrapper
G_AGUG # apt-get upgrade wrapper
G_AGI <package> # apt-get install <package> wrapper
```

lhrique, 2020-05-02:
Here it is the result of g_AGUP:

`root@DietPi:~# G_AGUP
[ INFO ] APT update, please wait...
Hit:1 https://dtcooper.github.io/raspotify raspotify InRelease
Hit:2 https://archive.raspberrypi.org/debian buster InRelease
Err:3 http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian buster InRelease
  Could not connect to mirror.ox.ac.uk:80 (163.1.221.67), connection timed out Could not connect to mirror.ox.ac.uk:80 (163.1.160.163), connection timed out
Reading package lists...
W: Failed to fetch http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian/dists/buster/InRelease  Could not connect to mirror.ox.ac.uk:80 (163.1.221.67), connection timed out Could not connect to mirror.ox.ac.uk:80 (163.1.160.163), connection timed out
W: Some index files failed to download. They have been ignored, or old ones used instead.
[  OK  ] G_AGUP
root@DietPi:~# ping  mirror.ox.ac.uk
PING mirror.ox.ac.uk (163.1.221.67) 56(84) bytes of data.
64 bytes from mirror0.it.ox.ac.uk (163.1.221.67): icmp_seq=1 ttl=41 time=201 ms
64 bytes from mirror0.it.ox.ac.uk (163.1.221.67): icmp_seq=2 ttl=41 time=197 ms
64 bytes from mirror0.it.ox.ac.uk (163.1.221.67): icmp_seq=3 ttl=41 time=194 ms
64 bytes from mirror0.it.ox.ac.uk (163.1.221.67): icmp_seq=4 ttl=41 time=196 ms
64 bytes from mirror0.it.ox.ac.uk (163.1.221.67): icmp_seq=5 ttl=41 time=197 ms
`

lhrique, 2020-05-02:
And here it is G_AGUG:

`root@DietPi:~# G_AGUG
[ INFO ] APT upgrade, please wait...
E: Failed to fetch http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian/pool/main/b/base-files/base-files_10.3+rpi1+deb10u3_armhf.deb  Could not connect to mirror.ox.ac.uk:80 (163.1.160.163), connection timed out Could not connect to mirror.ox.ac.uk:80 (163.1.221.67), connection timed out
E: Failed to fetch http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian/pool/main/s/systemd/libsystemd0_241-7~deb10u3+rpi1_armhf.deb  Unable to connect to mirror.ox.ac.uk:http:
E: Failed to fetch http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian/pool/main/s/systemd/systemd_241-7~deb10u3+rpi1_armhf.deb  Unable to connect to mirror.ox.ac.uk:http:
E: Failed to fetch http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian/pool/main/s/systemd/udev_241-7~deb10u3+rpi1_armhf.deb  Unable to connect to mirror.ox.ac.uk:http:
E: Failed to fetch http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian/pool/main/s/systemd/libudev1_241-7~deb10u3+rpi1_armhf.deb  Unable to connect to mirror.ox.ac.uk:http:
E: Failed to fetch http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian/pool/main/s/systemd/systemd-sysv_241-7~deb10u3+rpi1_armhf.deb  Unable to connect to mirror.ox.ac.uk:http:
E: Failed to fetch http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian/pool/main/libi/libidn2/libidn2-0_2.0.5-1+deb10u1_armhf.deb  Unable to connect to mirror.ox.ac.uk:http:
E: Failed to fetch http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian/pool/main/g/gnutls28/libgnutls30_3.6.7-4+deb10u3_armhf.deb  Unable to connect to mirror.ox.ac.uk:http:
E: Failed to fetch http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian/pool/main/e/e2fsprogs/libext2fs2_1.44.5-1+deb10u3_armhf.deb  Unable to connect to mirror.ox.ac.uk:http:
E: Failed to fetch http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian/pool/main/e/e2fsprogs/e2fsprogs_1.44.5-1+deb10u3_armhf.deb  Unable to connect to mirror.ox.ac.uk:http:
E: Failed to fetch http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian/pool/main/e/e2fsprogs/libcom-err2_1.44.5-1+deb10u3_armhf.deb  Unable to connect to mirror.ox.ac.uk:http:
E: Failed to fetch http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian/pool/main/c/curl/curl_7.64.0-4+deb10u1_armhf.deb  Unable to connect to mirror.ox.ac.uk:http:
E: Failed to fetch http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian/pool/main/c/curl/libcurl4_7.64.0-4+deb10u1_armhf.deb  Unable to connect to mirror.ox.ac.uk:http:
E: Failed to fetch http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian/pool/main/e/e2fsprogs/libss2_1.44.5-1+deb10u3_armhf.deb  Unable to connect to mirror.ox.ac.uk:http:
E: Failed to fetch http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian/pool/main/s/sudo/sudo_1.8.27-1+deb10u2_armhf.deb  Unable to connect to mirror.ox.ac.uk:http:
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
[FAILED] G_AGUG
`

lhrique, 2020-05-02:
But, after some minutes, result was different...

`root@DietPi:~# G_AGUP
[ INFO ] APT update, please wait...
Hit:1 https://dtcooper.github.io/raspotify raspotify InRelease
Hit:2 http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian buster InRelease
Hit:3 https://archive.raspberrypi.org/debian buster InRelease
Reading package lists...
[  OK  ] G_AGUP

root@DietPi:~# G_AGUG
[ INFO ] APT upgrade, please wait...
(Reading database ... 21540 files and directories currently installed.)
Preparing to unpack .../base-files_10.3+rpi1+deb10u3_armhf.deb ...
Unpacking base-files (10.3+rpi1+deb10u3) over (10.3+rpi1+deb10u2) ...
Setting up base-files (10.3+rpi1+deb10u3) ...
Installing new version of config file /etc/debian_version ...
(Reading database ... 21540 files and directories currently installed.)
Preparing to unpack .../libsystemd0_241-7~deb10u3+rpi1_armhf.deb ...
Unpacking libsystemd0:armhf (241-7~deb10u3+rpi1) over (241-7~deb10u2+rpi1) ...
Setting up libsystemd0:armhf (241-7~deb10u3+rpi1) ...
(Reading database ... 21540 files and directories currently installed.)
Preparing to unpack .../systemd_241-7~deb10u3+rpi1_armhf.deb ...
Unpacking systemd (241-7~deb10u3+rpi1) over (241-7~deb10u2+rpi1) ...
Preparing to unpack .../udev_241-7~deb10u3+rpi1_armhf.deb ...
Unpacking udev (241-7~deb10u3+rpi1) over (241-7~deb10u2+rpi1) ...
Preparing to unpack .../libudev1_241-7~deb10u3+rpi1_armhf.deb ...
Unpacking libudev1:armhf (241-7~deb10u3+rpi1) over (241-7~deb10u2+rpi1) ...
Setting up libudev1:armhf (241-7~deb10u3+rpi1) ...
Setting up systemd (241-7~deb10u3+rpi1) ...
(Reading database ... 21540 files and directories currently installed.)
Preparing to unpack .../systemd-sysv_241-7~deb10u3+rpi1_armhf.deb ...
Unpacking systemd-sysv (241-7~deb10u3+rpi1) over (241-7~deb10u2+rpi1) ...
Preparing to unpack .../libidn2-0_2.0.5-1+deb10u1_armhf.deb ...
Unpacking libidn2-0:armhf (2.0.5-1+deb10u1) over (2.0.5-1) ...
Setting up libidn2-0:armhf (2.0.5-1+deb10u1) ...
(Reading database ... 21540 files and directories currently installed.)
Preparing to unpack .../libgnutls30_3.6.7-4+deb10u3_armhf.deb ...
Unpacking libgnutls30:armhf (3.6.7-4+deb10u3) over (3.6.7-4) ...
Setting up libgnutls30:armhf (3.6.7-4+deb10u3) ...
(Reading database ... 21540 files and directories currently installed.)
Preparing to unpack .../libext2fs2_1.44.5-1+deb10u3_armhf.deb ...
Unpacking libext2fs2:armhf (1.44.5-1+deb10u3) over (1.44.5-1+deb10u2) ...
Setting up libext2fs2:armhf (1.44.5-1+deb10u3) ...
(Reading database ... 21540 files and directories currently installed.)
Preparing to unpack .../0-e2fsprogs_1.44.5-1+deb10u3_armhf.deb ...
Unpacking e2fsprogs (1.44.5-1+deb10u3) over (1.44.5-1+deb10u2) ...
Preparing to unpack .../1-libcom-err2_1.44.5-1+deb10u3_armhf.deb ...
Unpacking libcom-err2:armhf (1.44.5-1+deb10u3) over (1.44.5-1+deb10u2) ...
Preparing to unpack .../2-curl_7.64.0-4+deb10u1_armhf.deb ...
Unpacking curl (7.64.0-4+deb10u1) over (7.64.0-4) ...
Preparing to unpack .../3-libcurl4_7.64.0-4+deb10u1_armhf.deb ...
Unpacking libcurl4:armhf (7.64.0-4+deb10u1) over (7.64.0-4) ...
Preparing to unpack .../4-libss2_1.44.5-1+deb10u3_armhf.deb ...
Unpacking libss2:armhf (1.44.5-1+deb10u3) over (1.44.5-1+deb10u2) ...
Preparing to unpack .../5-sudo_1.8.27-1+deb10u2_armhf.deb ...
Unpacking sudo (1.8.27-1+deb10u2) over (1.8.27-1+deb10u1) ...
Setting up systemd-sysv (241-7~deb10u3+rpi1) ...
Setting up libcom-err2:armhf (1.44.5-1+deb10u3) ...
Setting up udev (241-7~deb10u3+rpi1) ...
Setting up libss2:armhf (1.44.5-1+deb10u3) ...
Setting up sudo (1.8.27-1+deb10u2) ...
Setting up libcurl4:armhf (7.64.0-4+deb10u1) ...
Setting up curl (7.64.0-4+deb10u1) ...
Setting up e2fsprogs (1.44.5-1+deb10u3) ...
Processing triggers for install-info (6.5.0.dfsg.1-4+b1) ...
Processing triggers for libc-bin (2.28-10+rpi1) ...
Processing triggers for systemd (241-7~deb10u3+rpi1) ...
[  OK  ] G_AGUG
`
Then, I believe we can put the blame on some temporary instability on servers/internet connection

closed 2020-05-02 by lhrique

MichaIng, 2020-05-02:
@lhrique 
> Then, I believe we can put the blame on some temporary instability on servers/internet connection

I think so. However feel free to reopen if you face this issue again.