MichaIng/DietPi #3493: Time Zone Issues

gary2002, 2020-04-29:
- DietPi version
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=28
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version:  10.3
- Kernel version:  Linux xx 4.9.219+ #1 SMP PREEMPT Tue Apr 14 23:21:53 CEST 2020 aarch64 GNU/Linux
- SBC device:  Odroid N2 (aarch64)
- Power supply used:  12V 2A as supplied by vendor
- SDcard used:  32GB mmc as supplied by vendor

#### Additional Information (if applicable)
- Software title:  Scripts such as diet-services; starting SABnzbd

Error messages such as:
/DietPi/dietpi/func/dietpi-globals: line 173: warning: setlocale: LC_ALL: cannot change locale (en_GB.UTF-8): No such file or directory

Line 173 of /Dietpi/dietpi/func/dietpi-globals is as follows (showing GB set):
export LC_ALL='en_GB.UTF-8'

Should this setting be picked up from /etc/default/locale which, on my system, contains
LC_ALL=en_AU.UTF-8
LANG=en_AU.UTF-8
LANGUAGE=en_AU.UTF-8

I changed line 173 of dietpi-globals to en_AU.UTF-8 and the problem has gone - at least for dependent dietpi scripts. I'm trying to see if it resolves the message from SABnzbd.

Many thanks. Gary

MichaIng, 2020-04-29:
@gary2002 
Many thanks for your report.

Note that we explicitely set `en_GB.UTF-8` in all our scripts to be able to reliably scrape localised command outputs. If you set this to `en_AU.UTF-8`, you break this intention and some scripts may fail (e.g. time sync, AFAIK) as they are checking for output that is not printed anymore due to wrong language/locale.

When setting any language via `dietpi-config`, `en_GB.UTF-8` is always compiled as well. So I guess you configured locales manually and removed it by this?

But it does not matter anymore since with v6.29 we switched to `C.UTF-8` as default locale in our scripts. This locale is always present as hardcoded part of the `libc-bin` package: https://packages.debian.org/buster/amd64/libc-bin/filelist
So it is always present and does not need to be enabled/compiled explicitely.

gary2002, 2020-04-29:
Many thanks.

I had set locale (via dietpi-config) to en_AU.UTF-8 but saw an error message in SABnzbd.

I then saw an error when I ran dietpi-services.

So I had a poke around the error message (line 173) of dietpi-globals.

I can change it back - but it was an annoyance that I hadn't seen in previous versions of dietpi.

Thanks! Gary

MichaIng, 2020-04-29:
@gary2002 
Hmm probably SABnzbd has/had a locale configured that was not compiled or such. Not sure if it uses the system locales. You can add/set locales manually but when you do it via `dpkg-reconfigure locales`, select `en_GB.UTF-8` as well, or join the open beta 😃: #3452

closed 2020-05-03 by MichaIng