MichaIng/DietPi #3494: Update to latest version fails

biggestbear19, 2020-04-29:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=24
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
9.11
- Kernel version | `uname -a`
Linux DietPi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
RPi 3 Model B (armv7l)
- Power supply used | (EG: 5V 1A RAVpower)
5V 1,5A
- SDcard used | (EG: SanDisk ultra)
SanDisk ultra 8GB

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
video surveillance package 'motion' installed, running ok for 1+ year
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. sudo dietpi-update
  Update available:                                                                                               │
│  - Current version : v6.24.1                                                                                    │
│  - Latest version  : v6.28.0 
....
press OK
....
	DietPi-Update
─────────────────────────────────────────────────────
 Phase: Installing new DietPi code

######################################################################## 100.0%
[  OK  ] DietPi-Update | Unpacking update archive
[  OK  ] DietPi-Update | Installing new DietPi scripts to RAMdisk
[  OK  ] DietPi-Update | Installing new DietPi system files to disk
[  OK  ] DietPi-Update | Setting execute permissions for DietPi scripts

#### Expected behaviour

- update finish OK

#### Actual behaviour

- [ SUB1 ] DietPi-Set_software > verify_dietpi.txt ()
[FAILED] DietPi-Set_software | Connection test: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
[ INFO ] DietPi-Set_software | If problems persist, please report at https://github.com/MichaIng/DietPi/issues for investigation.

#### Extra details

- ...

Joulinar, 2020-04-29:
Hi,

many thanks for your report. Maybe a temp issue to connect to GitHub. pls can you try following

`wget --spider https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt`

bbsixzz, 2020-05-13:
That worked for me, had same problem.

MichaIng, 2020-05-18:
Marking as closed, indeed a temporary network issue very likely.

closed 2020-05-18 by MichaIng