MichaIng/DietPi #3495: Can’t connect to WiFi

IBOYHK, 2020-04-30:
The first boot can find the WiFi ssid, but the ssid have some weird name to it as “iBoy’s Home” to as “iBoy\xE2\x80\x99s Home” after select it, and entering the passcode and apply the dietpi still not connect to internet

MichaIng, 2020-05-01:
@IBOYHK 
Many thanks for your report. Hmm it seems that the WiFi scan command prints found SSIDs with encoded special characters. Probably we can prevent this or otherwise would need to manually decode before applying it to `wpa_supplicant.conf`.

You should be able to fix it in `/etc/wpa_supplicant/wpa_supplicant.conf` manually for now.

To verify the assumption, could you please run the following from console and print the output here: `iwlist wlan0 scan` (if "0" is the correct WiFi interface index).

IBOYHK, 2020-05-01:
I need more father details

IBOYHK, 2020-05-01:
The pervious of dietpi doesn’t have this erro

MichaIng, 2020-05-01:
@IBOYHK 
Please run the following in console and copy+paste the output of the command here: `iwlist wlan0 scan`

It should verify my assumption, then I can give you a quick fix command and fix it in our code as well.

IBOYHK, 2020-05-01:
wlan0     Scan completed :
          Cell 01 - Address: 48:5D:36:8E:06:00
                    Channel:1
                    Frequency:2.412 GHz (Channel 1)
                    Quality=39/70  Signal level=-71 dBm  
                    Encryption key:on
                    ESSID:"FiOS-FBRFK"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 18 Mb/s
                              24 Mb/s; 36 Mb/s; 54 Mb/s
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 48 Mb/s
                    Mode:Master
                    Extra:tsf=0000000000000000
                    Extra: Last beacon: 80ms ago
                    IE: Unknown: 000A46694F532D464252464B
                    IE: Unknown: 010882848B962430486C
                    IE: Unknown: 030101
                    IE: Unknown: 0706555320010B1E
                    IE: Unknown: 2A0100
                    IE: Unknown: 32040C121860
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    IE: Unknown: 0B0509009E0000
                    IE: Unknown: 46050300000000
                    IE: Unknown: 2D1AAD091BFFFFFF0000000000000000000000000000000000000000
                    IE: Unknown: 3D1601081500000000000000000000000000000000000000
                    IE: Unknown: 7F080400080000000040
                    IE: Unknown: DD860050F204104A0001101044000102103B00010310470010CD71DBAE5F65B3C09B3F543DA9FE925C10210009477265656E576176651023000342485210240001341042000F4731413131363031313130353733391054000800060050F20400011011000E477265656E576176652042485234100800022008103C0001031049000600372A000120
                    IE: Unknown: DD090010180209001C0000
                    IE: Unknown: DD180050F2020101840003A4000027A4000042435E0062322F00
          Cell 02 - Address: F8:E4:FB:2B:91:09
                    Channel:11
                    Frequency:2.462 GHz (Channel 11)
                    Quality=36/70  Signal level=-74 dBm  
                    Encryption key:on
                    ESSID:"BZ336"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 6 Mb/s
                              9 Mb/s; 12 Mb/s; 18 Mb/s
                    Bit Rates:24 Mb/s; 36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=0000000000000000
                    Extra: Last beacon: 80ms ago
                    IE: Unknown: 0005425A333336
                    IE: Unknown: 010882848B960C121824
                    IE: Unknown: 03010B
                    IE: Unknown: 200100
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    IE: Unknown: 2A0100
                    IE: Unknown: 32043048606C
                    IE: Unknown: DD180050F2020101040003A4000027A4000042435E0062322F00
                    IE: Unknown: 2D1A8C131BFFFF000000000000000000000000000000000000000000
                    IE: Unknown: 3D160B080800000000000000000000000000000000000000
                    IE: Unknown: DD0900037F01010000FF7F
                    IE: Unknown: DD0A00037F04010000000000
                    IE: Unknown: 0706555320010B1B
                    IE: Unknown: DDBE0050F204104A0001101044000102103B00010310470010010102020303040405050606070708081021001A416374696F6E74656320456C656374726F6E69637320494E432E10230019576972656C6573732042726F616462616E6420526F75746572102400043030303110420004303030311054000800060050F204000110110019576972656C6573732042726F616462616E6420526F75746572100800020082103C000103104900140024E26002000101600000020001600100020001
          Cell 03 - Address: 40:9F:38:4C:E5:C7
                    Channel:8
                    Frequency:2.447 GHz (Channel 8)
                    Quality=37/70  Signal level=-73 dBm  
                    Encryption key:off
                    ESSID:"Shark_RV-409f384ce5c7"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 6 Mb/s
                              9 Mb/s; 12 Mb/s; 18 Mb/s
                    Bit Rates:24 Mb/s; 36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
[... 411 lines cut ...]
                    Mode:Master
                    Extra:tsf=0000000000000000
                    Extra: Last beacon: 14780ms ago
                    IE: Unknown: 001600000000000000000000000000000000000000000000
                    IE: Unknown: 01088C129824B048606C
                    IE: Unknown: 030101
                    IE: Unknown: 050400030000
                    IE: Unknown: 0706555320010B1E
                    IE: Unknown: 200100
                    IE: Unknown: 23021200
                    IE: Unknown: 2A0100
                    IE: Unknown: 2F0100
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    IE: Unknown: 2D1A3C1916FFFF000000000000000000000000000000000000000000
                    IE: Unknown: 3D1601081100000000000000000000000000000000000000
                    IE: Unknown: 7F080400000000000040
                    IE: Unknown: DD090010180200004C0000
                    IE: Unknown: DD180050F2020101800003A4000027A4000042435E0062322F00
                    IE: Unknown: DD050050F20500
                    IE: Unknown: DD520050F204104A00011010440001021047001022210203040506070809AC3A7ADD39BB1049000600372A0001201054000800060050F2040001101100164449524543542D726F6B752D3134302D354142444130
                    IE: Unknown: DD12506F9A09020200272B030600AE3A7ADD39B9
                    IE: Unknown: DD0D506F9A0A00000601111C440096

MichaIng, 2020-05-02:
@IBOYHK 
Misterious:
```
ESSID:"HP-Print-00-Officejet Pro 8630"
ESSID:"iBoy's Guest "
ESSID:"iBoy\xE2\x80\x99s Home "
```
So it's `iBoy's Home` only that has encoded characters, iBoy's Guest not. Did you use the exact same apostrophe in noth cases? E.g. `'`and ` or ´ might be treated different.

The following should reveal the character btw: `printf "%b" 'iBoy\xE2\x80\x99s Home '`

So far I couldn't find some info about which characters are printed in encoded way. In theory we could use the above decoding on all SSIDs but theoretically it is possible that a string that is meant literally contains a part that would be falsely decoded. So I would prefer to know about the likely very few special characters and their encodings to decode only those specifically.

There is an alternative command, even the more modern/supported one, probably it does not show this encoding:
```
iw dev wlan0 scan | grep SSID
```

IBOYHK, 2020-05-02:
I got this 

 SSID: iBoy\xe2\x80\x99s Home\x20
        SSID: iBoy's Guest\x20
        SSID: iBoy\xe2\x80\x99s Home\x20
        SSID: iBoy's Guest\x20
        SSID: iBoy\xe2\x80\x99s Home\x20
        SSID: iBoy's Guest\x20
        SSID: HP-Print-19-iBoy's Printer\x20
        SSID: FiOS-FBRFK
        SSID: BZ336
        SSID: 
        SSID: HP-Print-00-Officejet Pro 8630
        SSID: FIOS-FBRFK
        SSID: Shark_RV-409f384ce5c7
        SSID: FiOS-PK5U7
        SSID: 
        SSID: \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00

MichaIng, 2020-05-02:
Okay conversion works:
```
root@dietpi:~# echo -e 'iBoy\xe2\x80\x99s Home\x20'
iBoy’s Home
root@dietpi:~# printf '%b' 'iBoy\xe2\x80\x99s Home\x20'
iBoy’s Home 
```
@IBOYHK 
As you can see, you indeed used a non-standard apostrophe here, while `iBoy's Guest` contains a standard apostrophe. This seems to be one of the rare characters which `iwlist` encodes. `iw` btw encodes the trailing white space as well, since otherwise you would not recognise it in the output, compared to `iwlist` here double quotes make them visible.

However, since the SSID now is clear, the following should allow you to connect to WiFi:
```
sed -i '/ssid="/c\ssid="iBoy’s Home "' /etc/wpa_supplicant/wpa_supplicant.conf
sed -i "/aWIFI_SSID[0]='/c\aWIFI_SSID[0]='iBoy’s Home '" /var/lib/dietpi/dietpi-wifi.db
ifdown wlan0
ifup wlan0
```

I'll think about either adding the conversion of this single special character (very specific since 6 hexadecimal digits), or adding a generic hexadecimal code detection `\x[[:xdigit:]][[:xdigit:]]` and leave choice to user whether to decode it or not or enter/edit the SSID manually for correction.

IBOYHK, 2020-05-02:
Is ok I got it fix

MichaIng, 2020-05-03:
I'll keep this open as feature request, as mentioned above.

johannrichard, 2024-05-27:
I recently ran into the same problem (hex-encoded characters in ssid). I think the solution might be relatively simple if the dietpi-config script used printf escaped characters in these cases: `wpa_supplicant.conf` [supports these hex-encoded characters](https://unix.stackexchange.com/questions/679862/wpa-supplicant-conf-escaping-characters) if provided as a ["printf-escaped" string](https://sources.debian.org/src/wpa/2%3A2.10-12/wpa_supplicant/wpa_supplicant.conf/#L848):

```
# ssid: SSID (mandatory); network name in one of the optional formats:
#	- an ASCII string with double quotation
#	- a hex string (two characters per octet of SSID)
#	- a printf-escaped ASCII string P"<escaped string>"
```