MichaIng/DietPi #3496: Git blocking DIETPI?

zarigata, 2020-04-30:
Git blocking DIETPI?
after many trials ive encounter with this...

Warning: The unit file, source configuration file or drop-ins of systemd-timesyncd.service changed on disk. Run 'systemctl daemon-reload' to reload units.
[ INFO ] DietPi-Run_NTPD | NTPD: Waiting for completion of systemd-timesyncd (2/60)
Warning: The unit file, source configuration file or drop-ins of systemd-timesyncd.service changed on disk. Run 'systemctl daemon-reload' to reload units.
[ INFO ] DietPi-Run_NTPD | NTPD: Waiting for completion of systemd-timesyncd (3/60)
Warning: The unit file, source configuration file or drop-ins of systemd-timesyncd.service changed on disk. Run 'systemctl daemon-reload' to reload units.
[ INFO ] DietPi-Run_NTPD | NTPD: Waiting for completion of systemd-timesyncd (4/60)
Warning: The unit file, source configuration file or drop-ins of systemd-timesyncd.service changed on disk. Run 'systemctl daemon-reload' to reload units.
[ INFO ] DietPi-Run_NTPD | NTPD: Waiting for completion of systemd-timesyncd (5/60)
Warning: The unit file, source configuration file or drop-ins of systemd-timesyncd.service changed on disk. Run 'systemctl daemon-reload' to reload units.
[ INFO ] DietPi-Run_NTPD | NTPD: Waiting for completion of systemd-timesyncd (6/60)
Warning: The unit file, source configuration file or drop-ins of systemd-timesyncd.service changed on disk. Run 'systemctl daemon-reload' to reload units.
[ INFO ] DietPi-Run_NTPD | NTPD: Waiting for completion of systemd-timesyncd (7/60)
Warning: The unit file, source configuration file or drop-ins of systemd-timesyncd.service changed on disk. Run 'systemctl daemon-reload' to reload units.
[ INFO ] DietPi-Run_NTPD | NTPD: Waiting for completion of systemd-timesyncd (8/60)
Warning: The unit file, source configuration file or drop-ins of systemd-timesyncd.service changed on disk. Run 'systemctl daemon-reload' to reload units.
[ INFO ] DietPi-Run_NTPD | NTPD: Waiting for completion of systemd-timesyncd (9/60)

Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 https://archive.raspberrypi.org/debian buster InRelease [25.1 kB]
Reading package lists... Done
E: Release file for http://raspbian.raspberrypi.org/raspbian/dists/buster/InRelease is not valid yet (invalid for another 217d 16h 18min 42s). Updates for this repository will not be applied.
E: Release file for https://archive.raspberrypi.org/debian/dists/buster/InRelease is not valid yet (invalid for another 216d 6h 47min 2s). Updates for this repository will not be applied.

Joulinar, 2020-04-30:
Hi,

the INFO message for time sync service could be fixed by running `systemctl daemon-reload`

Regarding the blocking, well this is not a DietPi issue directly. DietPi is not an own OS, it's a Debian. I see you try to connect to `raspberrypi.org `. Therefore I assume you are using a Raspbian Pi. And on RPi, DietPi is build on Raspbian Lite. 

I found a similar issue reported at raspberrypi.org. It was pointing to an incorrect system time. Pls can you check that time sync is working correctly.

https://www.raspberrypi.org/forums/viewtopic.php?t=245121

zarigata, 2020-05-01:
found the issue, the time was correct, but it somehow put in the timezone of en-UK BUT timezone on US... even so, locale should not be issue with update and downloads seeing that only timestamp are the issue

Joulinar, 2020-05-02:
maybe you where redirected to a `raspberrypi.org` mirror that was expecting a different time stamp than set on the system

MichaIng, 2020-05-13:
I'm pretty sure it was the system time:
> (invalid for another 217d 16h 18min 42s)

No timezone lacks 7 months behind 😉.

The warning indicates that there was an update/change for systemd-timesyncd, I remember there was indeed a systemd update some days ago. Perhaps that somehow messed with the system time 🤔.

MichaIng, 2020-05-21:
Marking as closed. Feel free to reopen if required.

closed 2020-05-21 by MichaIng