MichaIng/DietPi #3498: A puzzle about the nano PI m3 serial port

hanfengcan, 2020-05-02:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.29.0_dev
- Distro version | 10.3
- Kernel version | Linux DietPi 4.14.141-s5p6818 #7 SMP Mon Sep 2 07:10:36 CEST 2019 aarch64 GNU/Linux
- SBC device |NanoPi M3
- Power supply used | notebook USB
- SDcard used | SanDisk microSD 8G

#### My puzzle
I am not sure that is appropriate to ask questions here

I am new for Linux-arm, I has a nanopi m3, and found this excellent system. After some effort, the system is working and working very well. 

But, the serialport do not work. I found the ttySAC0 and ttySAC1 in /dev/. And the offical document
said that is only uart3 and uart4 can be use.

I alse try the tty1~tty4. Have no effect too.

so, is it some thing wrong in config, or I made a mistake

this is the nanopi m3 GPIO map
[image]

this is the dietpi env
[image]

MichaIng, 2020-05-02:
@hanfengcan 
Many thanks for your report/question.

I just checked the docs: http://wiki.friendlyarm.com/wiki/index.php/Serial_port
So indeed ttySAC3 + ttySAC4 should be available.

> tty1~tty4

Those are locale console devices, used for login with keyboard and monitor, no serial devices indeed.

I checked the image for some dtoverlay or similar to enable the serial devices, but sadly couldn't find some. But actually the main dtb file contains definitions for UART3 and UART4 🤔. The original image is EOL from Armbian side, so from experience they will not help to debug enable this feature: https://www.armbian.com/nanopi-m3/

However, probably the device nodes for the serial devices are different on Armbian. Do you want to use them as serial console or for other things?

hanfengcan, 2020-05-02:
Thank you for your replay. I wanna to use the serial devices to control other device.

Has any other way to rebuild pass the git for https://github.com/friendlyarm

I am not good at Linux, but I maybe a try to fix this problem

MichaIng, 2020-05-02:
@hanfengcan 
One thing you could try is:
```
dietpi-config
# Goto Avanced Options > Serial/UART
# Disable all serial consoles, ttySAC0 and ttySAC1 should be enabled by default
# Try to use UART3 or UART4 as of the GPIO map
```
Probably the Armbian image reveals those UART devices at different dev nodes compared to the default/official image. The serial console/TTY must be disabled before the device itself can be used with different purpose. Change is not high but worth to give it a try IMO.

hanfengcan, 2020-05-03:
@MichaIng , Unfortunately, it didn't work. I going to try the offical image, and test my code.😅

closed 2020-05-03 by hanfengcan

hanfengcan, 2020-05-04:
@MichaIng , Hi, the offical image is work.

[image]

[image]

but, I like DietPi better than offical.😂

MichaIng, 2020-05-04:
@hanfengcan 
This was the FriendlyCore image, right? Means one from here: https://drive.google.com/drive/folders/1P-Vky3I785gZtqqEeSHPXR-mxzT84CH1

hanfengcan, 2020-05-05:
yes

MichaIng, 2020-05-07:
@StephanStS
Probably you tried to connect a serial console or device as well?

I would have expected the Armbian kernel+dtb to be based on the official one from FriendlyARM since it is legacy version as well, but obviously some features are not enabled or some requirement or setting for it is missing.

StephanStS, 2020-05-08:
@MichaIng , @hanfengcan: I never used the serial port.
As I see in http://wiki.friendlyarm.com/wiki/index.php/NanoPi_M3, there is also the debug port (UART0) shown below the GPIO port table. 
Possibly ttySAC0 is the debug port and ttySAC3 and ttySAC4 may be the UART3 and UART4.
Maybe it could be an option to test UART0 whether this corresponds to ttySAC0 to get a step forward.
But I am not aware how then to get UART3 and UART4 running.

hanfengcan, 2020-05-08:
@StephanStS thanks

MichaIng, 2020-05-08:
Debug port at ttySAC0 should definitely work, as this is explicitly used for kernel/boot messages (`/boot/boot.cmd`). But not sure what this "debug UART" is capable of besides what it is intended for (debugging failing boot).

Just searched through Armbian forum but could not find something about UART on NanoPi M3 or T3/Fire3. The dts file btw contains uart1 uart2 uart3 uart4 strings, but probably those are somehow disabled. If the issue is the same with NanoPi Fire3, we could ask on Armbian forum about it and have it probably fixed or useful information for all those boards.

StephanStS, 2020-05-09:
@MichaIng: Is this the same issue as #3514?
First of all I can generate a "fresh" image for M3/Fire3 based on Armbian [20.02.7](https://docs.armbian.com/Release_Changelog/). 
Base would be: https://dl.armbian.com/nanopim3/ and https://dl.armbian.com/nanopifire3/. But they are based on kernel 4.14.174 and issue #3514 deals with kernel 5.4.
Is this an option to wait for a fix for #3514 and then try to adopt this to this issue?

hanfengcan, 2020-05-09:
@MichaIng I fresh the `https://dl.armbian.com/nanopim3/archive/Armbian_20.02.7_Nanopim3_buster_legacy_4.14.174_minimal.7z` before. It has SCA0 and SCA1 only.

StephanStS, 2020-05-09:
@hanfengcan: I also generated quickly a DietPi image of the Armbian 20.02.7 (kernel version 4.14.174 ). It also only shows /dev/ttySAC0 and /dev/ttySAC1.
I could do a test with a Armbian image for Fire3 and convert this to DietPi, checking whether it runs on the M3. Then we could wait for Armbian using kernel version 5.4 on the s5p6818 cpu and start with a DietPi image based on this...

MichaIng, 2020-05-10:
@StephanStS 
On NanoPi NEO there are device tree overlays shipped with the Armbian image to enable UART1-4 but on their NanoPi M3 image this is indeed not the case. If someone has some time and knowledge about reading into the M3 base device tree file, we could derive which part is missing/wrong and could create own overlays actually.

Armbian will high likely not fix or change anything about the M3 anymore, but only update the kernel/firmware packages in a generic way to newer Linux versions. So without us doing some own research and fixes, there will be non, I'm afraid.