MichaIng/DietPi #3499: Nextcloud installation fails

harmlessdrudge, 2020-05-02:
### Creating a bug report/issue

#### Required Information
- DietPi version  6.23
- Distro version DietPi_RPi-ARMv6-Buster
- Kernel version 
- SBC device RPi3
- Power supply used Pi original
- SDcard used Sandisk Ultra 8Gb

#### Additional Information (if applicable)
- Software title Nextcloud, Nextcloud Talk
- Was the software title installed freshly or updated/migrated? Fresh install
- Can this issue be replicated on a fresh installation of DietPi? Yes
- Bug report ID didn't record ID, sorry, uploaded a few (c.12) hours ago

#### Steps to reproduce
1. Use dietpi-software to install Nextcloud and Nextcloud Talk

#### Expected behaviour
Successful installation

#### Actual behaviour
Installation fails with error about inability to find an app called spreed

#### Extra details
I've used DietPi to run Pi-Hole successfully and hoped it would be as simple with Nextcloud. I switched to Ubuntu 20.04 and will try Nextcloud using that. I've searched here and in forum on spreed and not found anything useful; apologies if I've missed something.

Joulinar, 2020-05-02:
Hi

Many thanks for your report. Installation of NextCloud is quite easy on DietPi. Pls can you past exactly the error you are facing and we could try to help.

Furthermore you are on an older version of DietPi. I would make sense to update to latest version 6.28 for better support possibilities.

MichaIng, 2020-05-02:
`spreed` is the old name of the Nextcloud Talk app, but it is still the name that one must use to install it via cmd: https://apps.nextcloud.com/apps/spreed (see URL and download links)

The exact command that failed is: `ncc app:install spreed`
Probably this was simply a temporary server issue. I'll retry on VM now.

MichaIng, 2020-05-02:
Very strange, first I ran into exactly the same issue, tried with "talk" but indeed "spreed" is the correct app name:
```
[  OK  ] DietPi-Software | systemctl start mariadb
[  OK  ] DietPi-Software | systemctl start redis-server
[  OK  ] DietPi-Software | ncc maintenance:mode --off
[FAILED] DietPi-Software | ncc app:install spreed
[FAILED] DietPi-Software | ncc app:install spreed
[ INFO ] DietPi-Software | Executing alternative command: ncc app:install talk
Error: Could not download app talk
[FAILED] Alternative command execution | An issue has occurred
Press any key to return to error handler menu...
```
Opened new great subshell feature of DietPi v6.29 to test/try some other commands:
```
2020-05-02 15:24:38 root@VM-Buster:/tmp/DietPi-Software# ncc app:install apporder
apporder installed
apporder enabled
2020-05-02 15:25:17 root@VM-Buster:/tmp/DietPi-Software# ncc app:install spreed
PHP Fatal error:  Cannot declare class OCA\Talk\Migration\Version2000Date20170707093535, because the name is already in use in /var/www/nextcloud/apps/spreed/lib/Migration/Version2000Date20170707093535.php on line 127
2020-05-02 15:26:37 root@VM-Buster:/tmp/DietPi-Software# apt update
Hit:1 https://deb.debian.org/debian buster InRelease
Hit:2 https://deb.debian.org/debian buster-updates InRelease
Hit:3 https://deb.debian.org/debian-security buster/updates InRelease
Hit:4 https://deb.debian.org/debian buster-backports InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
2020-05-02 15:26:53 root@VM-Buster:/tmp/DietPi-Software# ncc app:update apporder
2020-05-02 15:27:19 root@VM-Buster:/tmp/DietPi-Software# ncc app:update spreed
2020-05-02 15:27:34 root@VM-Buster:/tmp/DietPi-Software# ncc app:update --all
2020-05-02 15:27:38 root@VM-Buster:/tmp/DietPi-Software# ncc app:enable spreed
spreed enabled
2020-05-02 15:28:04 root@VM-Buster:/tmp/DietPi-Software# ncc app:install spreed
spreed already installed
2020-05-02 15:28:42 root@VM-Buster:/tmp/DietPi-Software# ncc app:install spreed && echo yes
spreed already installed
```
So now it is installed and enabled as it should. A bid strange that it first failed to download, then obviously downloaded successfully but threw an error that looks like the first attempts already created some database entries.

No idea why it failed in the first place. What we definitely should do is checking for `* already installed` and in case exit with success, since in such case the ncc command exits with error code for some reason. I'll suggest a patch to change this to NC, which would be consistent with `ncc app:enable spreed` succeeding even if the app is enabled already.

For now I think it's temporary NC server issues. I'll retry later.

MichaIng, 2020-05-02:
Change as suggested above has been added + some minor config fixes: https://github.com/MichaIng/DietPi/commit/39d77a6c91eedba7c04f5177bf469775419ecc62

MichaIng, 2020-05-02:
Okay, while some hours ago it still failed, now it succeeded on first try. But the error I recognised above returns 255 exit code:
```
PHP Fatal error:  Cannot declare class OCA\Talk\Migration\Version2000Date20170707093535, because the name is already in use in /var/www/nextcloud/apps/spreed/lib/Migration/Version2000Date20170707093535.php on line 127
```
A bug it appears: https://help.nextcloud.com/t/cannot-install-talk-on-fresh-nextcloud-18-install/68613
Needs to be reported to Talk devs.
When ignoring the exit code, all goes on fine. So a second special exit code exclusion we need to add 😠.

closed 2020-05-03 by MichaIng