MichaIng/DietPi #3500: Image | Odroid C4

MichaIng, 2020-05-02:
### Creating an image request

#### Formal device information
- Device name | Odroid C2
- Official product URL | https://wiki.odroid.com/odroid-c4/odroid-c4
- GitHub resource URL (if available) | https://github.com/hardkernel
- Image download URL |
  - https://wiki.odroid.com/odroid-c4/os_images/os_images (official, Ubuntu + Android only)
  - https://www.armbian.com/odroid-c4/ (Armbian, WIP)
  - https://forum.odroid.com/viewforum.php?f=203 (No Debian image by Meveric so far)
  - https://oph.mdrjr.net/meveric/images/ (all Meveric images to compare)

#### Is the SBC officially supported by the Debian installer?
- No

#### If not, is a reliable 3rd party Debian image available for this SBC?
- Armbian so far

#### If not, are there install instructions for Debian available?

- For Ubuntu only so far, however build info could be derived: https://wiki.odroid.com/odroid-c4/software

#### Vote for this image on FeatHub: https://feathub.com/MichaIng/DietPi/

MichaIng, 2020-05-17:
Meveric created a first testing image: https://oph.mdrjr.net/meveric/images/Buster/

MichaIng, 2020-05-17:
Initial support has been added: https://github.com/MichaIng/DietPi/commit/5996c9d5a9ab8c34dbd0509a27f5e8958cc90d6e

MDAR, 2020-05-29:
I would like to try the C4 with the view to moving from the C2.

If I can do anything to help, please let me know.

FYI, I have the following hardware that I can test on the devices.

ODroid 7 inch touch screen & speakers
ODroid 3.5 inch touch screen
ODroid RTC
Various USB sound cards
USB LAN dongles (dual LAN is quite important to the lighting setups I deploy)

MichaIng, 2020-05-29:
@MDAR 
I'll create a first image soon, would be great if you could test some hardware on it as well, especially since we could add the setup/install steps for those to `dietpi-config` then as well.

MDAR, 2020-05-29:
> @MDAR
> I'll create a first image soon, would be great if you could test some hardware on it as well, especially since we could add the setup/install steps for those to `dietpi-config` then as well.

@MichaIng 
No problem at all.

Please let me know when you are ready and what you would like me to test.

It would be great to offer multi LAN configuration or dual HDMI & LCD display options, but we can look at that later.

MichaIng, 2020-05-31:
#### Image up for testing: https://dietpi.com/downloads/images/testing/DietPi_OdroidC4-ARMv8-Buster.7z
**Linux version**: 4.9.218
**Debian version**: 10/Buster
**DietPi version**: 6.30.0-6.31.0 dev branch, will be updated to stable once v6.31 has been released!
**Credits**: Pre-image by Meveric: https://forum.odroid.com/viewtopic.php?f=203&t=38729
- Consider donating Meveric for his great work on the Odroid images and repository, providing GPU drivers, Kodi and other packages optimised and accelerated for Odroid: https://www.paypal.com/donate/?token=LMCbvftcq_bx4nbNVSB4PX-351eQSd2F7m1x-KoJ_zBE6boqH5cuKvvpwq4Jbs3h3oq0m0&country.x=US&locale.x=US

Please help us to estimate the stability state and give us a short report here if boot and first run setup went fine or if you faced any issue.

MichaIng, 2020-06-08:
**New image up, hopefully this boots 😉: https://dietpi.com/downloads/images/testing/DietPi_OdroidC4-ARMv8-Buster.7z**

cretudorin, 2020-06-15:
> **New image up, hopefully this boots wink: https://dietpi.com/downloads/images/testing/DietPi_OdroidC4-ARMv8-Buster.7z**

Tested with multiple sdcards, it doesn't boot

MichaIng, 2020-06-15:
Many thanks for testing. Then I'm afraid someone with the device needs to create the image as there seems to be some important package config step skipped when doing this on VM within container. What I found was missing gzipped kernel image creation and missing u-boot flash, which I did manually, respectively by removing the device checks from the package scripts. A pain those exist which makes automated image creation much more complicated...

Steps to do so:
1. Download Meverics current C4 image: https://oph.mdrjr.net/meveric/images/Buster/Debian-Buster64-1.0~RC1-20200505-C4.img.xz
    - Login: root:odroid
2. Update all packages and reboot to assure that it is not an issue with the latest version: `apt update; apt full-upgrade; reboot`
3. Run dev version of DietPi-PREP:
    ```
    bash -c "$(curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/dev/PREP_SYSTEM_FOR_DIETPI.sh)"
    ```
    - Choose as well `dev` branch, required to allow Odroid C4 device detection by DietPi.
4. `poweroff`
5. Plug SDcard into another DietPi system and run DietPi-Imager there:
    ```
    bash -c "$(curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/dev/.meta/dietpi-imager)"
    ```
    - Choose "Drive" as source and select the plugged SDcard.
    - The root partition is always the last one 😉.
    - Choose `DietPi_OdroidC4-ARMv8-Buster` for image name.
6. When done you should have an 7zip archive, ready to share.
7. Assure that the SDcard (still) boots fine by plugging it back into the Odroid C4.

cretudorin, 2020-06-15:
Flashed the Meverics image, it's no longer booting after `apt full-upgrade`

`Failed to process /etc/kernel/postrm.d at /var/lib/dpkg/info/linux-image-4.9.218+.postrm line 291.
dpkg: warning: old linux-image-4.9.218+ package post-removal script subprocess returned error exit status 1
dpkg: trying script from the new package instead ...`

[Full log here](https://gist.githubusercontent.com/cretudorin/b522653c4b2ce51e0beb860cfbbfbd26/raw/ccdd484986ba772aca3bfeca243dded4c7895f1c/odroid-c4.log)

MichaIng, 2020-06-15:
Ah yes I remember this `/var/lib/dpkg/info/linux-image-4.9.218+.postrm` where a bug is inside.

`/etc/kernel/postrm.d/clean_uInitrd` contains a faulty last line. To fix the uninstall replace
```
[ -f /boot/uInitrd-$version ] && rm /boot/uInitrd-$version
```
with
```
[ ! -f /boot/uInitrd-$version ] || rm /boot/uInitrd-$version
```
Interestingly the new version does not contain this file anymore at all.

This this is/was a postrm script, the old kernel files were removed already while the new ones have not been created yet. So you rebooted without any kernel in place.

I have the following idea:
- There is a Odroid N2 u-boot package installed on Meverics C4 image, but it is not present in the C4 repository (no u-boot available there currently).
- Since the installer checks some /proc content before flashing this u-boot, I flashed it manually.
- However I compared the parts of the image file where the u-boot binary would be flashed to, and the two images differ. Not sure how this can happen but it looks like the installed u-boot package indeed contains the wrong u-boot.
- I'll dd the original u-boot over to our image, lets see if this allows boot.

cretudorin, 2020-06-15:
I've followed you instructions and successfully created a working image.

There were two issues:  

- ` Error log:  E: Package 'u-boot' has no installation candidate `

For this one, just skipped the installation of `u-boot`

-  `[FAILED] DietPi-Software | Free space check: path=/ | available=132 MiB | required=500 MiB
[FAILED] DietPi-Software | Install aborted due to insufficient free space
` it didn't do an automatic disk resize.

I've done it manually with `dietpi-drive_manager` and then the minimal installation was successful.

Here is the [image](https://drive.google.com/uc?id=1L4AMNVPvDrqCbdiQYzOVQoi6mmmzquCE&export=download)

MichaIng, 2020-06-15:
Interesting, in my case APT just succeeded with the "old" u-boot package staying installed. However it seems to be wrong anyway. But I'll leave it inside DietPi-PREP since I guess Meveric will add it to the repo by times.

> it didn't do an automatic disk resize.

Yeah somehow `dietpi-fs_partition_resize.service` is not enabled on the image, same as `dietpi-firstboot` 🤔. Did you somehow boot it one time after DietPi-PREP has finished?

cretudorin, 2020-06-15:
Yes, I did had some issues with the sdcards. On the first one fsck failed even though everything seemed to be fine. So should I try again without booting after `DietPi-PREP` ?

MichaIng, 2020-06-15:
I'll create one. I also want to verify that it indeed was just the faulty u-boot image. I recognised that my image had a broken superblock as well and 7zip (from Windows) is not able to open the root partition. I guess u-boot was too large or I did something wrong when flashing it, breaking parts of the partition table or such, not sure.

cretudorin, 2020-06-15:
Well, I'm really confused right now. For some reason I had to flash the same image 3 times because the first 2 times didn't work. I've encountered this issue also with CoreElec so maybe there is also some hardware glitch. Also it finally worked after disconnecting everything from it (HDMI, Ethernet, wifi stick, UART)

MDAR, 2020-06-15:
It looks like you chaps are putting in some serious effort.

I'm sorry I've been too busy to join in.

I'll try to find some time tomorrow to give it a try.

@MichaIng 
Thanks for the tip regarding creating an image, I'll try that as soon as I can.

MichaIng, 2020-06-15:
Btw guys, hold on for now. Meveric seems to be busy with the kernel package as well currently. I just wanted to create the new image and the kernel package has been gone while a new linux-libc-dev package appeared in the repo.

For reference I asked for clarification about the u-boot package: https://forum.odroid.com/viewtopic.php?p=296240#p296240

And about the kernel, were looking for the `linux-image-arm64-odroid-c4` package:
- https://oph.mdrjr.net/meveric/pool/c4/l/
- Our mirror: http://fuzon.co.uk/meveric/pool/c4/l/

MichaIng, 2020-06-18:
Updated image with new kernel package and identified two reasons why the old did not boot:
1. The installed `u-boot` package is only a placeholder for what will be released for C4 explicitly at a later date and must not be flashed.
2. I gzipped the kernel manually as this was skipped when booting on VM. However I now saw that the C4 kernel comes already in gzipped version and must only be copied.

Hence the new image should finally boot.

MDAR, 2020-08-18:
Hi

I've just recieved and powered up my lovely new C4.

Just one small question, is there a way to activate the HDMI output for the standard terminal?

(I'm adding a desktop to the device so I can address and issue with QLCplus, but after that it would be nice to remove the desktop and go back to a termnal.

MichaIng, 2020-08-18:
It should be enabled by default, i.e. getty@tty1.service is enabled, which means on tty1 you get a login prompt and as long as there is nothing else attached tty1 should be mapped to the HDMI screen.

A touchscreen with speakers? Check `/boot/boot.ini` if you see some HDMI/DVI switch setting to toggle HDMI sound output. Check `dietpi-config` > `Audio Options` or `aplay -l` if the HDMI sound device is listed.

MDAR, 2020-08-18:
>A touchscreen with speaker?

Sure, why not 😉 

https://www.odroid.co.uk/odroid-accessories/odroid-displays/odroid-vu7-plus

With these
https://www.odroid.co.uk/odroid-accessories/odroid-sound?product_id=936

(or a pair of headphones in the 3.5mm socket)

>It should be enabled by default

There might have been a power issue, so I'll try again when I'm done with compiling QLCplus for the Arm7_64

MDAR, 2020-08-18:
> It should be enabled by default, i.e. [getty@tty1.service](mailto:getty@tty1.service) is enabled, which means on tty1 you get a login prompt and as long as there is nothing else attached tty1 should be mapped to the HDMI screen.
> 
> A touchscreen with speakers? Check `/boot/boot.ini` if you see some HDMI/DVI switch setting to toggle HDMI sound output. Check `dietpi-config` > `Audio Options` or `aplay -l` if the HDMI sound device is listed.

Arrr...

Looks like you have set VU7 support disabled to true by default, I'll change that to false and reboot

MDAR, 2020-08-18:
Here's a curious one for you

I've removed MATE desktop and rebooted (Althogh I did forget to change the default login) and it gives me a fully operational MATE desktop when it powers up.

BUT the MATE desktop software isn't shown in the Uninstall list anymore...

MichaIng, 2020-08-18:
> I've removed MATE desktop and rebooted (Althogh I did forget to change the default login) and it gives me a fully operational MATE desktop when it powers up.

Lol indeed, seems we changed mate-desktop-environment-extra to mate-desktop-environment-core on install but forgot to as well change this on uninstall. So the following should purge MATE:
```
apt-mark auto mate-desktop-environment-core
apt autopurge
```
Fixed with: https://github.com/MichaIng/DietPi/commit/832fc3cdce0f2d77dcacd05d417342740b957926

> Looks like you have set VU7 support disabled to true by default

Ah sorry, that boot.ini does not come from us, we that what we get from the Meveric image. I did not yet find time to create an own one. Although disabling VU7 by default matches our XU4 boot.ini: https://github.com/MichaIng/DietPi/blob/master/boot_xu4.ini#L27-L28

> TTY1 is now on the HDMI, but no Chromium. just a load of normal boot up text

You mean kiosk mode, right? Since enabling it in dietpi-autostart (when installed) leads to autologin and xinit start, you should see at least an error message if anything fails.

MDAR, 2020-08-18:
>So the following should purge MATE:

Yep, that's cleared it.

>Ah sorry, that boot.ini does not come from us
Right ho.... Tell me what I can do to help ....

>Since enabling it in dietpi-autostart (when installed) leads to autologin and xinit start

I'm not sure what I've done, but I now have a fully operational terminal via the HDMI.

Would be useful to you if I paste the boot.ini contents? (as I have them)

MichaIng, 2020-08-18:
> I'm not sure what I've done, but I now have a fully operational terminal via the HDMI.

Great to hear.

> Would be useful to you if I paste the boot.ini contents? (as I have them)

Of course I can check in in our image as well, but not bad to have it laid out here. Actually the defaults are IMO quite reasonable. I'd probably only style it a bid different, similar as with XU4 one. Earlier or later it would be great to make the settings available via `dietpi-config` 🤔.

MDAR, 2020-08-19:
Thanks for you hard work @MichaIng 

I've found some odd things in the chromium-autostart.sh file and I've taken the liberty of adding a Sleep option, to account for the time it take for a local webserver to boot up (In this case an instance of openHAB2)

This is what my C4 files look like now

Chromium-Autostart.sh (including the debugging echo comands)

```
#!/bin/bash
# Autostart run script for Kiosk mode, based on @AYapejian https://github.com/MichaIng/DietPi/issues/1737#issue-318697621
# - Please see /root/.chromium-browser.init (and /etc/chromium.d/custom_flags) for additional egl/gl init options

# Command line switches https://peter.sh/experiments/chromium-command-line-switches/
# --test-type gets rid of some of the chromium warnings that you may or may not care about in kiosk on a LAN
# --pull-to-refresh=1
# --ash-host-window-bounds="400,300"

# Resolution to use for kiosk mode, should ideally match current system resolution
RES_X=$(sed -n '/^[[:blank:]]*SOFTWARE_CHROMIUM_RES_X=/{s/^[^=]*=//p;q}' /boot/dietpi.txt)
RES_Y=$(sed -n '/^[[:blank:]]*SOFTWARE_CHROMIUM_RES_Y=/{s/^[^=]*=//p;q}' /boot/dietpi.txt)
DELAY=$(sed -n '/^[[:blank:]]*SOFTWARE_CHROMIUM_DELAY=/{s/^[^=]*=//p;q}' /boot/dietpi.txt)

 CHROMIUM_OPTS="--kiosk --test-type --window-size=$RES_X,$RES_Y --start-fullscreen --start-maximized --window-position=0,0"
# CHROMIUM_OPTS="--kiosk --test-type --window-size=$RES_X,$RES_Y --start-fullscreen --start-maximized --window-position=0,0"

# echo $CHROMIUM_OPTS

# If you want tablet mode, uncomment the next line.
# CHROMIUM_OPTS=$CHROMIUM_OPTS+" --force-tablet-mode --tablet-ui --no-sandbox --noerrordialogs"
# echo $CHROMIUM_OPTS

# Add URL for first run:
URL=$(sed -n '/^[[:blank:]]*SOFTWARE_CHROMIUM_AUTOSTART_URL=/{s/^[^=]*=//p;q}' /boot/dietpi.txt)
CHROMIUM_OPTS=$CHROMIUM_OPTS+" --homepage $URL"

# CHROMIUM_OPTS=$CHROMIUM_OPTS+" --app $URL"

# echo $CHROMIUM_OPTS

# Find absolute filepath location of Chromium binary.
FP_CHROMIUM=$(command -v chromium)
#if [[ ! $FP_CHROMIUM ]]; then

# Assume RPi
#FP_CHROMIUM="$(command -v chromium-browser)"

#fi
# echo $FP_CHROMIUM

echo Will atempt to run xinit with $FP_CHROMIUM and these options $CHROMIUM_OPTS

echo Waiting  $DELAY

sleep $DELAY # wait for server / machine to start up

xinit $FP_CHROMIUM $CHROMIUM_OPTS

```

And here's the slightly alterned dietpi.txt file for completness

```
# IMPORTANT:
# - This is intended for advanced users, unless you know what you are doing, do not edit this file. Please use the DietPi programs instead.
# - Do not remove uncommented lines, as the items are scraped by DietPi programs, on demand.

#------------------------------------------------------------------------------------------------------
# D I E T - P I
# DietPi-Automation settings, applied on first boot of DietPi only, ONCE!
[... 223 lines cut ...]
SOFTWARE_CHROMIUM_DELAY=30s
SOFTWARE_CHROMIUM_RES_X=2048
SOFTWARE_CHROMIUM_RES_Y=1200
SOFTWARE_CHROMIUM_AUTOSTART_URL=http://127.0.0.1:8080

# Home Assistant
# - Optional Python build dependencies and modules, possibly required for certain HA components
#	Space separated list (no quotation!), will be installed together with Home Assistant automatically, if present
SOFTWARE_HOMEASSISTANT_APT_DEPS=
#	Add Python modules with version string at best, e.g.: firstModule==1.2.3 secondModule==4.5.6
SOFTWARE_HOMEASSISTANT_PIP_DEPS=

#------------------------------------------------------------------------------------------------------
# D I E T - P I
# Dev settings
#------------------------------------------------------------------------------------------------------
DEV_GITBRANCH=dev
DEV_GITOWNER=MichaIng

#------------------------------------------------------------------------------------------------------
# D I E T - P I
# Settings, automatically added by dietpi-update
#------------------------------------------------------------------------------------------------------
AUTO_SETUP_ACCEPT_LICENSE=0
```

MDAR, 2020-08-19:
Would you like a copy of the working image? (Complete with openHAB and Node-RED)

MDAR, 2020-08-19:
Curiously... this error keeps apprearing when running Chroumium in kiosk mode

```
ERROR:object_proxy.cc(632)] Failed to call method: org.freedesktop.NetworkManager.GetDevices: object_path= /org/freedesktop/NetworkManager: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.NetworkManager was not provided by any .service files
libpng warning: iCCP: known incorrect sRGB profile
```

MDAR, 2020-08-19:
>(FYI, the 7inch odroid HDMI touch screen is working well with Mate desktop, but no sound from it's speakers)

FYI

Appears to be sound from the 3.5mm headphone socket, just not from the speakers.

I'll chat to Odroid and see what needs to be done

MichaIng, 2020-08-19:
> org.freedesktop.NetworkManager.GetDevices

Whatever its trying here, NetworkManager is not installed on DietPi, plain `ifupdown` `/etc/network/interfaces` is used to setup the network interfaces. A webserver is not required to for Chromium and network interfaces are assured to be available as long as network time sync is enabled (which delays service starts + autologin), else you should simply see some regular "website not reachable" page in Chromium and not an error on startup 🤔. However, glad it's sorted.

MDAR, 2020-08-20:
> However, glad it's sorted.

It was something to do with the openHAB2 landing page, once I navigated away from that the error messages stopped coming up in the terminal.

Expect ones about the RGB profiles, but it doesn't seem to be affecting the performance, so I'll ignore them.

closed 2020-08-31 by MichaIng