MichaIng/DietPi #3505: dietpi-update fails to set files/file permissions

PeterLacknase, 2020-05-04:
#### Required Information
- DietPi version | `cat /boot/dietpi/.version`

G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=29
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`

buster

- Kernel version | `uname -a`

Linux hostname 4.19.97+ #1294 Thu Jan 30 13:10:54 GMT 2020 armv6l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)

rpi3b+ (as well two rpi zero w - it's all the same)

- Power supply used | (EG: 5V 1A RAVpower)

5V / 2A dedicated

- SDcard used | (EG: SanDisk ultra)

SanDisk Extreme 64GB

#### Additional Information (if applicable)
- Can this issue be replicated on a fresh installation of DietPi?

not sure

#### Steps to reproduce

1. sudo dietpi-update
2. reboot
3. login (via ssh)

#### Expected behaviour

1. the actual file(s) should exist (or be touched)
2. the permissions should be "right" (depends on the files, but .network definitely needs 666)
(maybe: 3. ownership should be right)

#### Actual behaviour

error message: 

/boot/dietpi/func/dietpi-banner: line 222: /run/dietpi/.dietpi_motd: no permission
/boot/dietpi/func/obtain_network_details: line 143: /run/dietpi/.network: no permission

#### Extra details

Nothing to declare ;-)

Joulinar, 2020-05-04:
Many thanks for your report. What upset you use to perform the activities?

PeterLacknase, 2020-05-04:
update: 

1. `sudo -s`
2. `dietpi-update`

login:

1. `ssh -vvv localuser@machine`

Joulinar, 2020-05-04:
So you are using an own created user and not `root` or `dietpi`?

PeterLacknase, 2020-05-04:
For updates: root

Joulinar, 2020-05-04:
But not for login after reboot, correct?

PeterLacknase, 2020-05-04:
correct

MichaIng, 2020-05-04:
@PeterLacknase 
Many thanks for reporting. Strange, the `.network` permissions should have been set by: https://github.com/MichaIng/DietPi/blob/master/dietpi/patch_file#L98
From next reboot on, it should be created with correct permissions on boot: https://github.com/MichaIng/DietPi/blob/master/dietpi/boot#L60-L63

I'll check. Meanwhile for MOTD, this fixes it: https://github.com/MichaIng/DietPi/commit/31747d2256d5ee049a73ef714c675f72024e2233

Joulinar, 2020-05-04:
@MichaIng 
nope files are created with `read` only. But it should be `write` as well as it was on v6.28, ins't it?

```
dietpi@DietPiVM3:~$ ls -la /run/dietpi/
total 8
drwxr-xr-x  2 root root  80 May  4 11:43 .
drwxr-xr-x 15 root root 460 May  4 11:43 ..
-rw-r--r--  1  666 root  51 May  4 11:42 .network
-rw-r--r--  1 root root   2 May  4 11:43 .timesync_exit_status
dietpi@DietPiVM3:~$
```

PeterLacknase, 2020-05-04:
sure

Joulinar, 2020-05-04:
@MichaIng 
I tried `dev` branch and `/run/dietpi` has now `777` but files inside created still with read only for `non-root` user

```
root@DietPiVM1:/run/dietpi# ls -la /run|grep die
drwxrwxrwx  2 root root  100 May  4 12:01 dietpi
root@DietPiVM1:/run/dietpi# ls -la /run/dietpi/
total 12
drwxrwxrwx  2 root root  100 May  4 12:01 .
drwxr-xr-x 16 root root  480 May  4 12:01 ..
-rw-r--r--  1 root root 1649 May  4 12:01 .dietpi_motd
-rw-r--r--  1  666 root   51 May  4 12:01 .network
-rw-r--r--  1 root root    2 May  4 12:01 .timesync_exit_status
root@DietPiVM1:/run/dietpi#
```

MichaIng, 2020-05-04:
@Joulinar @PeterLacknase 
Dammit, chown was used where chmod was wanted... Just fixed it: https://github.com/MichaIng/DietPi/commit/8fd97bc60bce31e14af964504077318479a0f43b

Too few people (including me) use non-root users regularly it seems, so it slipped through the long beta phase.

Joulinar, 2020-05-04:
> Too few people (including me) use non-root users regularly it seems, so it slipped through the long beta phase.

jup was thinking the same already. Will change me testing approach next time.

MichaIng, 2020-05-04:
I'll wait a few hours, just in case something else appears, and then release v6.29.3 as hotfix. It is nothing urgent but simply ugly.

Joulinar, 2020-05-04:
what about the `.dietpi_motd` ??

MichaIng, 2020-05-04:
@Joulinar 
MOTD is fixed with: https://github.com/MichaIng/DietPi/commit/31747d2256d5ee049a73ef714c675f72024e2233

Joulinar, 2020-05-04:
ok will do some test on `dev` once I'm back home

PeterLacknase, 2020-05-04:
@all:

many thanks!

Joulinar, 2020-05-04:
@MichaIng 
ok did a test on my RPi3B+ with 2 users. Seems ok so far

MichaIng, 2020-05-07:
I also pushed fix via MOTD hack 😉. But Let's see the fstab bug that may require a hotfix release anyway.

Joulinar, 2020-05-07:
yes there are more thinks that would need attention next to `fstab`

closed 2020-05-10 by MichaIng