MichaIng/DietPi #3507: DietPi-Software | Allo Web Gui - Upgrade from 6.28 to 6.29 => Query & PD0 Exceptions

Heroldgray, 2020-05-04:
#### Required Information
- DietPi version: 6.29
- Distro version: 9.12
- Kernel version: Linux USBridge 3.10.38
- SBC model: Sparky SBC (armv7l)

#### Steps to reproduce
- DietPi update from 6.28 to 6.29
- Reboot 
- Allo Web Gui login 
- Exception web page displayed (query and PD0) see below
- Everything else is OK , only issue is with ALLO Web GUI ( PHP / SQL database access ?) 

Thank's for you work
Regards

ALLO web GUI Exception Page (ecerpt) 

**(2/2) QueryExceptioncould not find driver (SQL: select * from `users` where `email` = admin@allo.com limit 1)**
in Connection.php (line 647)
at Connection->runQueryCallback('select * from `users` where `email` = ? limit 1', array('admin@allo.com'), object(Closure))in Connection.php (line 607)
at Connection->run('select * from `users` where `email` = ? limit 1', array('admin@allo.com'), object(Closure))in Connection.php (line 326)
.
=============== cuted here

**(1/2) PDOExceptioncould not find driver**
in Connector.php (line 68)
at PDO->__construct('mysql:host=localhost;port=22;dbname=allo_db', 'allo_db', 'dietpi', array(0, 2, 0, false, false))in Connector.php (line 68)
at Connector->createPdoConnection('mysql:host=localhost;port=22;dbname=allo_db', 'allo_db', 'dietpi', array(0, 2, 0, false, false))in Connector.php (line 44)
at Connector->createConnection('mysql:host=localhost;port=22;dbname=allo_db', array('driver' => 'mysql', 'host' => 'localhost', 'port' => '22', 'database' => 'allo_db', 'username' => 'allo_db', 'password' => 'dietpi', 'unix_socket' => '', 'charset' => 'utf8mb4', 'collation' => 'utf8mb4_unicode_ci', 'prefix' => '', 'strict' => true, 'engine' => null, 'name' => 'mysql'), array(0, 2, 0, false, false))in MySqlConnector.php (line 24)
at MySqlConnector->connect(array('driver' => 'mysql', 'host' => 'localhost', 'port' => '22', 'database' => 'allo_db', 'username' => 'allo_db', 'password' => 'dietpi', 'unix_socket' => '', 'charset' => 'utf8mb4', 'collation' => 'utf8mb4_unicode_ci', 'prefix' => '', 'strict' => true, 'engine' => null, 'name' => 'mysql'))in ConnectionFactory.php (line 183)
at ConnectionFactory->Illuminate\Database\Connectors\{closure}()
at call_user_func(object(Closure))in Connection.php (line 883)
at Connection->getPdo()in Connection.php (line 904)
at Connection->getReadPdo()in Connection.php (line 392)

=============== cuted here

Joulinar, 2020-05-05:
Hi,

many thanks for the report. Looks like SQL driver is missing. What DB as well as PHP version you are running?

Heroldgray, 2020-05-05:
Hi,
Running mysqld and both php7.0 & php7.3 (mysql / php required for Allo Web GUI).
I presume missing SQL adapter due to some mismatch in php config after 6.28 to 6.29 upgrade.

Joulinar, 2020-05-05:
usually php7.3 should be sufficient tu run Allo. pls can you check if `php7.3-mysql` package is available?

`dpkg -l php7.3-mysql`

as well you could force re-installation as follow

```
apt update
apt --reinstall install php7.3-mysql
reboot
```

Heroldgray, 2020-05-06:
Thank's @Joulinar, i've reinstalled php7.3-mysql but no joy, same error.
Before dietpi upgrade Allo Web GUI was probably using php 7.0.

Something wrong in php.ini  ?

> usually php7.3 should be sufficient tu run Allo. pls can you check if `php7.3-mysql` package is available? `dpkg -l php7.3-mysql`
```
root@USBridge:~# dpkg -l php7.0-mysql
dpkg-query: no packages found matching php7.0-mysql

root@USBridge:~# dpkg -l php7.3-mysql
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                        Version            Architecture       Description
+++-===========================-==================-==================-============================================================
ii  php7.3-mysql                7.3.17-1+0~2020041 armhf              MySQL module for PHP

```

Joulinar, 2020-05-06:
can you have a look which PHP version is in use.

`cat /etc/lighttpd/conf-enabled/15-fastcgi-php.conf`

Heroldgray, 2020-05-06:
PHP 7.0 in use

```
root@USBridge:~# cat /etc/lighttpd/conf-enabled/15-fastcgi-php.conf
# -*- depends: fastcgi -*-
# /usr/share/doc/lighttpd/fastcgi.txt.gz
# http://redmine.lighttpd.net/projects/lighttpd/wiki/Docs:ConfigurationOptions#mod_fastcgi-fastcgi

## Start an FastCGI server using php-fpm
fastcgi.server += ( ".php" =>
        ((
                "socket" => "/var/run/php/php7.0-fpm.sock",
                "broken-scriptfilename" => "enable"
        ))
)
```

Joulinar, 2020-05-06:
did you tried to install php7.0-mysql??

`apt install php7.0-mysql`

or switching to PHP7.3 by changing settings in `15-fastcgi-php.conf`?

Heroldgray, 2020-05-06:
> did you tried to install php7.0-mysql??
> apt install php7.0-mysql

Bingo @Joulinar , thank you ...

Joulinar, 2020-05-06:
just a side note, but pls keep in mind that support for PHP7.0 ended Jan 2019, means it's already eol

https://www.php.net/supported-versions.php

Heroldgray, 2020-05-07:
Yes I agree, but isn't it a problem with ALLO Web GUI especially when DietPi is updated ? 
ALLO users are often not experts in software versions. Are you in touch with ALLO to make things more robust ? However thank's for your work and excellent support.

Joulinar, 2020-05-07:
Personally I don't think it's an issue of ALLO directly. Basically you missed a PHP extension that is needed to communicate between web server and database. It was basically the underlying infrastructure and would have similar impact to other web based application. ALLO is just something on top of this. 

Regarding the update: Before DietPi update is executed themselves, DietPi is calling standard Debian update function `apt-get update` and `apt-get upgrade`. This will update all packages that are managed by `apt`, including web server and PHP extensions. Nothing special on this.

Unfortunately we don't know what happen during your update as we don't have any logs. Therefore we are not really able to find out why the php7.0-mysql package was missing.

MichaIng, 2020-05-07:
@Heroldgray @Joulinar 
Many thanks for report and solution guys. Indeed since a few versions ago DietPi systems were migrated to PHP7.3. Strange is that Lighttpd config was not migrated as well, that should have been done here, actually: https://github.com/MichaIng/DietPi/blob/master/dietpi/patch_file#L1593

If it is working now, this is okay, but generally I suggest to keep/switch to using PHP7.3 and purge PHP7.0. It is relatively safe to test:
```
sed -i "s@\"socket\".*\$@\"socket\" => \"/run/php/7.3-fpm.sock\",@" /etc/lighttpd/conf-available/15-fastcgi-php.conf
systemctl restart lighttpd
```
Try to access Allo web GUI since it (the webserver) now uses PHP7.3. If it works, PHP7.0 can be safely purged:
```
apt purge php7.0-*
apt autoremove --purge
```

Heroldgray, 2020-05-10:
Thank you @MichaIng and @Joulinar,
I've edited /etc/lighttpd/conf-available/15-fastcgi-php.conf to use php7.3-fpm.sock and it's now  OK (i've purged php7.0). 

Additional question: Any hint to avoid some services (RoonBridge, NEtData, SqueezelLite ...) to auto-start at Boot ?

Joulinar, 2020-05-10:
@Heroldgray 
you can disable the services. This should prevent them to start during boot

`systemctl disable <your.service>`

MichaIng, 2020-05-10:
@Heroldgray 
Additionally the services which are started/controlled by DietPi can be prevented from starting via `dietpi-services` command. Select the service you don't want to have started, and `exclude` it.

closed 2020-07-09 by MichaIng